apache/nuttx #8616: gdb function info_nxthreads show the PC incorrectly for cortex-m

chirping78, 2023-02-22:
Here are some steps to reproduce the issue:

1. build a cortex-m board for qemu
```
./tools/configure.sh -E lm3s6965-ek:qemu-flat
make
```
2. start qemu
```
qemu-system-arm   -M lm3s6965evb   -device loader,file=nuttx   -serial mon:stdio   -nographic  -s
```
3. gdb attach
```
gdb-multiarch -ix tools/nuttx-gdbinit nuttx
(gdb) target extended-remote localhost:1234
(gdb) info_nxthreads
saved current_tcb (pid=0)
* 0 Thread 0x20001538  (Name: Idle Task, State: Running, Priority: 0, Stack: 456/1000) PC: 0x96b2 in up_idle()
saved current_tcb (pid=0)
  1 Thread 0x20005060  (Name: hpwork, State: Waiting,Semaphore, Priority: 224, Stack: 344/1992) PC: 0x487d in work_thread()
saved current_tcb (pid=0)
  2 Thread 0x20005c30  (Name: nsh_main, State: Waiting,Semaphore, Priority: 100, Stack: 1080/2000) PC: 0x1 in _vectors()
saved current_tcb (pid=0)
  3 Thread 0x200069f0  (Name: NTP daemon, State: Waiting,Semaphore, Priority: 100, Stack: 1688/1952) PC: 0x1 in _vectors()
saved current_tcb (pid=0)
  4 Thread 0x20008430  (Name: telnetd, State: Waiting,Semaphore, Priority: 100, Stack: 616/2008) PC: 0x20008ec4 in No()
saved current_tcb (pid=0)
saved current_tcb (pid=0)
saved current_tcb (pid=0)
saved current_tcb (pid=0)
(gdb)
```
You can see the PC of  thread 2 and 3 is `0x1`, that is an invalid address. Try the gdb function `nxthread_all_bt` will show incorrect backtrace for these threads.

chirping78, 2023-02-22:
This issue can also be reproduced on real hardware. Here are steps on stm32f4discovery.

1. build nuttx
```
./tools/configure.sh -E stm32f4discovery:nsh
make
kconfig-tweak --enable CONFIG_DEBUG_SYMBOLS
make oldconfig
make
```

2. connect USB cable and start openocd
```
openocd -f interface/stlink-v2.cfg -f target/stm32f4x.cfg 
```

3. gdb attach
```
gdb-multiarch -ix tools/nuttx-gdbinit nuttx
(gdb) target extended-remote localhost:3333
(gdb) load
(gdb) c

^C
Program received signal SIGINT, Interrupt.
up_idle () at chip/stm32_idle.c:191
191       END_IDLE();
(gdb) info_nxthreads
saved current_tcb (pid=0)
* 0 Thread 0x20000230  (Name: Idle Task, State: Running, Priority: 0, Stack: 0/1000) PC: 0x8004e6e in up_idle()
saved current_tcb (pid=0)
saved current_tcb (pid=0)
  2 Thread 0x10000db0  (Name: nsh_main, State: Waiting,Semaphore, Priority: 100, Stack: 0/2000) PC: 0x1 in No()
saved current_tcb (pid=0)
saved current_tcb (pid=0)
(gdb)

```

acassis, 2023-02-25:
Yes, same issue here:
```
(gdb) info_nxthreads
target examined
_target_arch.name=armv7
$_target_has_fpu : 1
$_target_has_smp : 0
saved current_tcb (pid=0)
* 
0 Thread 0x20001538  (Name: Idle Task, State: Running, Priority: 0, Stack: 464/1000) PC: 0x9612 in up_idle()
saved current_tcb (pid=0)
  
1 Thread 0x20005060  (Name: hpwork, State: Waiting,Semaphore, Priority: 224, Stack: 320/1992) PC: 0x47dd in work_thread()
saved current_tcb (pid=0)
  
2 Thread 0x20005c30  (Name: nsh_main, State: Waiting,Semaphore, Priority: 100, Stack: 1016/2000) PC: 0x1 in _vectors()
saved current_tcb (pid=0)
  
3 Thread 0x20006b40  (Name: NTP daemon, State: Waiting,Signal, Priority: 100, Stack: 864/1952) PC: 0x0 in _vectors()
saved current_tcb (pid=0)
  
4 Thread 0x20008540  (Name: telnetd, State: Waiting,Semaphore, Priority: 100, St--Type <RET> for more, q to quit, c to continue without paging--
ack: 616/2008) PC: 0x20008fd4 in No()
saved current_tcb (pid=0)
saved current_tcb (pid=0)
saved current_tcb (pid=0)
saved current_tcb (pid=0)
(gdb) 
```

acassis, 2023-02-25:
@chirping78 I created a documentation guide using your steps here, as I don't know your name I just cited this Issue here, I hope you are ok with it.

chirping78, 2023-02-28:
Take some time look into this today. It seem relates to how the gdb script calculate whether the core has FPU.

chirping78, 2023-05-06:
Confirmed this PR #9198 fixed for both `lm3s6965-ek:qemu-flat` and `stm32f4discovery:nsh`.
Thanks @anchao

closed 2023-05-07 by xiaoxiang781216