apache/nuttx #8666: Exporting LDELFFLAGS elf module definitions with CONFIG_BUILD_KERNEL

g2gps, 2023-02-27:
While working on #8643 I found that the ELF linker flags from the toolchain are not exported, in order to be used when building in the application directory. For example, in `arch/risc-v/src/common/Toolchain.defs`, the linker flags are defined.

```
ifeq ($(CONFIG_ARCH_RV32),y)
LDFLAGS += -melf32lriscv
else
LDFLAGS += -melf64lriscv
endif
```

But after exporting, and importing into apps, with:
```
make export V=1
cd ../apps
./tools/mkimport.sh -z -x ../nuttx/nuttx-export-*.tar.gz
```

I get the following, under `apps/import/scripts/Make.defs`:
```
ARCHCFLAGS       = -fno-common -Wall -Wstrict-prototypes -Wshadow -Wundef
ARCHCPUFLAGS     = -march=rv32imac -mabi=ilp32 -mcmodel=medany
ARCHCXXFLAGS     = -fno-common -nostdinc++ -Wall -Wshadow -Wundef -fno-exceptions -fcheck-new -fno-rtti
ARCHPICFLAGS     = -fpic -msingle-pic-base
ARCHWARNINGS     = 
ARCHWARNINGSXX   = 
ARCHOPTIMIZATION = -fno-omit-frame-pointer -fno-optimize-sibling-calls -ffunction-sections -fdata-sections -g
CROSSDEV         = riscv64-unknown-elf-
CC               = riscv64-unknown-elf-gcc
CXX              = riscv64-unknown-elf-g++
CPP              = riscv64-unknown-elf-gcc -E -P -x c
LD               = riscv64-unknown-elf-ld
AR               = riscv64-unknown-elf-ar rcs
NM               = riscv64-unknown-elf-nm
STRIP            = riscv64-unknown-elf-strip --strip-unneeded
OBJCOPY          = riscv64-unknown-elf-objcopy
OBJDUMP          = riscv64-unknown-elf-objdump
NXFLATLDFLAGS1   = 
NXFLATLDFLAGS2   = 
OBJEXT           = .o
LIBEXT           = .a
EXEEXT           = 
HOSTCC           = cc
HOSTINCLUDES     = 
HOSTCFLAGS       = -O2 -Wall -Wstrict-prototypes -Wshadow -DHAVE_STRTOK_C=1
HOSTLDFLAGS      = 
HOSTEXEEXT       = 
LDNAME           = ld-kernel32.script
```

The ELF link flags are not included, and would not be used either way, as they are not picked up in import's make definitions (`/apps/import/Make.defs`)
```
LDELFFLAGS = -r -e _start -Bstatic
LDELFFLAGS += $(addprefix -T,$(call CONVERT_PATH,$(TOPDIR)/scripts/gnu-elf.ld))
```

I can hack them in:
```
diff --git a/import/Make.defs b/import/Make.defs
index 0405e412b..7b6fb7ed5 100644
--- a/import/Make.defs
+++ b/import/Make.defs
@@ -90,5 +90,6 @@ endif
 
 # ELF module definitions
 
-LDELFFLAGS = -r -e _start -Bstatic
+LDELFFLAGS = -melf32lriscv
+LDELFFLAGS += -r -e _start -Bstatic
 LDELFFLAGS += $(addprefix -T,$(call CONVERT_PATH,$(TOPDIR)/scripts/gnu-elf.ld))
```

But obviously, that is not a good solution.

Perhaps this is something which is unique to the risc-v toolchain setup? It seems that the LD flags aren't uniquely exported either way, so this issue may have not come up previously. Does the export script need to be extended to pickup any extra required linker flags, or am I overlooking something simple?

pussuw, 2023-03-07:
I would start by looking into tools/mkexport.sh

That is the script that generates the file apps/import/scripts/Make.defs

Via that you can add something like "ARCHLDFLAGS" or something along those lines, there are many examples of adding such "extra flags" from the arch folder into the full link command.

casaroli, 2026-11-08:
Fixed by #8870.

closed 2026-11-08 by casaroli