apache/nuttx #8690: Enabling WIFI Networking compile failed, did i forgot to enable something?

onyx99, 2023-03-01:
OS: Ubuntu 22.04.2 LTS
Nuttx commit: 16ea80e

errors: 
implicit declaration of function 'task_tls_alloc', 'task_tls_get_value', 'task_tls_set_value', 'iob_update_pktlen'

---
this is what i tried:

mkdir ~/nuttxspace
cd ~/nuttxspace

git clone https://bitbucket.org/nuttx/tools.git
git clone https://github.com/apache/incubator-nuttx.git nuttx
git clone https://github.com/apache/incubator-nuttx-apps.git apps

cd ~/nuttxspace/tools/kconfig-frontends
autoreconf -f -i
make
sudo make install

mkdir ~/nuttxspace/esp-bins
curl -L "https://github.com/espressif/esp-nuttx-bootloader/releases/download/latest/bootloader-esp32.bin" -o ~/nuttxspace/esp-bins/bootloader-esp32.bin
curl -L "https://github.com/espressif/esp-nuttx-bootloader/releases/download/latest/partition-table-esp32.bin" -o ~/nuttxspace/esp-bins/partition-table-esp32.bin

cd ~/nuttxspace/nuttx

./tools/configure.sh esp32-devkitc:nsh

make menuconfig

* Networking Support: yes
        * Link layer support
            * Late driver initialization: yes
    * System Type
        * ESP32 Peripheral Selection
            * Wireless: yes
    * RTOS Features
        * Work queue support
            * Generic work notifier
            * High priority (kernel) worker thread
        * Pthread Options
            *  Enable mutex types
    * Device Drivers
        * Wireless Device Support
            * IEEE 802.11 Device Support

make

Create version.h
LN: platform/board to /home/user/nuttxspace/apps/platform/dummy
Register: nsh
Register: sh
Downloading: ESP Wireless Drivers
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
100 9992k    0 9992k    0     0  2364k      0 --:--:--  0:00:04 --:--:-- 2900k
Unpacking: ESP Wireless Drivers
CC:  chip/esp32_wifi_adapter.c chip/esp32_wifi_adapter.c: In function 'esp_thread_semphr_get':
chip/esp32_wifi_adapter.c:1259:13: warning: implicit declaration of function 'task_tls_alloc'; did you mean 'esp_malloc'? [-Wimplicit-function-declaration]
       ret = task_tls_alloc(esp_thread_semphr_free);
             ^~~~~~~~~~~~~~
             esp_malloc
chip/esp32_wifi_adapter.c:1269:17: warning: implicit declaration of function 'task_tls_get_value'; did you mean 'nxsem_get_value'? [-Wimplicit-function-declaration]
   sem = (void *)task_tls_get_value(wifi_task_key);
                 ^~~~~~~~~~~~~~~~~~
                 nxsem_get_value
chip/esp32_wifi_adapter.c:1279:13: warning: implicit declaration of function 'task_tls_set_value'; did you mean 'nxsem_get_value'? [-Wimplicit-function-declaration]
       ret = task_tls_set_value(wifi_task_key, (uintptr_t)sem);
             ^~~~~~~~~~~~~~~~~~
             nxsem_get_value
CC:  arp/arp_format.c arp/arp_format.c: In function 'arp_format':
arp/arp_format.c:112:3: warning: implicit declaration of function 'iob_update_pktlen' [-Wimplicit-function-declaration]
   iob_update_pktlen(dev->d_iob, sizeof(struct arp_hdr_s));
   ^~~~~~~~~~~~~~~~~

tmedicci, 2023-03-03:
Hi!

Instead of using the `esp32-devkitc:nsh`, you can try `./tools/configure.sh esp32-devkitc:wapi` or `./tools/configure.sh esp32-devkitc:sta_softap`. 

In order to check the active macros, you can just compare the `defconfig` files on https://github.com/apache/nuttx/tree/master/boards/xtensa/esp32/esp32-devkitc/configs

Please, take a look at the documentation on https://nuttx.apache.org/docs/latest/platforms/xtensa/esp32/index.html#wi-fi

onyx99, 2023-03-04:
Hi Tiago, thank you very much!

closed 2023-03-04 by onyx99