apache/nuttx #8719: Xtensa_user_panic when trying to ping from nsh console in a telnet session

onyx99, 2023-03-05:
Wifi Telnet Session on ESP32 esp32-devkitc (wifinsh config)

nsh> ping google.com

xtensa_user_panic: User Exception: EXCCAUSE=001c task: Telnet session
_assert: Current Version: NuttX  12.0.0 e696e99d9a Mar  4 2023 22:33:58 xtensa
_assert: Assertion failed panic: at file: common/xtensa_assert.c:187 task: Telnet session 0x400d7a4c
up_dump_register:    PC: 400dda98    PS: 00060f30
up_dump_register:    A0: 800ddb84    A1: 3ffed950    A2: ffffffe7    A3: 00000118
up_dump_register:    A4: 0000001f    A5: 3ffed990    A6: 00008b00    A7: 3ffecba8
up_dump_register:    A8: 00000100    A9: 3ffed930   A10: ffffffe7   A11: 00000118
up_dump_register:   A12: 0000001f   A13: 3ffedb18   A14: 3ffec4a0   A15: 00000000
up_dump_register:   SAR: 0000001f CAUSE: 0000001c VADDR: 00000050
up_dump_register:  LBEG: 400014fd  LEND: 4000150d  LCNT: fffffffe
dump_stack: User Stack:
dump_stack: sp:     0x3ffed738
dump_stack:   base: 0x3ffed090
dump_stack:   size: 00003008
stack_dump: 0x3ffed720: 3f4005a5 000000bb 800d6e90 3ffed818 3f4005a5 000000bb 3ffec91c 400d7a4c
stack_dump: 0x3ffed740: 00000000 00000000 7474754e 00000058 00000000 00000000 800ed2e4 3ffe00a8
stack_dump: 0x3ffed760: 00000000 3f4005bc 00000046 400f3038 400f300c 401363dc 0000000a 2e323100
stack_dump: 0x3ffed780: 00302e30 00000000 3ffed7a8 3ffed8b0 36650000 39653639 61396439 72614d20
stack_dump: 0x3ffed7a0: 20342020 33323032 3a323220 353a3333 00000038 3ffed7e0 3ffed818 3ffed7f8
stack_dump: 0x3ffed7c0: 00000008 657478f0 0061736e 3ffed7f8 00000008 00000000 800d67ea 3ffed838
stack_dump: 0x3ffed7e0: 0000001c 3ffed878 00000bc0 3ffed7f8 00000008 3ffed840 3ffed738 3ffec8a0
stack_dump: 0x3ffed800: 00000001 3ffedc50 800d67ea 3ffed838 0000001c 3ffed878 3f4001dd 0000001c
stack_dump: 0x3ffed820: 3ffec91c 3ffbcc78 4008096b 3ffed858 0000001c 3ffed878 3ffb00d4 3ffed910
stack_dump: 0x3ffed840: 00000000 00000118 00040023 3ffed878 00000001 3ffed8e0 00000100 3ffed930
stack_dump: 0x3ffed860: ffffffe7 00000118 3ffae950 3ffb00d8 00000bc0 3ffed050 400dda98 00060f30
stack_dump: 0x3ffed880: 800ddb84 3ffed950 ffffffe7 00000118 0000001f 3ffed990 00008b00 3ffecba8
stack_dump: 0x3ffed8a0: 00000100 3ffed930 ffffffe7 00000118 0000001f 3ffedb18 3ffec4a0 00000000
stack_dump: 0x3ffed8c0: 0000001f 0000001c 00000050 00000000 400014fd 4000150d fffffffe 40080955
stack_dump: 0x3ffed8e0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
stack_dump: 0x3ffed900: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
stack_dump: 0x3ffed920: 00000000 00000000 400f0078 00000118 800d259f 3ffed960 3ffec4a0 00000000
stack_dump: 0x3ffed940: 800d46d0 3ffed980 3ffecba8 00000118 3ffed9b0 3ffed990 00000008 3ffec4a0
stack_dump: 0x3ffed960: 0000001f 3ffed990 00000008 3ffecba8 800dfe61 3ffed9d0 3ffec3e8 00000118
stack_dump: 0x3ffed980: 3ffed9b0 3ffed990 00000008 00000064 3ffedb14 3ffec860 0000001f 3ffec4a0
stack_dump: 0x3ffed9a0: 00000800 3f402e40 00000001 00000000 0000001f 3ffec4a0 00000800 3f402e40
stack_dump: 0x3ffed9c0: 800dffc6 3ffed9f0 3ffec3e8 00000118 0000001f 00000800 400eedfc 3ffedb14
stack_dump: 0x3ffed9e0: 800d87f3 3ffeda70 00000000 00000118 800dbf7d 3ffeda50 0000000c 400eedfc
stack_dump: 0x3ffeda00: 00000000 00000000 400dbb68 00000000 800d4822 3ffeda40 3ffecba8 3ffecbb4
stack_dump: 0x3ffeda20: 00000000 00000000 00ff0000 ff000000 400eedfc 00000000 3ffedb14 3ffedaa0
stack_dump: 0x3ffeda40: 3ffedaa0 3ffeda80 00000008 3ffedb14 0000030b 0000001f 3ffeb040 3ffedaa0
stack_dump: 0x3ffeda60: 800daf16 3ffedac0 3ffedc60 00000118 3ffedaa0 3ffeda80 00000008 3ffec3e8
stack_dump: 0x3ffeda80: 3ffedab4 3ffedaa0 0000001f 3ffedaa0 3ffedb14 00000000 3ffedc60 3ffedec8
stack_dump: 0x3ffedaa0: 0000001f 3ffedaa0 3ffedb14 00000000 800d84dc 3ffedae0 3ffedc60 3ffedec8
stack_dump: 0x3ffedac0: 0000001f 00000000 3f402e68 00000114 800d8701 3ffedb10 00000002 3ffedec8
stack_dump: 0x3ffedae0: 00000000 3ffedb20 00000001 3f401c8f 0000001f 400d87e4 3ffede60 3f400d77
stack_dump: 0x3ffedb00: 800dac29 3ffedbb0 3ffedc60 3ffedec8 3ffedec8 3ffedecd 00000000 00000000
stack_dump: 0x3ffedb20: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
stack_dump: 0x3ffedb40: 00000000 00000000 00000000 00000000 00000000 3ffedb90 3ffedec8 0000012c
stack_dump: 0x3ffedb60: 0000012c 3ffeded7 3ffec9a4 3f400d77 00000008 0000000e 400dbb68 00000000
stack_dump: 0x3ffedb80: 00000000 3ffedbd0 3ffedc60 3ffedec8 3ffedc60 3ffede60 00000000 00000000
stack_dump: 0x3ffedba0: 800d7a5f 3ffedbd0 3ffedc60 3ffedec8 00000001 3ffede60 3ffedec8 ff000000
stack_dump: 0x3ffedbc0: 800d4b73 3ffedbf0 00000001 3ffed070 3f400d41 3ffed070 3ffebe90 00000000
stack_dump: 0x3ffedbe0: 800d29fb 3ffedc10 400d7a4c 00000001 3ffedc60 3ffecea8 3ffec940 3ffb0350
stack_dump: 0x3ffedc00: 00000000 3ffedc30 00000000 00000000 3ffed070 00000016 00000000 3ffebf78
stack_dump: 0x3ffedc20: 00000000 3ffedc50 00000000 00000000 00000000 00000000 00000000 00000000
show_tasks:    PID GROUP PRI POLICY   TYPE    NPX STATE   EVENT      SIGMASK  STACKBASE  STACKSIZE   COMMAND
dump_task:       0     0   0 FIFO     Kthread N-- Ready              00000000 0x3ffbbd50      3040   Idle Task
dump_task:       1     1 100 RR       Kthread --- Waiting Semaphore  00000000 0x3ffaeda0      1984   lpwork 0x3ffb0488
dump_task:       2     2 100 RR       Task    --- Waiting Semaphore  00000000 0x3ffe04a0      3008   nsh_main
dump_task:       3     3 223 RR       Kthread --- Waiting Semaphore  00000000 0x3ffe3670      1984   rt_timer
dump_task:       4     4 253 RR       Kthread --- Waiting MQ empty   00000000 0x3ffe4470      6608   wifi
dump_task:       5     5 100 RR       Task    --- Waiting Semaphore  00000000 0x3ffeb860      2000   telnetd
dump_task:      30    30 100 RR       Task    --- Running            00000000 0x3ffed090      3008   Telnet session
dump_task:      31    31 100 RR       Task    --- Ready              00000000 0x3ffee210      1984   ping google.com

no1wudi, 2023-03-06:
I meet the same issue from rv-virt:
```
NuttShell (NSH) NuttX-12.0.0
nsh> [    9.850000] [CPU1] ubsan_prologue: ================================================================================
[    9.860000] [CPU1] ubsan_prologue: UBSAN: misaligned-access in netdev/netdev_ioctl.c:1250:12
[    9.860000] [CPU1] handle_misaligned_access: member access within misaligned address 0x9 for type 'struct arpreq'
[    9.860000] [CPU1] handle_misaligned_access: which requires 2 byte alignment
[    9.860000] [CPU1] ubsan_epilogue: ================================================================================
[    9.870000] [CPU1] ubsan_prologue: ================================================================================
[    9.880000] [CPU1] ubsan_prologue: UBSAN: misaligned-access in netdev/netdev_ioctl.c:1251:26
[    9.880000] [CPU1] handle_misaligned_access: member access within misaligned address 0x9 for type 'struct arpreq'
[    9.900000] [CPU1] handle_misaligned_access: which requires 2 byte alignment
[    9.900000] [CPU1] ubsan_epilogue: ================================================================================
[    9.930000] [CPU1] riscv_exception: EXCEPTION: Load access fault. MCAUSE: 00000005, EPC: 80045450, MTVAL: 0000003a
[    9.940000] [CPU1] riscv_exception: PANIC!!! Exception = 00000005
[    9.940000] [CPU1] _assert: Current Version: NuttX  12.0.0 88cac02e09-dirty Mar  6 2023 16:29:33 risc-v
[    9.950000] [CPU1] _assert: Assertion failed panic: at file: common/riscv_exception.c:85 task(CPU1): Telnet session 0x8002a9be
[    9.960000] [CPU1] up_dump_register: EPC: 80045450
[    9.970000] [CPU1] up_dump_register: A0: 00000016 A1: 00000076 A2: 800be414 A3: 00000053
[    9.980000] [CPU1] up_dump_register: A4: 0000000a A5: ffffffcf A6: a0000000 A7: 0000a000
[    9.990000] [CPU1] up_dump_register: T0: 00000000 T1: 10000000 T2: 00000000 T3: 00000000
[    9.990000] [CPU1] up_dump_register: T4: 00000000 T5: 00000000 T6: 00000000
[   10.000000] [CPU1] up_dump_register: S0: 00000009 S1: 0000003a S2: 00000118 S3: 800b2650
[   10.000000] [CPU1] up_dump_register: S4: 00000000 S5: 8008b150 S6: 80017ae6 S7: 00000000
[   10.020000] [CPU1] up_dump_register: S8: 80134824 S9: 00000000 S10: 800b47e8 S11: 800b9564
[   10.030000] [CPU1] up_dump_register: SP: 80134610 FP: 00000009 TP: 00000000 RA: 80045434
[   10.030000] [CPU1] dump_stack: IRQ Stack:
[   10.040000] [CPU1] dump_stack: sp:     0x8010e250
[   10.040000] [CPU1] dump_stack:   base: 0x8010cbc0
[   10.040000] [CPU1] dump_stack:   size: 00002048
[   10.050000] [CPU1] dump_stack: ERROR: IRQ Stack pointer is not within the stack
[   10.050000] [CPU1] dump_stack: User Stack:
[   10.060000] [CPU1] dump_stack: sp:     0x80134610
[   10.060000] [CPU1] dump_stack:   base: 0x80133d60
[   10.070000] [CPU1] dump_stack:   size: 00003008
```
It seems relative to arp (netdev_arp_ioctl), the req is a invalid pointer with value (0x9) on riscv.

wengzhe, 2023-03-06:
Hi, @onyx99 could you help us verify if this patch (https://github.com/apache/nuttx/pull/8739) solves the problem?

closed 2023-03-06 by acassis