apache/nuttx #8731: ESP32C3 risc-v usb-serial not working correctly

janantos, 2023-03-06:
in recent master (last about 7 days) something changed. usb-serial console on esp32s3 started misbehaving, all input characters are doubled and when backspace is pressed if freeze completely. I am using minimal readline SYSTEM_READLINE(=y) as default in configure esp32c3-devkit:nsh, only thing changed from this config is ARCH_CHIP_ESP32C3MINI1, ESP32C3_USBSERIAL=y, ESP32C3_UART0=n. previously this was working perfectly

xiaoxiang781216, 2023-03-06:
Can you rebase both nuttx and apps to the last master?  Especially, the change come from @no1wudi :
https://github.com/apache/nuttx/pulls?q=is%3Apr+author%3Ano1wudi
https://github.com/apache/nuttx-apps/pulls?q=is%3Apr+author%3Ano1wudi
Some change recently need both side work together.

janantos, 2023-03-06:
rebased to latest master and no change. 
tried with `tools/configure.sh -m esp32c3-devkit:usbconsole` with only change `ESP32-C3-MINI-1=y`
and it does misbehave the same way. enabling `NSH_CLE=y` couses complete freeze of console after boot.

xiaoxiang781216, 2023-03-06:
@no1wudi could you look at this?

janantos, 2023-03-06:
also I realized now that NSH_CLE=y does not work in revision db69196 neither, but it works perfectly in release/12.0

no1wudi, 2023-03-07:
Emm, I don't have such a esp32c3 board with USB support, but I confirmed that raspberrypi-pico:usbnsh works fine, this issue is seems relative to esp32c3's usb serial driver.

@gustavonihei Could you have check ?

janantos, 2023-03-07:
Ok, I have received yesterday stm32f429i-disc1 board, will also test there. Do you think it might be also linked to that I am using Apple Silicon as dev machine?

no1wudi, 2023-03-07:
It should be OK with Apple Silicon, could you try `python -m serial.tools.miniterm --raw --eol lf /dev/tty.xxxxx 115200` ?
This is my terminal tool to work with nsh shell, if serial is not found, you can install it by `pip[3] install pyserial'

janantos, 2023-03-07:
tested with pyserial and it freezes on backspace,
tried with `tools/configure.sh -m esp32c3-devkit:usbconsole` with only change `ESP32-C3-MINI-1=y`

no1wudi, 2023-03-08:
Now I can reproduce this issue, I'll investigate it

janantos, 2023-03-08:
I am. Not sure if related but as I mentioned in db69196 revision command line editor is not working, but readline does. However for example when you enable Lua and try to run it, the console also freezes. This also works perfectly in realease/12.0 so this might be some deeper issue or just coincidence, but it might point you somewhere. Unfortunately I am not skilled in NuttX internals yet. :-(

no1wudi, 2023-03-08:
@janantos I've found the root case for this issue, I'll submit the fix later.

slorquet, 2023-03-08:
Please fix this as soon as possible.

no1wudi, 2023-03-08:
> Please fix this as soon as possible.

@slorquet https://github.com/apache/nuttx/pull/8764 this PR fix this issue.

janantos, 2023-03-08:
HI @pkarashchenko and @no1wudi , I would declare this partialy fixed.
readline works, however when enabled NSH_CLE and SYSTEM_CLE console after boot freezes on first connect to serial console, on second connect after couple of enters `nsh: nsh_session: cle failed: Invalid argument` is shown and console sort of works. hovever when trying to run lua, console freezes, again this works perfectly on release/12.0

xiaoxiang781216, 2023-03-08:
let's reopen this issue util all problems are fixed.

no1wudi, 2023-03-09:
@janantos Thanks for information, I'll continue to work on it until all issues fixed.

no1wudi, 2023-03-09:
@janantos Could you share me how do you test with lua?

no1wudi, 2023-03-09:
I guess this issue with cle is relative to esp32c3's usb-serial driver.

You can checkout to previous point of https://github.com/apache/nuttx-apps/commit/95f32fd018be7a66354942647d9e1c778ad5d488,such as 6a686ba93d082cf098cd1dc70943dad30e20d90d, nuttx still use the master, you will find the nsh works well.

Beacuse that commit remove the stdio, then nsh prompt will write direct to the driver instead of the stdio's buffer first, but it seems the driver is not ready in this  time. `readline` works since `cle` is more complex, `cle` will send some vt100 command sequence to terminal and wait the response but `readline` does not, if the uart is not really ready, maybe some bytes lost from command sequence, then you get the `nsh: nsh_session: cle failed: Invalid argument`, but can get works by type more enter.

Another way to validate the root case is by adding a short sleep in `apps/system/nsh/nsh_main.c:56` in latest code:

```c

int main(int argc, FAR char *argv[])
{
  struct sched_param param;
  int ret = 0;

 // Sleep 1ms, new added to debug

  usleep(1000);

  /* Check the task priority that we were started with */

  sched_getparam(0, &param);
  if (param.sched_priority != CONFIG_SYSTEM_NSH_PRIORITY)
    {
      /* If not then set the priority to the configured priority */

      param.sched_priority = CONFIG_SYSTEM_NSH_PRIORITY;
      sched_setparam(0, &param);
    }

  /* Initialize the NSH library */

  nsh_initialize();

#ifdef CONFIG_NSH_CONSOLE
  /* If the serial console front end is selected, run it on this thread */

  ret = nsh_consolemain(argc, argv);

  /* nsh_consolemain() should not return.  So if we get here, something
   * is wrong.
   */

  dprintf(STDERR_FILENO, "ERROR: nsh_consolemain() returned: %d\n", ret);
  ret = 1;
#endif

  return ret;
}

```
Then the nsh works correctly.

@xiaoxiang781216 @janantos @gustavonihei What's your opinion?
BTW, @gustavonihei Is there any way to check the usb-serial is really ready for use ?

janantos, 2023-03-09:
@no1wudi the Lua can be tested just by executing Lua and it should go to interpreter REPL. In release/12.0 you will end up in REPL showing you Lua 5.4 welcome message. In master it just freeze.

no1wudi, 2023-03-09:
@janantos I confirmed lua works fine with `readline` in master, and if `cle` works correctly, lua also works fine.

no1wudi, 2023-03-09:
I'll try to fix this issue by enhance `cle`'s error handling.

janantos, 2023-03-09:
yep, we should take Lua out of this. it may be separate ticket.

no1wudi, 2023-03-09:
@janantos Could you try this patch https://github.com/apache/nuttx-apps/pull/1650?

I confirmed `cle` and `lua` works correctly with usbconsole now.

janantos, 2023-03-09:
@no1wudi Lua works, but CLE is still sort of dodgy on my esp32c3 olimex board. The nsh prompt does not show, but is shows after exactly 11 enter presses, it does not matter how quickly you press.  

https://user-images.githubusercontent.com/19323286/224083373-b7630414-2ec2-4efe-bd17-843dfda02cb5.mov

no1wudi, 2023-03-10:
@janantos 
[image]
Emm, too many getch here without timeout in `cle_getcursor`, this is why you must 11 enter presses.

Only one poll in https://github.com/apache/nuttx-apps/pull/1650 is not enough, I'll continue update the PR.

But I can't reproduce this issue the same as you:

https://user-images.githubusercontent.com/9030616/224197088-f1318d53-97e1-44af-8871-76e5299838d2.mp4

no1wudi, 2023-03-10:
@janantos With latest patch, you don't need to press enter 11 times anymore, but may get some error message at first line:
···
) NuttX-12.0.0
nsh> nsh: nsh_session: cle failed: Invalid argument
nsh> 0;6R
···
The root case is that cle don't get the complete response during nsh startup from your PC's terminal, but works later until the driver become more stable, this issue is related with esp32c3's usb serial driver.

janantos, 2023-03-10:
Ok got it, however don't we have some regression here? while this is perfectly working in release/12.0 ?

janantos, 2023-03-10:
I am in work now, but tomorrow I will test if this error is acceptable for time of being as it gives you instantly console. However I will test tomorrow if this does not affect login when console login enabled.

no1wudi, 2023-03-10:
> Ok got it, however don't we have some regression here? while this is perfectly working in release/12.0?
Emm, I think current implementation is not the real solution.

I'll try to let `cle` get rid of getcursor command.

no1wudi, 2023-03-10:
@janantos You can try the latest PR (https://github.com/apache/nuttx-apps/pull/1650),  it works fine on my board just like release/12.0, but still need your feedback.

janantos, 2023-03-10:
@no1wudi will test during tomorrow.

no1wudi, 2023-03-14:
> @no1wudi will test during tomorrow. 

@janantos do you have any update for this issue？ thanks

janantos, 2023-03-15:
Hi @no1wudi I can confirm that issue is fixed. Thx again for your great job.

closed 2023-03-17 by acassis