apache/nuttx #8750: dd: open failed: 6 on rv-virt:nsh64

GooTal, 2023-03-07:
Hi, i built and run rv-virt:nsh64. I tried dd command. But it seems not working.

[image]

The related configure, make  and run command:
```
./tools/configure -l rv-virt:nsh64
make -j12
qemu-system-riscv64 -semihosting -M virt,aclint=on -cpu rv64 -smp 8 -bios none -kernel nuttx -nographic
```

Did i miss something?

acassis, 2023-03-08:
Good finding @GooTal confirmed on ESP32:
```
NuttShell (NSH) NuttX-12.0.0
nsh> ?
help usage:  help [-v] [<cmd>]

    .         cd        dmesg     free      mkrd      rm        time      uptim    
    [         cp        echo      help      mount     rmdir     true      uslee    
    ?         cmp       env       hexdump   mv        set       truncate  xd        
    basename  dirname   exec      kill      printf    sleep     uname     
    break     dd        exit      ls        ps        source    umount    
    cat       df        false     mkdir     pwd       test      unset     

Builtin Apps:
    nsh  sh   
nsh> mkrd 256
nsh> ls -l /dev
/dev:
 crw-rw-rw-       0 console
 crw-rw-rw-       0 null
 brw-rw-rw-  131072 ram0
 crw-rw-rw-       0 ttyS0
 crw-rw-rw-       0 urandom
 crw-rw-rw-       0 zero
nsh> dd if=/dev/zero of=/dev/ram0 bs=1 count=131072
nsh: dd: open failed: 6
nsh> dd if=/dev/urandom of=/dev/ram0 bs=1 count=131072
nsh: dd: open failed: 6
nsh> 
```

It was broken recently because it used to work on NuttX 11!

closed 2023-04-23 by xiaoxiang781216

GooTal, 2023-05-04:
Wow, Nice work!