apache/nuttx #8766: HCI support over RPTUN

raiden00pl, 2023-03-08:
Hi, is there anyone currently working on the support of BT HCI over RPTUN ? 

A question probably mainly for the Xiaomi team. I prefer to ask to avoid duplication of work before I start.

xiaoxiang781216, 2023-03-08:
> Hi, is there anyone currently working on the support of BT HCI over RPTUN ?
> 

How about btuart_shim+rpmsg-uart?

> A question probably mainly for the Xiaomi team. I prefer to ask to avoid duplication of work before I start implementing it myself.

Or you want to make bt_driver remote directly?

raiden00pl, 2023-03-08:
> How about btuart_shim+rpmsg-uart?

TIL: [shim](https://en.wikipedia.org/wiki/Shim_(computing)) is a valid computer programming term :)

I didn't look at `bt_uart_shim.c` before but I see now this is something that may be helpful.

> Or you want to make bt_driver remote directly?

I was thinking about remote bt_driver, something like HCI RPMSG in Zephyr (https://github.com/zephyrproject-rtos/zephyr/blob/main/drivers/bluetooth/hci/rpmsg.c), but for now I'll try the solution you proposed.

xiaoxiang781216, 2023-03-08:
Both is fine, but we didn't do the Zephyr approach on NuttX yet. If you finally select Zephyr approach, I can help review your change.

raiden00pl, 2023-03-10:
I ended up writing a Zephyr-like HCI RPMSG controller. After a dozen hardfaults and deadlocks, it seems to work. I'll prepare PR tomorrow.

closed 2023-03-14 by xiaoxiang781216