apache/nuttx #8780: ARMV7-M MPU base region not configured correctly

vome101, 2023-03-09:
I made a protected build on NXP 1052 chip which uses the ARMV7-M MCU. The user region has a base address of 0x80800048 with a size of 0xd15620. The alignedbase is 0x80000000 because the l2size is 0x19. The system will crash shortly after I enable the MPU. When I decrease the size of the user, the alignedbase is 0x8080000 and the device runs normally. I don't think the calculation for the aligned base is correct for all situations. I'm using nuttx 10.2.0.

[image]

patacongo, 2023-03-09:
> I made a protected build on NXP 1052 chip which uses the ARMV7-M MCU.
> The user region has a base address of 0x80800048 with a size of
> 0xd15620. The alignedbase is 0x80000000 because the l2size is [...]
I think this is a known issue.  Thanks for formally raising it. The 
issue is discussed in several places like:

https://github.com/apache/nuttx/blob/master/boards/arm/stm32/olimex-stm32-p407/scripts/memory.ld#L38

      * For MPU support, the kernel-mode NuttX section is assumed to be
    128Kb of
      * FLASH and 8Kb of SRAM.  (See
    boards/stm32f4discovery/scripts/kernel-space.ld).
      * Allowing additional memory permitis configuring debug
    instrumentation to
      * be added to the kernel space without overflowing the partition. 
    This could
      * just as easily be set to 256Kb or even 512Kb.
      *
      * Alignment of the user space FLASH partition is also a critical
    factor:
      * The user space FLASH partition will be spanned with a single
    region of
      * size 2**n bytes.  The alignment of the user-space region must be
    the same.
      * As a consequence, as the user-space increases in size, the alignment
      * requirement also increases.
      *
      * This alignment requirement means that the largest user space
    FLASH region
      * you can have will be 512KB at it would have to be positioned at
      * 0x08800000 (it cannot be positioned at 0x0800000 because vectors
    power-up
      * reset vectors are places at the beginning of that range). If you
    change
      * this address, don't forget to change the CONFIG_NUTTX_USERSPACE
      * configuration setting to match and to modify the check in
    kernel/userspace.c.
      *
      * With 112Kb of SRAM a 64Kb user heap would seem possible but it
    is not in
      * the current organization of SRAM memory (that could be changed
    with a
      * little effort).  The current ordering of SRAM is:  (1) kernel
    .bss/.data,
      * (2) user .bss/.data, (3) kernel heap (determined by
    CONFIG_MM_KERNEL_HEAPSIZE),
      * and (4) the user heap.  The maximum size of the user space heap
    is then
      * limited to 32Kb beginning at address 0x20008000.
      *
      * Both of these alignment limitations could be reduced by using
    multiple
      * regions to map the FLASH/SDRAM range or perhaps with some clever
    use of
      * subregions or with multiple MPU regions per memory region.
      *
      * NOTE: The MPU is used in a mode where mappings are not required for
      * kernel addresses and, hence, there are not alignment issues for
    those
      * case.  Only the user address spaces suffer from alignment
    requirements.
      * However, in order to exploit this fact, we would still need to
    change
      * the ordering of memory regions in SRAM.

If you wanted to improve the resolution of the MPU region allocation, 
that would be a very good contribution.

fjpanag, 2023-06-07:
Just for the reference, this seems duplicate of #1489.