apache/nuttx #8790: Errors getting started with C++ example

icolwell-as, 2023-03-10:
Hi There, I'm new to nuttx, so hopefully I'm just missing something simple here.

I'm running Ubuntu 20.04
I followed the instructions here on the `nuttx-12.0.0` release tag:
https://nuttx.apache.org/docs/12.0.0/quickstart/install.html
Installed all build tools and dependencies as specified.

Then, I moved onto this guide:
https://nuttx.apache.org/docs/12.0.0/guides/cpp_cmake.html

The first step is running the following command:
 ```
./tools/configure.sh stm32f4discovery:testlibcxx
```
When I run this command I get the following error:
```
...
LN: arch/arm/src/board/board to /home/ian/nuttx/boards/arm/stm32/stm32f4discovery/src
touch .dirlinks
make[1]: Leaving directory '/home/ian/nuttx'
Kconfig:2110: can't open file "/Kconfig"
make: *** [tools/Unix.mk:614: olddefconfig] Error 1
ERROR: failed to refresh
```

I'll start digging into the configure script to try to understand what's going on, but just curious if anyone else is able to reproduce this issue or can point me to something simple I might be missing.

icolwell-as, 2023-03-10:
Thank @TimJTi , I tried that and it results the same error unfortunately.

I'm starting to think it's some kinda directory or empty/missing variable issue. The error is `can't open file "/Kconfig"`, but it should likely be looking for `/home/ian/nuttx/Kconfig` which is a file that exists (or some other Kconfig in the tree, I just realized these Kconfig files are all over the place).

Is it maybe because I don't have an `apps` directory?

TimJTi, 2023-03-10:
I have certainly seen odd issues, worse recently perhaps, with path issues to the apps directory. I would certainly clone that repo and see.

acassis, 2023-03-11:
@icolwell-as normall the error
```
Kconfig:2204: can't open file "/Kconfig"
make: *** [tools/Unix.mk:656: olddefconfig] Error 1
```

Happens when you have the directoyu nuttxspace/nuttx but don't have the nuttexspace/apps

Please double check you cloned both directories inside same diretory (i.e. inside nuttxspace/)

acassis, 2023-03-12:
@icolwell-as please config that the issue was missing the apps/ directory and if so please close this issue.

closed 2023-03-13 by icolwell-as

pkarashchenko, 2023-03-13:
I think that is general the apps is optional, but only if your defconfig does not rely on anything from apps repo. For example if you want to build kernel only. But if you want to use NSH app for example then obviously apps repo becomes mandatory :)