apache/nuttx #8798: use esp8266 base on stm32

laoniaokkk, 2023-03-12:
I want use a module esp8266 as wifi(sta or ap mode) base on stm32,but there is no example tell me how to do it.I need help!

armsqr, 2023-03-13:
Take a look at ESP-AT
https://docs.espressif.com/projects/esp-at/en/latest/esp32/index.html
and LWESP
https://docs.majerle.eu/projects/lwesp/en/latest/

anjiahao1, 2023-03-15:
If it is simpler, you can send the at command through the serial port at the application layer to allow the stm32 and esp8266 to communicate to obtain network information.
If it is more complicated, it can be implemented at the kernel layer by writing an esp8266 network device driver and communicating through the serial port. I'm not very familiar with this.
It is recommended that you use the simple method.

laoniaokkk, 2023-04-02:
> Take a look at ESP-AT https://docs.espressif.com/projects/esp-at/en/latest/esp32/index.html and LWESP https://docs.majerle.eu/projects/lwesp/en/latest/

I found some errors in the file esp8266.c 
used in esp8266.c                          defined in esp8266.h
LESP_BSSID_SIZE                --->      lespBSSID_SIZE
LESP_SSID_SIZE                  --->      lespSSID_SIZE 
LESP_SECURITY_NBR         --->     lesp_eSECURITY_NBR
LESP_MODE_STATION       --->     lesp_eMODE_STATION
LESP_SECURITY_NONE     --->      lesp_eSECURITY_NONE
LESP_SECURITY_WEP        --->      lesp_eSECURITY_WEP
LESP_SECURITY_WPA_PSK       --->      lesp_eSECURITY_WPA_PSK
LESP_SECURITY_WPA2_PSK  ---> lesp_eSECURITY_WPA2_PSK
LESP_SECURITY_WPA_WPA2_PSK ---> lesp_eSECURITY_WPA_WPA2_PSK

And in function "lesp_bind", when lesp_check() return  value >=0 , there is an error,but in fact there is no error