apache/nuttx #8813: telnet

xiaotailang, 2023-03-14:
I am debugging the telnet-related functions on stm32f767. During the test, I found that stm32f767 can connect to PC via telnet as a server and enter the nuttx system normally. However, when stm32f767 works as a client to connect to PC, it cannot connect to PC normally. When I tracked and debugged the program, I found that the program can return normally during connect(), but when it comes to send(), it returns an error code "send() failed: 22". I don't know the reason for this and hope to get some suggestions and experience from you. My PC has installed the telnet server correctly and started the telnet service. My code is nuttx _8.2 and configuration is as follows:
CONFIG_NSH_TELNET=y
CONFIG_NSH_TELNETD_PORT=y
CONFIG_NSH_TELNETD_DAEMONPRIO=y
CONFIG_NSH_TELNETD_DAEMONSTACKSIZE=y
CONFIG_NSH_TELNETD_CLIENTPRIO=y
CONFIG_NSH_TELNETD_CLIENTSTACKSIZE=y
SYSTEM_TELNET_CLIENT=y

acassis, 2023-03-14:
Error -22 means "Invalid argument", please enable the debug NETWORK (err, warn and info) to find where and why this error is happening.

```
$ make menuconfig

Build Setup  --->
Debug Options  --->
[*] Enable Debug Features
[*]   Enable Error Output
[*]     Enable Warnings Output
[*]       Enable Informational Debug Output
...
[*]   Network Debug Features
[*]     Network Error Output
[*]     Network Warnings Output
[*]     Network Informational Output
```

xiaotailang, 2023-03-15:
> Error -22 means "Invalid argument", please enable the debug NETWORK (err, warn and info) to find where and why this error is happening.
>
> ```
> $ make menuconfig
>
> Build Setup  --->
> Debug Options  ---> [...]

okay! thank you very much.