apache/nuttx #8815: the md5's hash values on nuttx are different from that value in linux for the same file

gneworld, 2023-03-14:
With the same file, get two md5 hash values

1bea0cd956c139e21179c4213a10c1f3 on nuttx
f11fcab02b7cbc5adcd6d6acdff24490 on linux

so, What is the cause?

acassis, 2023-03-14:
@gneworld good finding! Probably you found an issue in NuttX MD5 algo.
I tested MD5 on NuttX using ESP32 internet MD5 hardware calculator and it worked: https://acassis.wordpress.com/2022/11/08/testing-hmac-md5-on-esp32/

But I didn't test it using NuttX MD5, if I had tested it probably should find this issue that you found! :-)

closed 2023-03-15 by gneworld

fjpanag, 2023-06-07:
I just did some quick tests of the md5 algorithm.

I checked NuttX's md5 against our "own" implementation which is known to be correct and agrees 100% with the Linux algorithm for the last several years.

For all files provided to the algorithms, the results were identical.  
I believe there is nothing wrong with NuttX MD5.

---

However. MD5 is **crucial** to be correct.  
In embedded systems is used extensively to validate OTA received files, firmware updates etc.

I would like someone else to repeat the test.  
Or maybe, such algorithms should have unit tests in CI?

Shall we re-open this issue, till it is verified correct?

@acassis what do you think? Can you help with a test of NuttX vs ESP32 MD5?