apache/nuttx #8831: BMP280 sensor is not working with sensortest

acassis, 2023-03-15:
I noticed that sensortest is not working with BMP280 sensor

How to test:

1) connect a BMP280 module to I2C pins 22 and 23 of ESP32-Devkitc
2) run: $ ./tools/configure.sh esp32-devkitc:bmp280
3) run: $ make flash ESPTOOL_PORT=/dev/ttyUSB1 -j

Use minicom or other serial console and run:

```
nsh> ls /dev
/dev:
 console
 i2c0
 kmsg
 null
 ttyS0
 uorb/

nsh> ls /dev/uorb
/dev/uorb:
 sensor_baro0

nsh> sensortest baro0
sensor_ioctl: cmd=a81 arg=000f4240
bmp280_set_standby: Failed to set value for standby time.
Failed to set interval for sensor:/dev/uorb/sensor_baro0, ret:Unknown error
```

I confirmed that commenting these lines solves the issue:

```
diff --git a/testing/sensortest/sensortest.c b/testing/sensortest/sensortest.c
index 3fd24d027..65b73d277 100644
--- a/testing/sensortest/sensortest.c
+++ b/testing/sensortest/sensortest.c
@@ -346,7 +346,7 @@ int main(int argc, FAR char *argv[])
       goto open_err;
     }
 
-  ret = ioctl(fd, SNIOC_SET_INTERVAL, interval);
+  /*ret = ioctl(fd, SNIOC_SET_INTERVAL, interval);
   if (ret < 0)
     {
       ret = -errno;
@@ -356,7 +356,7 @@ int main(int argc, FAR char *argv[])
                  devname, strerror(errno));
           goto ctl_err;
         }
-    }
+    }*/
 
   ret = ioctl(fd, SNIOC_BATCH, latency);
   if (ret < 0)
```

@xiaoxiang781216 could you please take a look?

Donny9, 2023-03-16:
@acassis i will try this config with sensortest, If there is a problem, I will synchronize in time.

acassis, 2023-05-03:
Hi @Donny9 did you test the sensor with some board config?