apache/nuttx #8833: w5500 ethernet can not work well

laoniaokkk, 2023-03-16:
When i use w5500 as a eth device,i found that when eth bring up, it only write the mac register, and ip netmask gateway is configured but not write into it's register. Then i wirte the config into ip register(netmask gateway etc.), i can ping my board by PC,but can not ping PC by my board. And in the driver of w5500, there is only socket0 is supported,in fact the w5500 is supported as much as 8 sockets. hope to fix this problem,thanks.

acassis, 2023-03-16:
@michi-jung did you face this issue when you created the driver?

michi-jung, 2023-03-16:
The w5500 driver uses the chip in MACRAW mode, i.e. only at the Ethernet level.  All IP processing is done by NuttX's networking stack.  This is why only the MAC is written to a chip register and why only 'socket0' is used to exchange raw Ethernet frames.

You wrote
> Then i wirte the config into ip register(netmask gateway etc.) [...]

Do you mean that you extended the code to write the e.g. the netmask gateway into the W5500's registers?  This does not make sense for the MACRAW mode.

acassis, 2023-03-17:
Thank you @michi-jung !!! BTW, do you have a board profile example that uses this w5500 that we could use as reference?

michi-jung, 2023-03-18:
At the moment I only have board profile for one of my customer's proprietary boards, which I can not share.  I am planning to provide a profile for the `b-u585i-iot02a` board, but can't give a date at this point.

laoniaokkk, 2023-03-18:
I just test it on stm32f407,and use the config of boards/arm/stm32/stm32f4discovery/ether_w5500 but change some pins of spi

acassis, 2023-03-18:
Hi @laoniaokkk which pins did you change? Do you please share your modifications?

laoniaokkk, 2023-03-19:
I use this pins for w5500:
PB12 ---> ETH_INT
PB13 ---> SPI_CLK
PB14 ---> SPI_MISO
PB15 ---> SPI_MOSI
PD8 ---> ETH_CS
PE14 ---> EHT_RST

and this is my pin configs:
#define GPIO_SPI2_MISO    GPIO_SPI2_MISO_1
#define GPIO_SPI2_MOSI    GPIO_SPI2_MOSI_1
#define GPIO_SPI2_SCK     GPIO_SPI2_SCK_2
#define GPIO_W5500_CS      (GPIO_OUTPUT|GPIO_PUSHPULL|GPIO_SPEED_50MHz|\
                             GPIO_OUTPUT_SET|GPIO_PORTD|GPIO_PIN8)

#define GPIO_W5500_RESET   (GPIO_OUTPUT|GPIO_PUSHPULL|GPIO_SPEED_50MHz|\
                             GPIO_OUTPUT_SET|GPIO_PORTE|GPIO_PIN14)

#define GPIO_W5500_INTR     (GPIO_INPUT|GPIO_FLOAT|GPIO_EXTI|\
                             GPIO_OPENDRAIN|GPIO_PORTB|GPIO_PIN12)

and this is the ifconfig show on console:
nsh> ifconfig
eth0    Link encap:Ethernet HWaddr 00:e0:de:ad:be:ef at UP
        inet addr:192.168.1.100 DRaddr:192.168.1.1 Mask:255.255.255.0

             IPv4   TCP   UDP  ICMP
Received     0000  0000  0000  0000
Dropped      0000  0000  0000  0000
  IPv4        VHL: 0000   Frg: 0000
  Checksum   0000  0000  0000  ----
  TCP         ACK: 0000   SYN: 0000
              RST: 0000  0000
  Type       0000  ----  ----  0000
Sent         0000  0000  0000  0000
  Rexmit     ----  0000  ----  ----
nsh> ping 192.168.1.3
PING 192.168.1.3 56 bytes of data
No response from 192.168.1.3: icmp_seq=0 time=1000 ms
No response from 192.168.1.3: icmp_seq=1 time=1000 ms
No response from 192.168.1.3: icmp_seq=2 time=1000 ms

I can not ping PC by board or ping board by PC.
Only when i write the registers(ip netmask gateway) in function "w5500_unfence(FAR struct w5500_driver_s *self)",i can ping board by PC,but can not ping PC by board

acassis, 2023-03-19:
Thank you very much!

laoniaokkk, 2023-07-13:
When I debug w5500, i found that: when send an icmp pack to PC, the PC will reply in 1ms. And in the w5500 driver ,it will disable the interrupt first when receive an interrupt ,and enable the interrupt after processed the receive data. It will take some time,and maybe lost  some interrupts(data). So i remove the code of disable interrupt from function "w5500_interrupt", after that the net work well. I don't know if this will cause other problems

michi-jung, 2023-07-13:
@laoniaokkk I believe you are loosing interrupte because of the issue described [here](https://github.com/apache/nuttx/blob/0ee304954a327394e493703ea78b8f6c221d2868/boards/arm/stm32/stm32f4discovery/src/stm32_w5500.c#L141).

In the w5500_lower implementation of my custom board the enable method looks as follows:

```
static void w5500_enable(const struct w5500_lower_s *lower,
                         bool enable)
{     
  struct stm32_w5500_lower_s *self = (struct stm32_w5500_lower_s *)lower;

  stm32_gpiosetevent(self->gpio_intn,
                     false, /* rising edge  */
                     true,  /* falling edge */
                     false, /* event        */
                     enable ? self->handler : NULL,
                     enable ? self->arg : NULL);
      
  /* If interrupt is still pending trigger interrupt handler. */
      
  if (enable && !stm32_gpioread(self->gpio_intn))
    {
      stm32_gpioswirq(self->gpio_intn);
    }   
}
```

I think stm32_w5500.c should be modified accordingly.

michi-jung, 2023-07-13:
[stm32_exti.h.diff.txt](https://github.com/apache/nuttx/files/12036753/stm32_exti.h.diff.txt)
[stm32_exti_gpio.c.diff.txt](https://github.com/apache/nuttx/files/12036754/stm32_exti_gpio.c.diff.txt)
[stm32_w5500.c.txt](https://github.com/apache/nuttx/files/12036761/stm32_w5500.c.txt)
[image]

laoniaokkk, 2023-07-13:
@michi-jung Thank you very much! The net is work well now.

michi-jung, 2023-07-13:
@laoniaokkk Would you be willing to bring your fix upstream?

laoniaokkk, 2023-07-13:
> 
I'm just modifying it(stm32_w5500.c  and stm32_exit_gpio.c) based on your docs. The drive of w5500 is correct, and actually your configuration of stm32_w5500 is valid

michi-jung, 2023-07-13:
@laoniaokkk It would be nice to have the `boards/arm/stm32/stm32f4discovery/ether_w5500` configuration running out of the box on upstream nuttx.  Could you take care of this?  I do not have access to this board.

laoniaokkk, 2023-07-14:
@michi-jung I am sorry but i don't have the stm32f4discovery board yet, i test it on my own board. And i find after some times(a few minutes or hours) late ,the network will "death", and then put "ifconfig" command, the status of net is up but can not  ping

laoniaokkk, 2023-08-04:
@michi-jung     I tested the network with examples/Simple TCP echo server, when sending data to the server in a period of 50ms, after a period of time the w5500 will reset, the network is not available, and then manually restart the network card with the 'ifdown eth0', 'ifup eth0' commands, the network is still unavailable. The same test environment and methodology, using LAN8720 did not cause similar problems.

michi-jung, 2023-08-04:
Hi @laoniaokkk,

could you please provide your w5500_lower implementation?

Thanks!

laoniaokkk, 2023-08-04:
@michi-jung Hi, the w5500.c was just used from nuttx 12.0.0
[config.txt](https://github.com/apache/nuttx/files/12258764/config.txt)
[stm32_exti_gpio.txt](https://github.com/apache/nuttx/files/12258767/stm32_exti_gpio.txt)
[stm32_w5500.txt](https://github.com/apache/nuttx/files/12258768/stm32_w5500.txt)
[w5500.txt](https://github.com/apache/nuttx/files/12258769/w5500.txt)