apache/nuttx #8835: Add a new .a library to the apps directory

xiaotailang, 2023-03-16:
Hello! I have a library file called my.a. I tried creating my own folder called my_dir under apps/examples, and placed the .a library file in it. I also wrote a my.c source file with a function called my_func, which calls a function inside my.a. In the Makefile, I added LDLIBS += -Lapps/examples/my_dir -lmy to link the my.a file. However, when I tried to generate nuttx.bin, it gave an error saying that the function my_func in my.o generated from my.c was undefined when calling the function inside the my.a library file. Is my approach incorrect? Is there anything else I need to consider?

acassis, 2023-03-16:
Hi @xiaotailang other alternative is using the EXTRA_LIBS:
https://acassis.wordpress.com/2020/10/06/linking-external-libraries-on-nuttx/

xiaotailang, 2023-03-17:
> Hi @xiaotailang other alternative is using the EXTRA_LIBS: https://acassis.wordpress.com/2020/10/06/linking-external-libraries-on-nuttx/

Thank you. Following your advice, the compilation has been successful and it can be used normally. Thank you very much for your help. Wish you all the best!

closed 2023-03-17 by pkarashchenko