apache/nuttx #8839: Photon P0 WiFi not functional

TheCranston, 2023-03-17:
Greetings,
I've taken a fresh git clone from gitbox.apache.org of NuttX and apps.  Ran ```./tools/configure.sh photon:wlan```  , ```make menuconfig``` to set the SSID and password for my WiFi AP.  Ran ```make``` without any issues.  Then dfu upload to the board.  

It does not connect to my WiFi network.  I've tried my hotspot on the iphone as well without success.  Follows is the serial console debug log from my attempts.

```
photon_wlan_initialize: Initializing SDIO slot 0
bcmf_sdio_initialize: minor: 0
bcmf_sdio_thread:  Enter
netdev_register: Registered MAC: 00:00:00:00:00:00 as dev: wlan0
netdev_ifr_ioctl: cmd: 1794
netdev_ifr_ioctl: cmd: 1796
netdev_ifr_ioctl: cmd: 1800
netdev_ifr_ioctl: cmd: 1818
bcmf_wl_active: Entered
sdio_probe: rca is 1
sdio_enable_function: Function 1 enabled
sdio_io_rw_extended: prep read 4
sdio_io_rw_extended: Transaction ends
bcmf_chipinitialize: bcm43362 chip detected
bcmf_core_upload_firmware: upload firmware
sdio_io_rw_extended: prep read 4
sdio_io_rw_extended: Transaction ends
sdio_io_rw_extended: prep write 4 1
sdio_io_rw_extended: Transaction ends
sdio_io_rw_extended: prep read 4
sdio_io_rw_extended: Transaction ends
sdio_io_rw_extended: prep read 4
sdio_io_rw_extended: Transaction ends
sdio_io_rw_extended: prep write 4 1
sdio_io_rw_extended: Transaction ends
sdio_io_rw_extended: prep read 4
sdio_io_rw_extended: Transaction ends
sdio_io_rw_extended: prep write 4 1
sdio_io_rw_extended: Transaction ends
sdio_io_rw_extended: prep read 4
sdio_io_rw_extended: Transaction ends
sdio_io_rw_extended: prep write 4 1
sdio_io_rw_extended: Transaction ends
sdio_io_rw_extended: prep read 4
sdio_io_rw_extended: Transaction ends
bcmf_core_upload_firmware: firmware size is 210412 bytes
sdio_io_rw_extended: prep write 64 256
sdio_io_rw_extended: Transaction ends
sdio_io_rw_extended: prep write 64 256
sdio_io_rw_extended: Transaction ends
sdio_io_rw_extended: prep write 64 256
sdio_io_rw_extended: Transaction ends
sdio_io_rw_extended: prep write 64 256
sdio_io_rw_extended: Transaction ends
sdio_io_rw_extended: prep write 64 256
sdio_io_rw_extended: Transaction ends
sdio_io_rw_extended: prep write 64 256
sdio_io_rw_extended: Transaction ends
sdio_io_rw_extended: prep write 64 256
sdio_io_rw_extended: Transaction ends
sdio_io_rw_extended: prep write 64 256
sdio_io_rw_extended: Transaction ends
sdio_io_rw_extended: prep write 64 256
sdio_io_rw_extended: Transaction ends
sdio_io_rw_extended: prep write 64 256
sdio_io_rw_extended: Transaction ends
sdio_io_rw_extended: prep write 64 256
sdio_io_rw_extended: Transaction ends
sdio_io_rw_extended: prep write 64 256
sdio_io_rw_extended: Transaction ends
sdio_io_rw_extended: prep write 64 216
sdio_io_rw_extended: Transaction ends
bcmf_core_upload_firmware: upload nvram configuration
bcmf_upload_nvram: nvram size is 1024 bytes
sdio_io_rw_extended: prep write 64 16
sdio_io_rw_extended: Transaction ends
sdio_io_rw_extended: prep write 4 1
sdio_io_rw_extended: Transaction ends
sdio_io_rw_extended: prep read 4
sdio_io_rw_extended: Transaction ends
sdio_io_rw_extended: prep write 4 1
sdio_io_rw_extended: Transaction ends
sdio_io_rw_extended: prep read 4
sdio_io_rw_extended: Transaction ends
sdio_io_rw_extended: prep write 4 1
sdio_io_rw_extended: Transaction ends
sdio_io_rw_extended: prep read 4
sdio_io_rw_extended: Transaction ends
sdio_io_rw_extended: prep write 4 1
sdio_io_rw_extended: Transaction ends
sdio_io_rw_extended: prep read 4
sdio_io_rw_extended: Transaction ends
sdio_io_rw_extended: prep read 4
sdio_io_rw_extended: Transaction ends
sdio_io_rw_extended: prep read 4
sdio_io_rw_extended: Transaction ends
sdio_enable_function: Function 2 enabled
sdio_io_rw_extended: prep write 4 1
sdio_io_rw_extended: prep read 4
sdio_io_rw_extended: Transaction ends
sdio_io_rw_extended: prep write 4 1
sdio_io_rw_extended: Transaction ends
sdio_io_rw_extended: prep read 4
sdio_io_rw_extended: Transaction ends
sdio_io_rw_extended: error 1
sdio_io_rw_extended: prep read 4
sdio_io_rw_extended: Transaction ends
bcmf_sdpcm_readframe: len: 12 Header checksum: 0xfff3
bcmf_sdpcm_readframe: Flow control
sdio_io_rw_extended: prep read 8
sdio_io_rw_extended: Transaction ends
sdio_io_rw_extended: prep read 4
sdio_io_rw_extended: Transaction ends
bcmf_sdpcm_readframe: No data
sdio_io_rw_extended: prep read 4
sdio_io_rw_extended: Transaction ends
sdio_io_rw_extended: prep write 4 1
sdio_io_rw_extended: Transaction ends
bcmf_wl_active: Interface activated
bcmf_cdc_control_request_unsafe: >>> Sending control 0 1 0x00000107 [4] set bus:txglom 
bcmf_cdc_control_request_unsafe: Error while waiting for control response -110
bcmf_ifup: bcmf_ifup done: -110
nsh_telnetstart: Starting the Telnet daemon
telnetd_daemon: Accepting connections on port 23
telnetd_daemon: ERROR: accept failed: 9
```
Please advise on other debugs I can generate to help.  I'm at a loss.  I've tried NuttX 11, 12, and git without success.  Thanks!

acassis, 2023-03-19:
Hi @TheCranston thank you for reporting! I think we don't have users testing it or we don't have users with experience to help to fix it.
So I decided to test it and got same issue as you in our current versions, but I got NuttX 9.0 working, see attached file.
I'm also including my board config for reference.

Could you please help us to git bisect and find each patch broken the WiFi support on Photon board?
[image]
[config_nuttx_photo_9.0.txt](https://github.com/apache/nuttx/files/11012238/config_nuttx_photo_9.0.txt)

acassis, 2023-03-19:
I found other duplicated issues about this Photon not connecting. I closed them and pointed all to this issue here, let to use this issue and reference and find an way to fix the issue.

@anchao @a-lunev https://github.com/apache/nuttx/issues/3663 was not fixed or it returned!

TheCranston, 2023-03-19:
Alan,I’m happy to help.  Let me give your board configuration a try on 9.0.  I’ll need a bit of help understanding the bisects you are requesting, but am eager to help figure it out.-BenSent from my iPhone

acassis, 2023-03-20:
Thank you @TheCranston there is a quick introduction to git bisect here: https://www.youtube.com/watch?v=z-AkSXDqodc

In NuttX case the git bisect is a little bit more trickier because we have two separated repositories to bisect (nuttx and nuttx-apps). These two repositories normally "move" in sync, so the git bisect done in a repository needs to be replicated in the other repository.

Maybe it could be possible to fuse/combine both repositories into a single one, but until now nobody did it this way.

TheCranston, 2023-03-20:
@acassis
I used your config (modified for my wifi) but am running into a compiler issue
```
arm-none-eabi-ar rcs  libbinfmt.a   binfmt_globals.o binfmt_initialize.o binfmt_register.o binfmt_unregister.o binfmt_loadmodule.o binfmt_unloadmodule.o binfmt_execmodule.o binfmt_exec.o binfmt_copyargv.o binfmt_dumpmodule.o builtin.o
make[1]: Leaving directory '/home/nonroot/src/nuttx-9.0/nuttx/binfmt'
IN: binfmt/libbinfmt.a -> staging/libbinfmt.a
make[1]: Entering directory '/home/nonroot/src/nuttx-9.0/nuttx/arch/arm/src'
make[2]: Entering directory '/home/nonroot/src/nuttx-9.0/nuttx/boards/arm/stm32/photon/src'
CC:  stm32_boot.c
CC:  stm32_bringup.c
CC:  stm32_spi.c
CC:  dfu_signature.c
CC:  stm32_appinit.c
CC:  stm32_userleds.c
CC:  stm32_wdt.c
CC:  stm32_wlan.c
CC:  stm32_wlan_firmware.c
arm-none-eabi-ar rcs  libboard.a   stm32_boot.o stm32_bringup.o stm32_spi.o dfu_signature.o stm32_appinit.o stm32_userleds.o stm32_wdt.o stm32_wlan.o stm32_wlan_firmware.o
make[2]: Leaving directory '/home/nonroot/src/nuttx-9.0/nuttx/boards/arm/stm32/photon/src'
LD: nuttx
arm-none-eabi-ld: Error: unable to disambiguate: -nostartfiles (did you mean --nostartfiles ?)
make[1]: *** [Makefile:185: nuttx] Error 1
make[1]: Leaving directory '/home/nonroot/src/nuttx-9.0/nuttx/arch/arm/src'
make: *** [tools/Makefile.unix:449: pass2] Error 2
```

perhaps its a version issue?  What are you running?  I'm on ubuntu 22.10

```
$ arm-none-eabi-ar -V
GNU ar (2.39-1ubuntu1+17ubuntu1) 2.39
Copyright (C) 2022 Free Software Foundation, Inc.
This program is free software; you may redistribute it under the terms of
the GNU General Public License version 3 or (at your option) any later version.
This program has absolutely no warranty.
```

TheCranston, 2023-03-20:
Nice video.  working thru it now. thanks.

> https://www.youtube.com/watch?v=z-AkSXDqodc

TheCranston, 2023-03-20:
git bisect log.. not sure if this helps

```
$ git bisect log
git bisect start
# status: waiting for both good and bad commits
# good: [725bdfb0e8c704823669d20931fd0a824c462212] tools/sethost.sh:  Correct error in setting a different host.
git bisect good 725bdfb0e8c704823669d20931fd0a824c462212
# status: waiting for bad commit, 1 good commit known
# bad: [561cbbca96a749e4d2e9ec2ea672f257dc5348aa] tcp: Remove incomplete support for TCP reassembly
git bisect bad 561cbbca96a749e4d2e9ec2ea672f257dc5348aa
# good: [3c829226f9f2f5495b22d3f7e239760ac54c589a] Fix typos and nxstyle complaints.
git bisect good 3c829226f9f2f5495b22d3f7e239760ac54c589a
# good: [b94a7698d917161dc85c23984ca09a4b6fe56e9f] Revert "CI: use 'git diff $commits' as a whole patchset to do checkpatch"
git bisect good b94a7698d917161dc85c23984ca09a4b6fe56e9f
# good: [fefd893b653a43dff56fc44961a8761478df2ba5] fs/nfs/nfs_util.c: Fix compilation error
git bisect good fefd893b653a43dff56fc44961a8761478df2ba5
# good: [ac84a5177d6ba40dc0ca3adbb1c7a14bb0022d33] libc/time: call _NX_OPEN/_NX_CLOSE instead of open/close
git bisect good ac84a5177d6ba40dc0ca3adbb1c7a14bb0022d33
# good: [24262a4ddbccbd33d9020f6e783c94b2ce43cc8c] Fix nxstyle issue
git bisect good 24262a4ddbccbd33d9020f6e783c94b2ce43cc8c
# good: [c3ba6031349cad8a86c69abbe4b3438bbfd07e7d] boards/arm/stm32/olimex-stm32-p407: Add joystick support.
git bisect good c3ba6031349cad8a86c69abbe4b3438bbfd07e7d
# good: [145205767dc165e133c3d6354234da64fe99fcc5] tools/zipme.sh: Add executable bit
git bisect good 145205767dc165e133c3d6354234da64fe99fcc5
# good: [3b1dca8167d8abc22a7e0c71759bfc560c4e71c5] mm: Do not memcopy more than oldsize when realloc
git bisect good 3b1dca8167d8abc22a7e0c71759bfc560c4e71c5
# good: [569477d36c44e27609ed17d3383d10474b46ce30] board/sim: Don't ignore any files under src/etc/ folder
git bisect good 569477d36c44e27609ed17d3383d10474b46ce30
# good: [b34096a4fa568c7a13e0c5d4f251ee79e8a2e454] net/tcp/tcp_input.c:  Correct bad check of urgent data length
git bisect good b34096a4fa568c7a13e0c5d4f251ee79e8a2e454
# first bad commit: [561cbbca96a749e4d2e9ec2ea672f257dc5348aa] tcp: Remove incomplete support for TCP reassembly
```

TheCranston, 2023-03-20:
Ok.  much bisecting later.... I believe, if I'm doing this correctly, I have the commit in question.

```
fa6ea23101bf9470253c2f60ca422027abdef537 is the first bad commit
commit fa6ea23101bf9470253c2f60ca422027abdef537
Author: chao.an <anchao@xiaomi.com>
Date:   Sun Jun 19 04:43:36 2022 +0800

    wireless/bcm43xxx: enable power saving on netdev up/down
    
    Move sdio/firmware de/initialize to ifup/down
    
    Signed-off-by: chao.an <anchao@xiaomi.com>

 drivers/wireless/ieee80211/bcm43xxx/bcmf_driver.c |  61 ++++++----
 drivers/wireless/ieee80211/bcm43xxx/bcmf_driver.h |   2 +
 drivers/wireless/ieee80211/bcm43xxx/bcmf_netdev.c | 133 +++++++++++++---------
 drivers/wireless/ieee80211/bcm43xxx/bcmf_sdio.c   | 126 +++++++++++---------
 drivers/wireless/ieee80211/bcm43xxx/bcmf_sdio.h   |   2 +
 5 files changed, 193 insertions(+), 131 deletions(-)
```
I think it is correct as when the WiFi fails I get this message on startup
```
ioctl(SIOCSIWMODE): 1
```

I can get 10.3 to work properly and 11.0 does not.  This lands between them.  Hopefully this helps the team out.

anchao, 2023-03-21:
It seems that the sdio driver has potential race condition, I submitted a PR to solve this issue, please review:
https://github.com/apache/nuttx/pull/8865

acassis, 2023-03-21:
Amazing! Thank you @TheCranston and @anchao !!!

acassis, 2023-03-21:
@anchao it worked, but only if I run:
```
nsh> wapi mode wlan0 2
```

Why Infrastructure mode is not default to WiFi on this chip?

acassis, 2023-03-21:
I didn't close it, github did it automatically when I merged the PR #8865

TheCranston, 2023-03-21:
@anchao @acassis It would appear the PR addresses the issue.  I'm doing some extended testing right now, but things look good.  Thanks!

TheCranston, 2023-03-21:
Welp, not 100% and perhaps another issue here:
After ~5 minutes of an extended ping I start getting

```
[ 4] sdio_io_rw_extended: timeout
[ 4] bcmf_sdpcm_sendframe: No credit to send frame
[ 1] bcmf_netdev_alloc_tx_frame: ERROR: Cannot allocate TX frame
```
repeated over and over and the ping fails on the timeout 

I did get an
```
ABCDF
[ 2] sdio_io_rw_extended: error 1
telnetd [5:100]

NuttShell (NSH) NuttX-12.0.0
nsh> 
```
Not sure what that sdio_rw error during boot up is related to...

TheCranston, 2023-03-22:
@anchao Thank you for the code changes.  I've applied them locally and it appears to have helped
```
ABCDF
[    1.190000] [ 2] sdio_io_rw_extended: error 1
telnetd [5:100]

NuttShell (NSH) NuttX-12.0.0
nsh> iperf -c 192.168.253.166
       IP: 192.168.253.17

 mode=tcp-client sip=192.168.253.17:5001,dip=192.168.253.166:5001, interval=3, time=30

           Interval         Transfer         Bandwidth

*float*-*float* sec     229376 Bytes *float* Mbits/sec
*float*-*float* sec     212992 Bytes *float* Mbits/sec
*float*-*float* sec     229376 Bytes *float* Mbits/sec
*float*-*float* sec     245760 Bytes *float* Mbits/sec
*float*-*float* sec     229376 Bytes *float* Mbits/sec
*float*-*float* sec     229376 Bytes *float* Mbits/sec
*float*-*float* sec     245760 Bytes *float* Mbits/sec
*float*-*float* sec     229376 Bytes *float* Mbits/sec
*float*-*float* sec     229376 Bytes *float* Mbits/sec
*float*-*float* sec     245760 Bytes *float* Mbits/sec
*float*-*float* sec    2326528 Bytes *float* Mbits/sec
iperf exit
nsh> 
```

The iperf server reported
```
$ iperf -s
------------------------------------------------------------
Server listening on TCP port 5001
TCP window size:  128 KByte (default)
------------------------------------------------------------
[  1] local 192.168.253.166 port 5001 connected with 192.168.253.17 port 4823 (icwnd/mss/irtt=12/1250/14035)
[ ID] Interval       Transfer     Bandwidth
[  1] 0.0000-30.5803 sec  2.23 MBytes   613 Kbits/sec
```
613Kb/s is not all that speedy, but at least it ran error free.  And I'll note, I'm not fully sure what a good metric is for this chip.
Thanks again, let me know what other debugs/tests I can perform.

That sdio_io_rw_extended error upon boot persists.  I also discovered that the wlan0 interface appears to "go to sleep" at some point of inactivity and does not wake back up:

```
ABCDF
[    1.190000] [ 2] sdio_io_rw_extended: error 1
telnetd [5:100]

NuttShell (NSH) NuttX-12.0.0
nsh> wapi show wlan0
wlan0 Configuration:
       IP: 192.168.253.17
  NetMask: 255.255.255.0
Frequency: *float*
     Flag: WAPI_FREQ_AUTO
  Channel: 1
Frequency: *float*
    ESSID: DaNet
     Flag: WAPI_ESSID_ON
     Mode: WAPI_MODE_MANAGED
       AP: 24:81:3b:b6:10:e0
  BitRate: 65000
     Flag: WAPI_BITRATE_FIXED
  TxPower: 31
     Flag: WAPI_TXPOWER_DBM
    Sense: -58
  Country: US
[   12.970000] [ 6] bcmf_ioctl: ERROR: Unrecognized IOCTL command: 8b3a
[   12.970000] [ 6] wapi_get_pta_prio: ioctl(SIOCGIWPTAPRIO): 25
nsh> ping 10.255.254.151
PING 10.255.254.151 56 bytes of data
No response from 10.255.254.151: icmp_seq=0 time=1000 ms
56 bytes from 10.255.254.151: icmp_seq=1 time=40.0 ms
56 bytes from 10.255.254.151: icmp_seq=2 time=30.0 ms
56 bytes from 10.255.254.151: icmp_seq=3 time=30.0 ms
56 bytes from 10.255.254.151: icmp_seq=4 time=30.0 ms
56 bytes from 10.255.254.151: icmp_seq=5 time=30.0 ms
56 bytes from 10.255.254.151: icmp_seq=6 time=30.0 ms
56 bytes from 10.255.254.151: icmp_seq=7 time=30.0 ms
56 bytes from 10.255.254.151: icmp_seq=8 time=30.0 ms
56 bytes from 10.255.254.151: icmp_seq=9 time=30.0 ms
10 packets transmitted, 9 received, 10% packet loss, time 10100 ms
rtt min/avg/max/mdev = 30.000/31.111/40.000/3.143 ms
nsh> wapi show wlan0
wlan0 Configuration:
       IP: 192.168.253.17
  NetMask: 255.255.255.0
Frequency: *float*
     Flag: WAPI_FREQ_AUTO
  Channel: 1
Frequency: *float*
    ESSID: DaNet
     Flag: WAPI_ESSID_ON
     Mode: WAPI_MODE_MANAGED
       AP: 00:00:00:00:00:00
  BitRate: 65000
     Flag: WAPI_BITRATE_FIXED
  TxPower: 31
     Flag: WAPI_TXPOWER_DBM
    Sense: 0
  Country: US
[  674.680000] [14] bcmf_ioctl: ERROR: Unrecognized IOCTL command: 8b3a
[  674.680000] [14] wapi_get_pta_prio: ioctl(SIOCGIWPTAPRIO): 25
nsh> ping 10.255.254.151
PING 10.255.254.151 56 bytes of data
No response from 10.255.254.151: icmp_seq=0 time=1000 ms
No response from 10.255.254.151: icmp_seq=1 time=1000 ms
No response from 10.255.254.151: icmp_seq=2 time=1000 ms
No response from 10.255.254.151: icmp_seq=3 time=1000 ms
No response from 10.255.254.151: icmp_seq=4 time=1000 ms
No response from 10.255.254.151: icmp_seq=5 time=1000 ms
No response from 10.255.254.151: icmp_seq=6 time=1000 ms
No response from 10.255.254.151: icmp_seq=7 time=1000 ms
No response from 10.255.254.151: icmp_seq=8 time=1000 ms
No response from 10.255.254.151: icmp_seq=9 time=1000 ms
10 packets transmitted, 0 received, 100% packet loss, time 10100 ms
nsh> 
```
The two pings were separated by ~10mins of idle time.  Not sure what's up there, but it's been consistently loosing network ability after a period of time.  wapi shows that the "SENSE" value goes from -58 to 0.

anchao, 2023-03-22:
@TheCranston 

>> That sdio_io_rw_extended error upon boot persists.
>> [ 1.190000] [ 2] sdio_io_rw_extended: error 1

This issue exists in nuttx-9.0, but it does not has any side effect for initialization of bcmf driver,
This sdio command will only succeed on the second retry, which may be related to the implementation of firmware.

>> *float*-*float* sec 229376 Bytes *float* Mbits/sec

about word "*float*", you can enable floating point printing to get more detailed data:
CONFIG_LIBC_FLOATINGPOINT

>> 613Kb/s is not all that speedy.

Yep, the throughput results I tested were even worse,

```bash
nsh> iperf -c 192.168.31.180 -i 1
        IP: 192.168.31.138

  mode=tcp-client sip=192.168.31.138:5001, dip=192.168.31.180:5001, interval=1, time=30

            Interval Transfer Bandwidth

    0.00- 1.01 sec 49152 Bytes 0.39 Mbits/sec
    1.01- 2.08 sec 49152 Bytes 0.37 Mbits/sec
    2.08- 3.09 sec 49152 Bytes 0.39 Mbits/sec
    3.09- 4.14 sec 49152 Bytes 0.37 Mbits/sec
    4.14- 5.19 sec 49152 Bytes 0.37 Mbits/sec
    5.19- 6.23 sec 49152 Bytes 0.38 Mbits/sec
...
    0.00- 30.09 sec 1507328 Bytes 0.40 Mbits/sec
```

I tried adjusting nvram parameters but it doesn't have any improvement:
https://github.com/apache/nuttx/blob/master/boards/arm/stm32/photon/src/stm32_wlan_firmware.c#L44-L103

>> The two pings were separated by ~10mins of idle time. Not sure what's up there, but it's been consistently loosing network ability after a period of time. wapi shows that the "SENSE" value goes from -58 to 0.

This issue also exists in nuttx-9.0, after the data communication is maintained for a certain period of time, the firmware seems enter like state of power saving, I don't have a clue yet.

acassis, 2023-04-06:
@TheCranston could you please confirm that the issue was fixed?

a-lunev, 2023-06-26:
> I found other duplicated issues about this Photon not connecting. I closed them and pointed all to this issue here, let to use this issue and reference and find an way to fix the issue.
> 
> @anchao @a-lunev #3663 was not fixed or it returned!

Hi Alan,
I am sorry, I am reading this just now. Currently there is not Photon board with me, so I can not test it.
The old #3663 issue was definitely resolved that time concerning namely Wi-Fi connection.

acassis, 2023-10-11:
@TheCranston I think I missed your last comment. Yes, maybe the driver needs to have more improvements to work after the sleep/wake-up. BTW I suggest you closing this issue here (since the WiFi is working if the chip doesn't sleep_ and open an issue related specifically to the WiFi sleep/wake-up issue.

TheCranston, 2023-10-18:
@acassis Sure thing. I'll take that direction.  thanks for the support.

closed 2023-10-18 by TheCranston