apache/nuttx #8857: use after free in `poll()` implementation

realprocrastinator, 2023-03-21:
In `fd/vfs/fs_poll.c`, the semaphore object is allocated on the stack meaning that its memory is valid until this function returns. However, there is a `poll_default_cb()` which may also access this semaphore object after this function returns. For example, when a `SIGINT` signal interrupts the `poll` function, and during the task exiting, `poll_notify` will be eventually invoked and internally invokes `poll_default_cb()`. At this time, there is a chance that the `poll` function has already returned which leads to invalid memory access to the semaphore object.
```
 int poll(FAR struct pollfd *fds, nfds_t nfds, int timeout)
 {
   FAR struct pollfd *kfds;
   sem_t sem;
   ...
 }
```
```
void poll_default_cb(FAR struct pollfd *fds)
{
  int semcount = 0;
  FAR sem_t *pollsem;

  if (fds->arg != NULL)
    {
      pollsem = (FAR sem_t *)fds->arg;
      nxsem_get_value(pollsem, &semcount);
      if (semcount < 1)
        {
          nxsem_post(pollsem);
        }
    }
}
```

acassis, 2023-03-24:
@realprocrastinator good catch! Do you think it is possible to create a testing application to demonstrate this issue?

realprocrastinator, 2023-03-25:
> @realprocrastinator good catch! Do you think it is possible to create a testing application to demonstrate this issue?

Sure, I will have a try! :-)