apache/nuttx #8858: Issues with adjtime implementation.

patacongo, 2023-03-21:
adjtime() is implemented in sched/clock/clock_timekeeping.  A quick review reveals several problems:

1. Missing system call.  The implemented of adjtime() is provided, but there is no call gate implemented in syscall/ and include/sys

3. Adjustment has limited range.  The logic calculates a separate adjustment that is added to the current time (a timespec).  The  adjustment value is a type long with an LSB of 1 nonosecond.  So the total range is only +/-2 seconds.  That is an insufficient range for systems that are always on.

This is because adjtime does not follow the man page:  "If the adjustment in delta is positive, then the system clock is  speeded up by some small percentage (i.e., by adding a small  amount of time to the clock value in each second) until the  adjustment has been completed.  If the adjustment in delta is negative, then the clock is slowed down in a similar fashion."

If small adjustments were add to a full timespec, overflow would not occur.

Hmmm..  There is no such word as speeded up, is there?  Isn't the past tense of speed sped, not speeded.

3. No guarantee of monotonically increasing time.  The requirement is:  "The adjustment that adjtime() makes to the clock is carried out   in such a manner that the clock is always monotonically  increasing."  But I see nothing that guarantees this.

cederom, 2023-03-21:
Here are some references :-)

https://github.com/freebsd/freebsd-src/blob/master/contrib/ntp/libntp/adjtime.c

https://github.com/freebsd/freebsd-src/blob/master/contrib/ntp/include/ntp_syscall.h

https://github.com/freebsd/freebsd-src/blob/main/sys/kern/kern_time.c

And the man page :-)

https://man.freebsd.org/cgi/man.cgi?query=adjtime&sektion=2&apropos=0&manpath=FreeBSD+13.1-RELEASE+and+Ports

     The adjtime() system call makes small adjustments to the system time, as
     returned by gettimeofday(2), advancing or retarding it by the time    speci-
     fied by the timeval delta.     If delta is negative, the clock is slowed
     down by incrementing it more slowly than normal until the correction is
     complete.    If delta is positive, a    larger increment than normal is    used.
     The skew used to perform the correction is    generally a fraction of    one
     percent.  Thus, the time is always    a monotonically    increasing function.
     A time correction from an earlier call to adjtime() may not be finished
     when adjtime() is called again.  If olddelta is not a null    pointer, the
     structure pointed to will contain,    upon return, the number    of microsec-
     onds still    to be corrected    from the earlier call.

     This call may be used by time servers that    synchronize the    clocks of com-
     puters in a local area network.  Such time    servers    would slow down    the
     clocks of some machines and speed up the clocks of    others to bring    them
     to    the average network time.

     The adjtime() system call is restricted to    the super-user.

Hope that helps :-)

michallenc, 2023-03-22:
Thank you Greg @patacongo  for detailed issue description. I agree clock rate adjustment would be better solution and would solve possible problems with time jumps. This is also how time synchronization is handled in RTEMS implementation of adjtime() function.

I will take a look into possible implementation to NuttX mainline.

patacongo, 2023-03-22:
> I will take a look into possible implementation to NuttX mainline.

There are lots of implementations of algorithms out there, so fixing the NuttX adjtime should not be too difficult (although it doesn't even compiler now if you enable TIMEKEEPING).

It will require special architecture-specific hooks to modify the clock rate, but these can be conditioned on CONFIG_HAVE_CLOCK_TIMEKEEPING as the logic is now, so it does not require modification of all architectures.

michallenc, 2023-04-03:
> adjtime() does depend on the architecture-specific code providing interfaces to adjust the clock rate, so perhaps it would need to depend on CONFIG_ARCH_HAVE_TIMEKEEPING

@patacongo That is the current state. I just changed it that adjtime() would depend on CONFIG_ARCH_HAVE_ADJTIME and CONFIG_CLOCK_ADJTIME. Mostly because current timekeeping implementation does not use traditional basetime structure

```
#ifndef CONFIG_CLOCK_TIMEKEEPING
struct timespec   g_basetime;
#endif
```

As can be seen here https://github.com/apache/nuttx/blob/master/sched/clock/clock_initialize.c#L58. Therefore there is a different aproach for settime and gettime. I think maybe we should keep the implementation of adjtime (under CONFIG_CLOCK_ADJTIME) and other timekeeping function (wall time etc. under CONFIG_CLOCK_TIMEKEEPING) separately. This way it will be possible to use adjtime just for clock period adjustment when required by the user.

closed 2023-04-25 by acassis