apache/nuttx #8869: Useless/incorrect stuff in signal.h should be removed?

patacongo, 2023-03-21:
With recent changes, the maximum signal number has been changed to OPEN_MAX, aka, FD_SETSIZE (actually FD_SETSIZE-1 since zero is not a signal number).  But there are still references to the maximal signal number being 31.  There is not such limit, correct?

As examples, in signal.h, are these not incorrect:

     38 /* Signal set management definitions and macros. */
     39
     40 #define NULL_SIGNAL_SET ((sigset_t)0x00000000)
     41 #define ALL_SIGNAL_SET  ((sigset_t)0xffffffff)
     42 #define MIN_SIGNO       1
     43 #define MAX_SIGNO       31
     44 #define GOOD_SIGNO(s)   ((((unsigned)(s)) <= MAX_SIGNO))
     45 #define SIGNO2SET(s)    ((sigset_t)1 << (s))
     46
     47 /* All signals are "real time" signals */
     48
     49 #define SIGRTMIN        MIN_SIGNO       /* First real time signal */
     50 #define SIGRTMAX        MAX_SIGNO       /* Last real time signal */
     51 #define _NSIG           (MAX_SIGNO + 1) /* Biggest signal number + 1 */

There are a few references to these in the code.  Are these not wrong too?

    $ find . -name *.c | xargs grep -l MAX_SIGNO
    ./fs/vfs/fs_signalfd.c
    ./sched/signal/sig_default.c
    ./sched/signal/sig_lowest.c

    $ find . -name *.c | xargs grep -l GOOD_SIGNO
    ./libs/libc/signal/sig_addset.c
    ./libs/libc/signal/sig_delset.c
    ./libs/libc/signal/sig_ismember.c
    ./libs/libc/signal/sig_set.c
    ./libs/libc/signal/sig_signal.c
    ./libs/libc/string/lib_strsignal.c
    ./sched/mqueue/mq_notify.c
    ./sched/signal/sig_action.c
    ./sched/signal/sig_default.c
    ./sched/signal/sig_kill.c
    ./sched/signal/sig_queue.c
    ./sched/signal/sig_tgkill.c
    ./sched/signal/sig_timedwait.c

    $ find . -name *.c | xargs grep -l ALL_SIGNAL_SET
    ./libs/libc/signal/sig_fillset.c
    ./sched/pthread/pthread_exit.c
    ./sched/signal/sig_cleanup.c

Since the signal set is an array, the definition NULL_SIGNAL_SET is also not meaningful.

patacongo, 2023-03-21:
The motivation for this is to provide more real-time signals (i.e., those with no pre-defined meaning or actions).  If all signal definitions are enabled, then that would be 29 signals, leaving only 2 real-time sizes.  Linux increased the number of signals from 32 to 64 just in order to support real-time signals.  Should we do the same?

pkarashchenko, 2023-03-21:
If we can think of getting rid of `SIGWORK` and reusing `SIGCONT` that can free up 1 value

patacongo, 2023-03-21:
I see an error in include/signal.h.  This really should be a separate Issue, but I will include it here since it applies to the same section of code:

The real time signals are all sequentially numbered starting at SIGRTMIN and going through SIGRTMAX.  But the standard signals are also in the range of SIGRTMIN through  SIGRTMAX.  It should not be

    #define SIGRTMIN        MIN_SIGNO       /* First real time signal */
    #define SIGRTMAX        MAX_SIGNO       /* Last real time signal */

It should be:

    #define SIGRTMIN        (SIGIO + 1)     /* First real time signal */
    #define SIGRTMAX        MAX_SIGNO       /* Last real time signal */

Per POSIX (https://pubs.opengroup.org/onlinepubs/7908799/xsh/signal.h.html):

> This header also declares the macros SIGRTMIN and SIGRTMAX, which evaluate to integral expressions and, if the Realtime Signals Extension option is supported, specify a range of signal numbers that are reserved for application use and for which the realtime signal behaviour specified in this specification is supported. The signal numbers in this range do not overlap any of the signals specified in the following table.
> 
> The range SIGRTMIN through SIGRTMAX inclusive includes at least RTSIG_MAX signal numbers. 
>

patacongo, 2023-03-21:
> If we can think of getting rid of `SIGWORK` and reusing `SIGCONT` that can free up 1 value

SIGWORK is a real time signal since it is simply used an IPC and has no default action.  It is used to kick off work in worker thread when new work is added.  It can be moved elsewhere.  It does not have to have global scope because it has meaning only to the worker thread task and associated signaling logic.

patacongo, 2023-03-21:
> If we can think of getting rid of `SIGWORK` and reusing `SIGCONT` that can free up 1 value

With regard to SIGCONT.  I don't understand your argument.  All real time signals must lie in the consecutive range of SIGRTMIN through SIGRTMAX per POSIX requirement.  That is so that the real time signal number can be assigned in a portable way.  I don't think that there is any way to re-use "holes" in the lower signal numbers for real time signals.

pkarashchenko, 2023-03-21:
As far as I see POSIX does not specify the signal values, but only a list of aliases for standard signals. Following Linux values and keeping compliance to POSIX will make implementation of realtime signals almost impossible (or actually possible with for 1 signal value `SIGRTMIN == SIGRTMAX`). Again I'm talking about current implementation that has 32 bit sigset `typedef uint32_t sigset_t;`.
My main goal is to understand how many realtime signals we would like to support and then define a way to achieve it. For example if we re-arrange standard signals values to be monotonically incremental values and get rid of the holes at values 16 and 28 plus extending `MAX_SIGNO` to 32 by `#define SIGNO2SET(s)    ((sigset_t)1 << (s))` -> `#define SIGNO2SET(s)    ((sigset_t)1 << ((s) - 1))` we can have at least 3 values for realtime signals.
From your explanation I understand that we can simply define `SIGWORK` to `SIGRTMIN` as it is an internal signal, so no need to expose it via `signal.h`.
So the main question is: do we need more than 3 realtime signals? If yes, then we should think of a bigger rework than just shuffling things around and can definitely follow Linux values for signal numbers.

patacongo, 2023-03-21:
Prior to e6227e19433c4999d500437d0a8a5c05f476ceb1, here were only 19 standard signals defined by default (with 21 defined with options enabled).  Then there were 11 real time signals.  Now there are 29 standard signal definitions with only 2 available for true real time signals.  Although it erroneously claims to have 31 real time signals.

patacongo, 2023-03-21:
> For example if we re-arrange standard signals values to be monotonically incremental values and get rid of the holes at values 16 and 28 plus extending `MAX_SIGNO` to 32 by `#define SIGNO2SET(s) [...]

Commit e6227e19433c4999d500437d0a8a5c05f476ceb1 made the NuttX signal numbers identical in value to the Linux signal numbers.  So I am assuming that some depends on this equivalence.  Perhaps Xiaomi?  Perhaps for the simulator.  Xiaomi made similar changes in the past to make error number and O_FLAGS the same as Linux to simplify the simulator.  Changing them now might break something???

Most of those standard signal numbers could be removed.  They are not even used in NuttX.  So I think you would have to first determine who cares about the exact values and why.

patacongo, 2023-03-21:
> From your explanation I understand that we can simply define `SIGWORK` to `SIGRTMIN` as it is an internal signal, so no need to expose it via `signal.h`.

I believe that is true, but it wouldn't hurt to grep and make sure that nothing outside of the work queue and the work queue interface uses it.

patacongo, 2023-03-21:
> So the main question is: do we need more than 3 realtime signals? If yes, then we should think of a bigger rework than just shuffling things around and can definitely follow Linux values for signal numbers.

Obviously I think we do.  Three is a very small number.  Many tasks can live with three but perhaps others cannot.  It is hard to know for certain and the kinds of failures that you would see if there are duplicate signal numbers could be difficult to debug.

pkarashchenko, 2023-03-21:
Yeah, 19 standard signals defined is a good from real use case perspective, but not good from POSIX compliance perspective that defines 28 standard signals. The POSIX compliance is stated as a top priority for NuttX, so we need to have all 28 standard signals even if 9 of those are ever going to be issued. The `SIGCONDTIMEDOUT` can be safely removed as no code is referring to it anymore. The `SIGWORK` can be any of realtime signals as it is internal signal and we can move its definition to `sched/paging/paging.h` and define to `SIGRTMAX` for example. So as I described above with current implementation and simple re-ordering of the signal values we can have `31 - 29 = 2` signals and by rising a `MAX_SIGNO` to `32` the `3` realtime signals without going to `sigset_t` expansion.

pkarashchenko, 2023-03-21:
While I was writing my single comment the 3 more are already issued, so sorry for duplicating parts that are already answered. I need to refresh the page before pressing the "Comment" button

patacongo, 2023-03-21:
MAX_SIGNO is zero based.  So for the 32-bit sigset,_t signal 31 == bit 31 is the maximum signal number.  Currently only 29 and 30 are available. 

I don't have any concerns about memory usage or performance loss by extending sigset_t.  It is not used represent a signal (that is just an integer).  It is used only for passing parameters at POSIX iinterfaces and only a few other places (like sigprocmask).  Manipulations on signal sets is all done through POSIX function calls so that would not change -- unless the POSIX functions are not used.  The implementation of set operations would be a little bit larger.

pkarashchenko, 2023-03-21:
> MAX_SIGNO is zero based. So for the 32-bit sigset,_t signal 31 == bit 31 is the maximum signal number. Currently only 30 and 31 are available.
> 
> I don't have any concerns about memory usage or performance loss by extending sigset_t. It is not used to represent a signal (that is just an integer). It is used only for passing parameters at POSIX iinterfaces and only a few other places (like sigprocmask). Manipulations on signal sets is all done through POSIX function calls so that would not change -- unless the POSIX functions are not used. The implementation of set operations would be a little bit larger.

Yes `MAX_SIGNO` is zero based, but bit 0 in `sigset_t` is always 0, so 32-bit sigset could actually address 32 signals and not 31.

pkarashchenko, 2023-03-21:
Regarding the `O_FLAGS` changes to be the same as Linux. Is that about `O_RDONLY`, `O_WRONLY` and others? If yes, then I think those are still different as `#define O_RDONLY    (1 << 0)` in NuttX, but `O_RDONLY` is `0` in Linux

patacongo, 2023-03-21:
> Yes `MAX_SIGNO` is zero based, but bit 0 in `sigset_t` is always 0, so 32-bit sigset could actually address 32 signals and not 31.

Signal 0 has a special function.  It is the do-nothing signal.  It is used with kill to test if the task is still alive.  But it does not appear in the sigset_t AFAIK.  Recovering that single bit would probably be more effort than it is work, however.

patacongo, 2023-03-21:
> Regarding the `O_FLAGS` changes to be the same as Linux. Is that about `O_RDONLY`, `O_WRONLY` and others? If yes, then I think those are still different as `#define O_RDONLY (1 << 0)` in NuttX, but `O_RDONLY` is `0` in Linux

Probably.  I don't know what all Xiaomi changed.  The point is that they did change some things to agree with arbitrary Linux definitions.

pkarashchenko, 2023-03-21:
We already have
```
#ifdef CONFIG_SYSTEM_TIME64
typedef uint64_t  time_t;         /* Holds time in seconds */
#else
typedef uint32_t  time_t;         /* Holds time in seconds */
#endif
```
so I was thinking about use cases for systems that do not have `CONFIG_HAVE_LONG_LONG` (about if 4 realtime signals would be enough). Using `uint64_t` for sigset will hide a lot of work behind the scene (done by the compiler) and modification to existing code would be minimal plus compiler can insert assembler if ISA can deal natively with 64 bit values, so I was considering this only from such perspective. Going with `uint32_t [2]` array is obviously the most portable solution.

pkarashchenko, 2023-03-21:
@xiaoxiang781216 what is the background for https://github.com/apache/nuttx/pull/8623 so we need to have values in sync with Linux?

patacongo, 2023-03-21:
> so I was thinking about use cases for systems that do not have `CONFIG_HAVE_LONG_LONG` (about if 4 realtime signals would be enough). Using `uint64_t` for sigset will hide a lot of work behind the [...]

That is reasonable.  There is no one using those retro platforms that do not support long long now.  They might in the future, however.  But my experience is that we always have to resuscitate the less often used platforms when they have not been used for awhile.

> Going with `uint32_t [2]` array is obviously the most portable solution.

I don't think this has much effect on code outside of the signal set logic.  But it would require a rather trivial modification to all signal set manipulation functions to break a signal number into an index and bit number via a macro.

This biggest impact that I know is where the avoids the standard signal set functions.  For example using ALL_SIGNAL_SET as an initialize vs. calling sigfillset();

patacongo, 2023-03-22:
I just did a quit glance through some of the code to see how real time signal numbers are being defined.  The appear:

1. to be hard-coded, and
2. to overlap the standard signals.

For example, in nuttx-apps I can do:

    $ find . -name *.c | xargs grep -l "signal[.]h" | xargs grep define | grep SIG

The only value real time signals at this moment are 30 and 31.  Those are never used.  All signals overlap the standard signal numbers and could have some really weird consequences.

The value of SIGRTMIN must be created to avoid the overlap, then all usage of real time signals must be defined something like:

    #define SIGNO_A (SIGRTMIN)
    #define SIGNO_B (SIGRTMIN + 1)

pkarashchenko, 2023-03-22:
Yes. That's something I thought about and you noted here. The FS automount signal or HW timer expiration notification via signal should use realtime signals only to make sure that signal action is queued properly and events are not lost.
But on the other hand each use case should be analyzed to see if it is agnostic to event loss and can deal with being notified only with the latest event.

patacongo, 2023-03-22:
> **Another Must-Fix Bug**. I just did a quit glance through some of the code to see how real time signal numbers are being defined. They appear: ...

Kconfig files that allow configurable signals have the same issue.

    $ find nuttx -name Kconfig | xargs grep SIG
    $ find nuttx-apps -name Kconfig | xargs grep SIG

**Real Time Signal Default Value**.  The default value for all real time signals must be in the range SIGRTMIN through SIGRTMAX.  Some are not.  For example, in nuttx-apps/examples/ajoystick/Kconfig:

    19 config EXAMPLES_AJOYSTICK_SIGNO
    20         int "Joystick signal"
    21         default 13

In that case, the joystick notification is given the same value as SIG_PIPE.

**Real Time Signal Range**.  Configurable real time signal number selections should be limited by range SIGRTMIN SIGRTMAX.

**Standard Signal Number Range.**  Standard signal numbers are also configurable in sched/Kconfig.  I am not sure why since they have very no options because there are no free signal numbers and they cannot be real time signals.  They all need to have `range 1 SIGRTMIN-1`.

pkarashchenko, 2023-03-23:
@patacongo yes, please. I started with `strsignal()` thinking that it is a low hanging fruit, but with your help the bigger inconsistency was found. I do not have anything particular planned now and actually I'm having flu right now, so my activity is near to zero these days. Please take a lead and complete everything you are willing to.

patacongo, 2023-03-23:
> Please take a lead and complete everything you are willing to.

@pkarashchenko Will do.  I have it about half finished.  I will make a Draft so that you can see what I am up to.  It will be a couple more days.

> I have one change that adds SIGSYS, but better to hold on with it since Linux value for SIGSYS is 31 and that means if we stick with Linux values approach this change should wait 63 signal number extension

Okay.  Now I understand why linux has 64 signals, but only 32 real time signals.

patacongo, 2023-03-24:
> I have one change that adds `SIGSYS`, but better to hold on with it since Linux value for `SIGSYS` is 31 and that means if we stick with Linux values approach this change should wait 63 signal number extension

Yes, it will need to be 32.  This Issue will close when #8881 closes.  There are a couple of issues that will still be open like:

- Missing logic that handles real time signal delivery priority, and
- All of the bad real time signal numbers in code and Kconfig files.

There is too much in this issue now.  When #8881 cloawa, I will create new issues for whatever remains.

xiaoxiang781216, 2023-03-31:
let's close this pr since the work is finished.

closed 2023-03-31 by xiaoxiang781216