apache/nuttx #8933: FS_LARGEFILE config disabled is causing issue with df -h output.

dlukian, 2023-03-31:
I am using the cxd56xx board. Installing 32GB SDHC card into the system, formatted FAT32 and running `df` displays correct block size, block number, and remaining blocks adding up to correct volume. Using `-h` parameter outputs incorrect _Size_ and _Available_ columns. Recent Spresense SDK update adapted Nuttx 11.0.0, and FS_LARGEFILE is now disabled by default. Manually enabling FS_LARGEFILE allows correct display of total and remaining memory with `df -h` option. Perhaps the `-h` option calculation is using incorrect parameters that have to do with the maximum file size instead of actual mounted media volume?

laoniaokkk, 2023-04-02:
When disable FS_LARGEFILE, the max size is 2^32 = 4GB. So when use the SD card > 4GB, you should enable FS_LARGEFILE.The same as USB msc device

acassis, 2023-04-05:
Good point @laoniaokkk it is not a BUG! Maybe we should select FS_LARGEFILE automatically when enabling SD/MMC driver because probably most of users will be using a SDCard bigger than 4GB.

dishilov, 2023-04-05:
@acassis @laoniaokkk I was under impression the 'df' command has nothing to do with file sizes and is used to display volume information - which it does correctly when used without '-h' option. Is the option not supposed to convert the block+block size data to human readable format only?