apache/nuttx #8959: Enabling sinfo printing and disabling CONFIG_SYSLOG_TIMESTAMP_REALTIME caused a recursive calling bug?

GooTal, 2023-04-04:
I enabled sinfo printing:
[image]

I found that sinfo called nx_vsyslog, then called clock_gettime
```
  if (OSINIT_HW_READY())
    {
#if defined(CONFIG_SYSLOG_TIMESTAMP_REALTIME)
      /* Use CLOCK_REALTIME if so configured */

      clock_gettime(CLOCK_REALTIME, &ts);

#else
      /* Prefer monotonic when enabled, as it can be synchronized to
       * RTC with clock_resynchronize.
       */

      clock_gettime(CLOCK_MONOTONIC, &ts);
#endif
```
But clock_gettime called sinfo again, which makes a recursive bug?

Here`s the backtrace from gdb:

[image]

anchao, 2023-04-28:
@GooTal already fixed by https://github.com/apache/nuttx/pull/8897

closed 2023-04-28 by anchao

GooTal, 2023-05-04:
@anchao Thanks a lot