apache/nuttx #8960: Code does not rebuild properly after 'make menuconfig'

pkarashchenko, 2023-04-04:
Initially observed during testing of https://github.com/apache/nuttx/pull/8951
Steps to reproduce:
1. `tools/configure.sh spresense:smp`
2. `make menuconfig` -> set `CONFIG_PRIORITY_INHERITANCE=y`
3. `make -j8`
4. `make menuconfig` -> set `CONFIG_SEM_PREALLOCHOLDERS=0`
5. `make -j8`
6. Run binary on the board and observe crash on system startup
7. `make clean`
8. `make -j8`
9. Run binary on the board successfully without crash

Comparing `.map` files generated after step 5 and after step 8 shows that there is a difference in code segment size for multiple sections. This mean that code was not rebuilt with a proper configuration after configuration was changed with `make menuconfig`.

davids5, 2023-04-05:
Was this closed because after changes to .config a clean build is required?

pkarashchenko, 2023-04-05:
Yes. This behaviour seems to be "as expected". Most of the code is rebuilt because of `#include <nuttx/config.h>` and `config.h` is regenerated after `make menuconfig` exits, but there are some files that do not include it directly, so stale objects are used.

davids5, 2023-04-05:
Given this, should make menuconfig run clean?

pkarashchenko, 2023-04-05:
Yeah, that sounds reasonable.

pkarashchenko, 2023-04-05:
Let me re-open an issue, so we can implement it

closed 2023-04-24 by pkarashchenko