apache/nuttx #8968: stm32_otgfsdev not working on stm32f411-minimum

ALTracer, 2023-04-05:
## Summary
I can't get USB OTG to work in device mode on WeAct BlackPill STM32F411CEU6 rev 3.1 in CDC/ACM, MSC and COMPOSITE configurations.
## Impact
Possibly all stm32f4-based boards and dwc2 IP users without VBUS_SENSE pad routed.
## Description
As I was bringing up some more configs for stm32f411-minimum, I stumbled upon non-working USBDEV. USBHOST won't work because I didn't bridge VBUS with onboard 5V pin yet, and I don't need USBHOST on this board.

I see several issues in stm32_otgfsdev driver code:
1. VBUSASEN/VBUSBSEN bits are written unconditionally. Does NOVBUSSENS actually override them in this IP core version?
```patch
diff --git a/arch/arm/src/stm32/stm32_otgfsdev.c b/arch/arm/src/stm32/stm32_otgfsdev.c
index a0a9ecb70..7bf89426c 100644
--- a/arch/arm/src/stm32/stm32_otgfsdev.c
+++ b/arch/arm/src/stm32/stm32_otgfsdev.c
@@ -5362,9 +5362,11 @@ static void stm32_hwinitialize(struct stm32_usbdev_s *priv)
    * Sense when Set
    */
 
-  regval  = (OTGFS_GCCFG_PWRDWN | OTGFS_GCCFG_VBUSASEN |
-             OTGFS_GCCFG_VBUSBSEN);
-# ifndef CONFIG_USBDEV_VBUSSENSING
+  regval  = OTGFS_GCCFG_PWRDWN;
+
+# ifdef CONFIG_USBDEV_VBUSSENSING
+  regval |= OTGFS_GCCFG_VBUSBSEN;
+# else
   regval |= OTGFS_GCCFG_NOVBUSSENS;
 # endif
 # ifdef CONFIG_STM32_OTGFS_SOFOUTPUT
 ```

2. CONFIG_STM32_USBDEV_REGDEBUG option is unreachable. I added it to Kconfig myself to achieve the following trace:

```patch
diff --git a/arch/arm/src/stm32/Kconfig b/arch/arm/src/stm32/Kconfig
index 214a31ef3..7094ea2a6 100644
--- a/arch/arm/src/stm32/Kconfig
+++ b/arch/arm/src/stm32/Kconfig
@@ -11088,6 +11088,15 @@ config STM32_USBFS_REGDEBUG
 
 endmenu
 
+menu "USB OTG Device Debug Configuration"
+
+config STM32_USBDEV_REGDEBUG
+       bool "Register-Level Debug for OTG in Device mode"
+       default n
+       depends on USBDEV && DEBUG_USB_INFO
+
+endmenu
+
 menu "OTG Configuration"
        depends on STM32_OTGFS
 
```

I input sercon, msconn, conn and disconn. The latter triggered an assert because my Gentoo Host cannot enumerate this device.

```
ern  :info  : [ +18.810180] usb 1-8: new full-speed USB device number 20 using xhci_hcd
kern  :err   : [  +0.204052] usb 1-8: device descriptor read/64, error -71
kern  :err   : [  +0.280011] usb 1-8: device descriptor read/64, error -71
kern  :info  : [  +0.281943] usb 1-8: new full-speed USB device number 21 using xhci_hcd
kern  :err   : [  +0.205060] usb 1-8: device descriptor read/64, error -71
kern  :err   : [  +0.280995] usb 1-8: device descriptor read/64, error -71
kern  :info  : [  +0.106046] usb usb1-port8: attempt power cycle
kern  :info  : [Apr 5 11:16] usb 1-8: new full-speed USB device number 22 using xhci_hcd
kern  :warn  : [  +0.026800] usb 1-8: Device not responding to setup address.
kern  :warn  : [  +0.236003] usb 1-8: Device not responding to setup address.
kern  :err   : [  +0.205205] usb 1-8: device not accepting address 22, error -71
kern  :info  : [  +0.179008] usb 1-8: new full-speed USB device number 23 using xhci_hcd
kern  :warn  : [  +0.026811] usb 1-8: Device not responding to setup address.
kern  :warn  : [  +0.237983] usb 1-8: Device not responding to setup address.
kern  :err   : [  +0.204205] usb 1-8: device not accepting address 23, error -71
kern  :err   : [  +0.004012] usb usb1-port8: unable to enumerate USB device
```

Wireshark usbmon0 traces later. STM32 DfuSe USB BootROM works.

ALTracer, 2023-04-05:
<details>
<summary> NuttX-12.0 / master serial log </summary>

```

NuttShell (NSH) NuttX-12.0.0
nsh> uname -a
NuttX 12.0.0 ca9b3271d Apr  5 2023 11:05:58 arm stm32f411-minimum
nsh> ps
  PID GROUP PRI POLICY   TYPE    NPX STATE    EVENT     SIGMASK           STACK   USED  FILLED    CPU COMMAND
    0     0   0 FIFO     Kthread N-- Ready              0000000000000000 001000 000564  56.4%  100.0% Idle Task
    1     1 100 RR       Task    --- Running            0000000000000000 002000 001204  60.2%    0.0% nsh_main
nsh> free
                   total       used       free    largest  nused  nfree
nsh> help
help usage:  help [-v] [<cmd>]

    .          cat        dmesg      memdump    mksmartfs  rmdir      truncate   
    [          cd         echo       help       mount      set        uname      
    ?          cp         env        hexdump    mv         sleep      umount     
    alias      cmp        exec       irqinfo    printf     source     unset      
    unalias    dirname    exit       kill       ps         test       uptime     
    basename   dd         false      ls         pwd        time       usleep     
    break      df         free       mkdir      rm         true       xd         

Builtin Apps:
    conn     disconn  msconn   msdis    nsh      sercon   serdis   sh       
nsh> srcon
nsh: srcon: command not found
nsh> sercon
sercon: Registering CDC/ACM serial driver
sercon: Successfully registered the CDC/ACM serial driver
nsh> msconn
mcsonn_mausbmsc_exportluns: Starting SCSI worker thread
in:usbmsc_scsi_main: Started
 usbmsc_exportluns: Waiting for the SCSI worker thread
Cusbmsc_exportluns: Signalling for the SCSI worker thread
rusbmsc_scsi_main: Waiting to be signaled
eusbmsc_scsi_main: Running
ating block drivers
mcsonn_main: Configuring with NLUNS=1
mcsonn_main: handle=0x2000db90
mcsonn_main: Bind LUN=0 to /dev/smart0
mcsonn_main: Connected
nsh> ps
  PID GROUP PRI POLICY   TYPE    NPX STATE    EVENT     SIGMASK           STACK   USED  FILLED    CPU COMMAND
    0     0   0 FIFO     Kthread N-- Ready              0000000000000000 001000 000564  56.4%   99.3% Idle Task
    1     1 100 RR       Task    --- Running            0000000000000000 002000 001204  60.2%    0.6% nsh_main
    6     6 128 RR       Kthread --- Waiting  Semaphore 0000000000000000 002008 000444  22.1%    0.0% scsid
nsh> conn
coboard_mscclassobject: Configuring with NLUNS=1
nboard_mscclassobject: MSC handle=n0x2000d9e0
_board_mscclassobject: Bind LUN=0 to /dev/smart0
musbmsc_exportluns: Starting SCSI worker thread
ain:usbmsc_scsi_main: Started
 usbmsc_exportluns: Waiting for the SCSI worker thread
Pusbmsc_exportluns: Signalling for the SCSI worker thread
eusbmsc_scsi_main: Waiting to be signaled
rusbmsc_scsi_main: Running
forminstm32_getreg: 50000804->00000002
stm32_putreg: 50000804<-00000000
gstm32_getreg: 50000804->00000000
stm32_putreg: 50000804<-00000000
 stm32_getreg: 50000014->04008c20
stm32_getreg: 50000014->04008c20
stm32_getreg: 50000018->000c3812
stm32_putreg: 50000014<-00000800
stm32_usbsuspend: resume: 0
stm32_getreg: 50000014->04008420
stm32_getreg: 50000018->000c3812
[... 469 lines cut ...]
dump_stack: IRQ Stack:
dump_stack: sp:     0x2000f110
dump_stack:   base: 0x200003c0
dump_stack:   size: 00002048
dump_stack: ERROR: IRQ Stack pointer is not within the stack
dump_stack: User Stack:
dump_stack: sp:     0x2000f110
dump_stack:   base: 0x2000ea98
dump_stack:   size: 00002008
stack_dump: 0x2000f100: 2000d548 080157ad 00000200 08020613 00000000 7474754e 20000058 0801c0da
stack_dump: 0x2000f120: 2000f1e8 080076e7 08020012 2000f1e0 0000001c 2000f1e8 00000000 080072f9
stack_dump: 0x2000f140: 2000f160 50000b00 2e323100 00302e30 20000bc8 50000b00 84008000 616300f0
stack_dump: 0x2000f160: 32336239 20643137 20727041 32203520 20333230 303a3131 38353a35 08020600
stack_dump: 0x2000f180: 2000f198 080145f9 2000f1ac 2000f198 6d7261bd 08020600 2000f1a4 00000000
stack_dump: 0x2000f1a0: 0801493d 0801c0da 2000d9e0 20000bd0 00000000 000000f0 2000d8e8 0000004c
stack_dump: 0x2000f1c0: 00000000 08006e93 00000002 08019627 2000d888 20000bd0 00000000 000000f0
stack_dump: 0x2000f1e0: 2000d8e8 0801b353 0801b2c9 20000bd0 2000d9d0 000000f0 00000080 00000000
stack_dump: 0x2000f200: 00000000 08002a3f 20001e90 2000d888 20001e90 2000ea88 00000000 0801b4fb
stack_dump: 0x2000f220: 2000f240 20001e90 2000ea88 08014111 00000203 00000000 20001e90 080157ed
stack_dump: 0x2000f240: 00000203 20001e90 2000ea88 080157ad 00000001 080070e3 00000000 080045c3
nsh>

```

</details>

ALTracer, 2023-04-05:
Also see discussion in #8906 and https://github.com/libopencm3/libopencm3-examples/issues/178.
Summoning @raiden00pl for good measure because of involvement in RNDIS, COMPOSITE and stm32h7. Maybe even @davids5 (thanks for allowing disabling GPIO_OTG_ID!)

ALTracer, 2023-04-05:
Wireshark:

with NuttX firmware, host gets -EPROTO and 0-length USB URB in response to GET DESCRIPTOR Request DEVICE to 1.0.0 0-address destination. Request-response time is 0.3ms-1.0ms.
Then hosts sends RESET to USBHUB (this is AMD Ryzen 3200G B450M desktop and USB2.0 front panel port, no issues with Full Speed devices) and repeats DESCRIPTOR Request. 0-length responses. I'll need to look at physical level later, something's wrong. NuttX throws a lot of USBDEV_REGDEBUG traces I enabled, into 0x50000000-0x50001000 region. Without them (synchronous 1-byte FIFO USART at 115200 is /dev/console aka arm bit-bucket, so might slow down) nothing much changes.

with STM32 DfuSe BootROM, host gets URB status success and 18-length data, also a DFU Mode descriptor. Then assigns an address (39 today) and repeats descriptor request-response to this address, all the usual handshake stuff. First request-response time is 2.7ms.

davids5, 2023-04-05:
@ALTracer Have a look at the reference manual and see of the IP block uses on the stm32411 look line the one on the F466/F469

see https://github.com/apache/nuttx/blob/master/arch/arm/src/stm32/stm32_otgfsdev.c#L5349-L5360