apache/nuttx #8986: Move debug.h from include to include/nuttx

xiaoxiang781216, 2023-04-08:
debug.h is a generic header file but not defined by posix, so it is very easy to conflict with external project.

alancassis, 2023-04-24:
Should it be move to include/nuttx/debug.h ?
If so I agree!

xiaoxiang781216, 2023-04-24:
Yes.

pkarashchenko, 2023-04-25:
Yes. That should be fine

PiyushPatle26, 2026-04-01:
Hello @xiaoxiang781216 @alancassis @pkarashchenko  
I've worked on this issue and have a fix ready.

Changes made:
- Added `include/nuttx/debug.h` as the new standard location
- Replaced `include/debug.h` with a backward-compatibility shim that emits a `#warning` and re-includes `<nuttx/debug.h>`, so out-of-tree code continues to build while migrating
- Updated all in-tree `#include <debug.h>` → `#include <nuttx/debug.h>` across ~5720 files in `arch/, boards/, drivers/, fs/, libs/, net/, sched/, wireless/ and documentation`
- Also made this changes for nuttx-apps repository

I have also tested with `sim:nsh` configuration:
```
./tools/configure.sh sim:nsh
make -j$(nproc)
./nuttx
```
Build was complete with no errors. Also the ostest passed:

[image]

Should I go ahead and open a PR for both nuttx and nuttx-apps?