apache/nuttx #9059: Using the ftpc client

xiaotailang, 2023-04-21:
Hello everyone ！
After compiling the FTP client program ftpc on a NuttX system and burning it onto a board, I set up an FTP server on a PC and attempted to connect to it from the board. After successfully logging in to the server, every time I entered the 'ls' command, I received an error. Upon debugging, I found that fopen was returning error code 22, which indicates that the file does not exist or the permissions are insufficient. However, I confirmed that the folder exists and has read/write permissions because I tested the 'put' and 'get' commands and was able to upload and download files successfully. Your team's extensive development experience is greatly appreciated - can you offer any suggestions for this issue?
error as follows:
ftpc_listdir: ERROR: Failed to create ftptest/TMP10.dat: 22                     
nfc> ls failed: 22                                                              
pwd                                                                             
fptc_getreply: Reply: 257 "/home" is the current directory                      
nfc> PWD: /home"                                                                
cd ftptest                                                                      
fptc_getreply: Reply: 250 Directory successfully changed.                       
fptc_getreply: Reply: 257 "/home/ftptest" is the current directory              
pwd                                                                             
fptc_getreply: Reply: 257 "/home/ftptest" is the current directory              
nfc> nfc> PWD: /home/ftptest"                                                   
ls                                                                              
ftpc_listdir: ERROR: Failed to create ftptest/TMP10.dat: 22                     
nfc> ls failed: 22                                                              
ls                                                                              
ftpc_listdir: ERROR: Failed to create ftptest/TMP10.dat: 22                     
nfc> ls failed: 22

alancassis, 2023-04-23:
Hi @xiaotailang
If you are new to NuttX RTOS or don't have enough experience with menuconfig configuration, then it is a good idea to start with an working configuration example. You need to find a board config that enables the feature you want to use, in this case: boards/arm/lpc17xx_40xx/olimex-lpc1766stk/configs/ftpc/defconfig could be a good reference.

BTW, I think there are some issues with NuttX FTPC:

```
NuttShell (NSH) NuttX-12.1.0
nsh> uname -a
NuttX  12.1.0 129a0703c4 Apr 23 2023 12:45:27 xtensa esp32-devkitc

nsh> ifconfig
wlan0   Link encap:Ethernet HWaddr 8c:aa:b5:b5:a7:e8 at RUNNING              
        inet addr:10.0.0.2 DRaddr:10.0.0.1 Mask:255.255.255.0                
                                                                             
             IPv4   TCP   UDP  ICMP                                          
Received     0002  0000  0002  0000                                          
Dropped      0000  0000  0000  0000                                          
  IPv4        VHL: 0000   Frg: 0000                                          
  Checksum   0000  0000  0000  ----                                          
  TCP         ACK: 0000   SYN: 0000                                          
              RST: 0000  0000                                                
  Type       0000  ----  ----  0000                                          
Sent         0000  0000  0000  0000
  Rexmit     ----  0000  ----  ----

nsh> renew wlan0
nsh> ifconfig
wlan0   Link encap:Ethernet HWaddr 8c:aa:b5:b5:a7:e8 at RUNNING
        inet addr:192.168.0.18 DRaddr:192.168.0.1 Mask:255.255.255.0

             IPv4   TCP   UDP  ICMP
Received     0005  0000  0004  0000
Dropped      0001  0000  0000  0000
  IPv4        VHL: 0000   Frg: 0000
  Checksum   0000  0000  0000  ----
  TCP         ACK: 0000   SYN: 0000
              RST: 0000  0000
  Type       0000  ----  ----  0000
Sent         0002  0000  0002  0000
  Rexmit     ----  0000  ----  ----

nsh> ftpc 192.168.0.2
NuttX FTP Client:
udp_input: WARNING: No listener on UDP port
udp_input: WARNING: No listener on UDP port
login ftpclient ********
pwd
nfc> nfc> PWD: /"
udp_input: WARNING: No listener on UDP port
udp_input: WARNING: No listener on UDP port

ls
udp_input: WARNING: No listener on UDP port
udp_input: WARNING: No listener on UDP port
tcp_input: WARNING: SYN with no listener (or old packet) .. reset
tcp_input: WARNING: SYN with no listener (or old packet) .. reset
ftpc_sockconnect: ERROR: connect() failed: 110
ftpc_xfrinit: ERROR: ftpc_sockconnect() failed: 110
ls failed: 9

quit
nfc> nfc> nfc> nfc> nfc> nfc> Exiting...
tcp_input: WARNING: SYN with no listener (or old packet) .. reset
nsh> udp_input: WARNING: No listener on UDP port
udp_input: WARNING: No listener on UDP port
```

Please see attached my ESP32 board configuration.
[esp32_wifi_ftpc.txt](https://github.com/apache/nuttx/files/11304228/esp32_wifi_ftpc.txt)

xiaotailang, 2023-04-24:
Hi@alancassis 
Thank you for your valuable suggestions during your assistance. Have a nice life！

alancassis, 2023-04-24:
You are welcome @xiaotailang 
Hey @wangchen61698 I saw that recently you submitted an improvement to ftpd did you test NuttX ftpc too?

xiaotailang, 2023-04-25:
Hi@alancassis
Thank you for your previous suggestion. Following your advice, I recompiled ftpc and it is now functioning properly. However, it can only be compiled and used in the flat build configuration. When compiling in the protected build configuration, an error occurs during the linking stage of generating the app executable program, stating that the watchdog-related function definitions cannot be found. After investigating, I discovered that these functions are implemented at the kernel layer, and cannot be directly used in the application layer in protected build mode. Besides implementing system calls and registering callback functions, are there any other methods within NuttX to allow application layer access to kernel-level function interfaces in protected build mode?

problem as follows:
staging//libapps.a(ftpc_connect.o): In function `ftpc_connect':
apps/netutils/ftpc/ftpc_connect.c:124: undefined reference to `wd_create'
apps/netutils/ftpc/ftpc_connect.c:171: undefined reference to `wd_start'
netutils/ftpc/ftpc_connect.c:229: undefined reference to `wd_cancel'

xiaotailang, 2023-04-25:
Hi@btashton 
Thank you for your suggestion. I will try to use POSIX timers。

xiaotailang, 2023-04-26:
@acassis My MCU is stm32f767