apache/nuttx #9080: littlefs on SD/eMMC read/write performance is poor

radek-pesina, 2023-04-24:
Hi,

Utilising the littlefs filesystem on SD/eMMC is providing read/write speeds of approximately 50KiB/sec (calculated using dd command). I've tried optimising the menuconfig settings to improve it, including options such as below but still the performance is lacking:

- Turning on CONFIG_MEMCPY_VIK gave slight improvement
- Setting USEC_PER_TICK to 1000 or below gave most improvement, however at the detriment of other aspects of the system.  The fastest speeds observed by adjusting this was ~370KiB/s write and 700KiB/s read though overall this was unacceptable given the effect on the rest of the system.
- Adjusting LittleFS parameters (e.g. CONFIG_FS_LITTLEFS_PROGRAM_SIZE_FACTOR, CONFIG_FS_LITTLEFS_READ_SIZE_FACTOR, CONFIG_FS_LITTLEFS_BLOCK_SIZE_FACTOR, CONFIG_FS_LITTLEFS_CACHE_SIZE_FACTOR, CONFIG_FS_LITTLEFS_LOOKAHEAD_SIZE 
- Ensuring SD/eMMC DMA read/writes are enabled
- Setting MMCSD_MULTIBLOCK_LIMIT to 0

Details of build:

- risc-v architecture using running kernel in S-mode, main clock running @ 75MHz, SD peripheral @ 25MHz
- Toolchain: riscv64-unknown-elf-gcc-8.3.0-2019.08.0-x86_64-linux-ubuntu14
- Nuttx GIT version: https://github.com/apache/nuttx/pull/8922 with local changes in littlefs to use kmm_malloc/free due to using DMA and S-mode kernel
- Built with commands:
```
   $ make distclean -j16
   $ ./tools/configure.sh rv-virt:knsh64
   $ make -j16 
   $ make export -j16
   $ make import -j16  
```
- Executed with command:
`  $ qemu-system-riscv64 -semihosting -M virt,aclint=on -cpu rv64 -smp 8 -bios none -kernel nuttx -nographic -serial mon:stdio`

Our architecture and hardware design requires the product to boot from eMMC and have a fast boot time for a firmware image of approximately 2MiB.  At 50KiB/s the image load itself would consume 40 seconds.

NOTE: Interestingly littlefs performs considerably better than vFAT on SD/eMMC, which on the same setup has read/writes speeds of ~12KiB/s.  Littlefs however on a RAM disk gives speeds of 560KiB/s for write which is considerably slower than vFAT, which measures ~3225KiB/s.

Does anyone have a rough feel on what  performance we could expect when using littefs on SD/eMMC in Nuttx for a configuration like described above and are there ways to improve this?

TimJTi, 2023-04-24:
> Hi,
> Utilising the littlefs filesystem on SD/eMMC is providing read/write speeds
> of approximately 50KiB/sec (calculated using dd command).

I have just tried the same - I have littlfs on GD25 flash memory. It takes 40 seconds to dd a 1.2Mbyte file from that filesystem to /dev/null. That equates to 30KiB/s - on a processor running at 164MHz and flash SPI at 45MHz.

The file is a wav file that should play in 3 seconds...and this might well explain why I've been having issues getting my in-development audio driver to behave, since I am relying on playing this wav file to check it is working!

Doesn't help you right now, but I will look at this later today/tomorrow to see if I can find if there's a reason: and a workaround or fix or whatever. But doesn't sound right, for sure. Might be optimisation or debug levels of course.

TimJTi

alancassis, 2023-04-24:
@radek-pesina do you think it could be LittleFS specific? I remember testing SDCards on iMX1052 3 years ago and it was fast (but I was using SDIO interface instead SPI). More info here: https://acassis.wordpress.com/2020/08/09/getting-the-sdcard-of-imxrt1050-evk-working-on-nuttx/

TimJTi, 2023-04-25:
I realise that my board was configured with SmartFS, but having set full optimisation and turned off all debug, and upped my SPI clock to 90MHz, I still get 29kbytes/s.

I have just changed to LittleFS, and that's a little better at 37kbyte/s. 

Using a ram drive it's 24Mbyte/s.

Since both filesystems behave "badly" I will try and investigate more later this week as it concerns me.

Or is it as-expected perhaps?

>-----Original Message-----
>From: Tim Hardisty
>Sent: Monday, April 24, 2023 1:02 PM
>To: apache/nuttx
>***@***.***>;
>apache/nuttx ***@***.***>
>Cc: Subscribed ***@***.***>
>Subject: RE: [apache/nuttx] littlefs on SD/eMMC read/write performance is
>poor (Issue #9080)
>
>
>> Hi,
>> Utilising the littlefs filesystem on SD/eMMC is providing read/write
>> speeds of approximately 50KiB/sec (calculated using dd command).
>
>I have just tried the same - I have littlfs on GD25 flash memory. It
>takes 40 seconds to dd a 1.2Mbyte file from that filesystem to /dev/null.
>That equates to 30KiB/s - on a processor running at 164MHz and flash SPI
>at 45MHz.
>
>The file is a wav file that should play in 3 seconds...and this might
>well explain why I've been having issues getting my in-development audio
>driver to behave, since I am relying on playing this wav file to check it
>is working!
>
>Doesn't help you right now, but I will look at this later today/tomorrow
>to see if I can find if there's a reason: and a workaround or fix or
>whatever. But doesn't sound right, for sure. Might be optimisation or
>debug levels of course.
>
>TimJTi
>

radek-pesina, 2023-04-25:
> @radek-pesina do you think it could be LittleFS specific? I remember testing SDCards on iMX1050 3 years ago and it was fast (but I was using SDIO interface instead SPI). More info here: https://acassis.wordpress.com/2020/08/09/getting-the-sdcard-of-imxrt1050-evk-working-on-nuttx/
> 
> Please try to use other FS like VFAT or SmartFS just to verify if the issue is on LittleFS or SPI/SD/MMC drivers.

Testing Fat32 provides poorer performance on our target than littlefs (~12KiB/s) therefore as supported by TimJTi's comment above, it does not appear to be related specifically to littlefs.

TimJTi, 2023-04-27:
> I now get 730 KB/s using LittleFS and the same file.

857 KB/s if I disable debug and re-enable normal code optimization

radek-pesina, 2023-05-01:
Greatfindings Tim, and a decent improvement.  On our existing embedded platform was operating at 10MiB/s so still relatively slow compared to where we want to get to but I will find some time soon to see if your findings can be applied to our platform - 800KiB/s will provide a MUCH better starting point than what it's currently performing at :)

xiaoxiang781216, 2023-05-31:
How fast block I/O without filesystem?

radek-pesina, 2023-05-31:
Hi,

Below are some measured speeds with the round-robin interval reduced to 10ms (was initially 200ms, however the improvement was marginal from reducing it to 10ms).  NOTE: This is running on a slightly faster core than the previous measurements, hence why the initial baseline is ~100KiB/s instead of 75KiB/s.

I believe you are more interested in the read/write speeds for eMMC without littlefs on it - I'm trying to DD from /dev/zero to /dev/mmcsd0 however I receive error 6 (no such device or address).  Is there a method you suggest could be used for testing the transfer speed for eMMC?

```
// From block device (RAM) to block device (RAM)
nsh> dd if=/dev/zero of=/dev/null bs=512 count=2000
1024000 bytes copied, 300000 usec, 3333 KB/s
nsh> dd if=/dev/zero of=/dev/null bs=1024 count=2000
2048000 bytes copied, 340000 usec, 5882 KB/s
nsh> dd if=/dev/zero of=/dev/null bs=4095 count=2000
8190000 bytes copied, 570000 usec, 14031 KB/s
nsh> dd if=/dev/zero of=/dev/null bs=16384 count=2000
32768000 bytes copied, 1520000 usec, 21052 KB/s

// From block device (RAM) to littleFS on (/sys - eMMC)
nsh> dd if=/dev/zero of=/sys/t512 bs=512 count=2000
1024000 bytes copied, 9650000 usec, 103 KB/s
nsh> dd if=/dev/zero of=/sys/t1024 bs=1024 count=100
102400 bytes copied, 680000 usec, 147 KB/s
nsh> dd if=/dev/zero of=/sys/t4096 bs=4096 count=100
409600 bytes copied, 2970000 usec, 134 KB/s
nsh> dd if=/dev/zero of=/sys/t16384 bs=16384 count=50
819200 bytes copied, 5940000 usec, 134 KB/s

// from littleFS (on /sys - eMMC) to block device (RAM)
nsh> dd if=/sys/t512 of=/dev/null bs=512
1024000 bytes copied, 9160000 usec, 109 KB/s
nsh> dd if=/sys/t1024 of=/dev/null bs=1024
102400 bytes copied, 510000 usec, 196 KB/s
nsh> dd if=/sys/t4096 of=/dev/null bs=4096
409600 bytes copied, 2800000 usec, 142 KB/s
```

xiaoxiang781216, 2023-05-31:
> Hi,
>
> Below are some measured speeds with the round-robin interval reduced to 10ms (was initially 200ms, however the improvement was marginal from reducing it to 10ms). NOTE: This is running on a [...]

Yes, to investigate file system performance, we must ensure the performance at block level match the hardware spec.

> - I'm trying to DD from /dev/zero to /dev/mmcsd0 however I receive error 6 (no such device or address). Is there a method you suggest could be used for testing the transfer speed for eMMC?

Do you enable CONFIG_BCH?

rtucker85, 2023-06-01:
> @radek-pesina do you think it could be LittleFS specific? I remember testing SDCards on iMX1050 3 years ago and it was fast (but I was using SDIO interface instead SPI). More info here: [...]

Hi @alancassis.
Radek is also using SDIO (4bit).
Out of interest, when you say "fast", exactly what sort of performance were you getting? There is no mention of throughput numbers in your blog post...

g2gps, 2023-06-15:
@xiaoxiang781216 

I did a bit of testing in this area.

First, I hacked in a crude virt_to_phys implementation so our DMA engine could use userspace addresses:

```diff
diff --git a/arch/risc-v/src/litex/litex_sdio.c b/arch/risc-v/src/litex/litex_sdio.c
index eec66ed295..583e574085 100644
--- a/arch/risc-v/src/litex/litex_sdio.c
+++ b/arch/risc-v/src/litex/litex_sdio.c
@@ -828,6 +828,23 @@ static int litex_cancel(struct sdio_dev_s *dev)
   return OK;
 }
 
+static uintptr_t virt_to_phys(uintptr_t vaddr)
+{
+  FAR struct tcb_s *tcb = nxsched_self();
+  uintptr_t         page;
+
+  if(tcb->addrenv_own) {
+      page = up_addrenv_find_page(&tcb->addrenv_own->addrenv, vaddr);
+      assert(page);
+      return page | (vaddr & 0xFFF);
+  }
+  else
+  {
+    /* Assume the address is in the kernel address space.*/
+    return vaddr;
+  }
+}
+
 /****************************************************************************
  * Name: litex_dmarecvsetup
  *
@@ -852,6 +869,9 @@ static int litex_recvsetup(struct sdio_dev_s *dev, uint8_t *buffer,
 {
   struct litex_dev_s *priv = (struct litex_dev_s *)dev;
   DEBUGASSERT(priv != NULL && buffer != NULL && nbytes > 0);
+
+  buffer = (uint8_t*)virt_to_phys((uintptr_t)buffer);
+
   DEBUGASSERT(((uint32_t)buffer & 3) == 0);
 
   mcinfo("buffer: %p bytes: %u\n", buffer, nbytes);
@@ -897,6 +917,9 @@ static int litex_sendsetup(struct sdio_dev_s *dev,
 {
   struct litex_dev_s *priv = (struct litex_dev_s *)dev;
   DEBUGASSERT(priv != NULL && buffer != NULL && nbytes > 0);
+
+  buffer = (uint8_t*)virt_to_phys((uintptr_t)buffer);
+
   DEBUGASSERT(((uint32_t)buffer & 3) == 0);
 
   mcinfo("buffer: %p bytes: %u\n", buffer, nbytes);
```

I then modified the DD command handling to accept larger block counts:
```diff
diff --git a/nshlib/nsh_ddcmd.c b/nshlib/nsh_ddcmd.c
index 7b005d3d2..45fc4dfca 100644
--- a/nshlib/nsh_ddcmd.c
+++ b/nshlib/nsh_ddcmd.c
@@ -76,8 +76,8 @@ struct dd_s
   uint32_t     skip;       /* The number of sectors skipped on input */
   bool         eof;        /* true: The end of the input or output file has been hit */
   bool         verify;     /* true: Verify infile and outfile correctness */
-  uint16_t     sectsize;   /* Size of one sector */
-  uint16_t     nbytes;     /* Number of valid bytes in the buffer */
+  long         sectsize;   /* Size of one sector */
[... 141 lines cut ...]
CONFIG_ARCH_TEXT_VBASE=0x10000000
CONFIG_ARCH_USE_MMU=y
CONFIG_ARCH_USE_MPU=y
CONFIG_ARCH_USE_S_MODE=y
CONFIG_BCH=y
CONFIG_BUILD_KERNEL=y
CONFIG_LITEX_COHERENT_DMA=y
CONFIG_LITEX_SDIO=y
CONFIG_MEMCPY_VIK=y
CONFIG_MEMSET_OPTSPEED=y
CONFIG_MM_PGALLOC=y
CONFIG_NSH_CMDOPT_DD_STATS=y
CONFIG_RISCV_TOOLCHAIN_GNU_RV32=y
CONFIG_RR_INTERVAL=200
CONFIG_SCHED_TICKLESS=y
CONFIG_SYSTEM_NSH=y
CONFIG_SYSTEM_NSH_PROGNAME="init"
CONFIG_USEC_PER_TICK=1
```

From the results, it seems like the throughput to and from the SD card @ `/dev/sd0` is sufficiently fast, once the block transfer has been initialised, capping out at ~16.2 MB/s read and ~9.7MB/s write. However, lower transfers with single, or lower block counts very slow. For the most part, this is expected, as there is significant overhead required to setup the transaction.

There is a generic rwbuffer under `drivers/misc/rwbuffer.c`, which has been used previously for `mtd`. It seems like adding this, optionally, around the `mmc_sdio` driver would allow for a reduction in block device access and improve performance significantly.

Is that something which has been considered previously? Would it be a feasible addition?

xiaoxiang781216, 2023-06-15:
> There is a generic rwbuffer under `drivers/misc/rwbuffer.c`, which has been used previously for `mtd`. It seems like adding this, optionally, around the `mmc_sdio` driver would allow for a reduction [...]

Yes, we consider this approach before, but not implement yet.

xiaoxiang781216, 2025-08-07:
littlefs is designed to work with the raw flash, so the performance is very poor compared with other block base file system(e.g. fat or ext4).

linguini1, 2025-08-07:
> littlefs is designed to work with the raw flash, so the performance is very poor compared with other block base file system(e.g. fat or ext4).

Right, but do you know what the underlying reason is?

xiaoxiang781216, 2025-08-10:
sd/emmc already have wear leveling in the card side, the wear leveling inside littlefs is useless, but slow down the performance.