apache/nuttx #9092: Removing app via menuconfig causes build fail

TimJTi, 2023-04-25:
If you deselect an app using` make menuconfig `the build always fails, even if you do a `make clean` and a `make apps_clean.`

The only solution is a `make distclean` which doesn't seem right to me?

acassis, 2023-04-25:
@TimJTi seems it is something specific for your setup or your board, etc. I didn't supply much information to duplicate the issue.

Here is a test confirming that building system is working as expected:

$ ./tools/configure.sh esp32-devkitc:nsh

$ make menuconfig

Enable HELLO

$ make -j

Compiled fine:

LD: nuttx
CP: nuttx.hex
MKIMAGE: ESP32 binary
esptool.py -c esp32 elf2image -fs 4MB -fm dio -ff 40m -o nuttx.bin nuttx
esptool.py v3.3
Creating esp32 image...
Merged 1 ELF section
Successfully created esp32 image.
Generated: nuttx.bin (ESP32 compatible)

$ make menuconfig

Disable HELLO

$ make -j

Compiled fine:

LD: nuttx
CP: nuttx.hex
MKIMAGE: ESP32 binary
esptool.py -c esp32 elf2image -fs 4MB -fm dio -ff 40m -o nuttx.bin nuttx
esptool.py v3.3
Creating esp32 image...
Merged 1 ELF section
Successfully created esp32 image.
Generated: nuttx.bin (ESP32 compatible)

TimJTi, 2023-04-25:
Hmmm....this has been the case (for me) for a few months or more, but wasn't an issue a year ago. I wonder it it is related to having out-of-tree apps which I only added relatively recently. I have a symbolic link to my custom apps folder - I will have a play around, and see if I can work out the reason.

TimJTi, 2023-04-27:
I went to investigate this and now can't replicate it. Either something has changed or I was being a numpty (most likely) so I will close for now and can always reopen if I see it again and can provide more information.

TimJTi, 2023-07-18:
Reopened in error!

closed 2023-07-18 by TimJTi