apache/nuttx #9182: MTD device may not behave correctly

michallenc, 2023-05-04:
I am using NOR flash W25QxxJV as MTD device wrapped in FTL layer (therefore no file system is used) and performing read/write operations normally via `read` and `write` POSIX interfaces. Everything works fine until I have to write to next block page (one block page is 256 bytes for my flash). In that case the whole erase page (4096 bytes) is erased for some reason and previously written data are lost.

So far what I was able to get from the code is that FTL layer reads the entire erase page, saves it to buffer, adds new data to the buffer on correct offset, erases the erase page and then saves the buffer back to flash. This all happens in this function https://github.com/apache/nuttx/blob/master/drivers/mtd/ftl.c#L243.

Has someone experienced similar behavior? I am trying to figure out whether the problem is in FTL layer, MTD layer or W25QxxJV flash driver.

For reference here is my MTD device configuration:

```C
static struct mtd_partition_s g_mtd_partition_table[] =
{
  {
    .size    = 0x2000, /* 8KiB */
    .devpath = "/dev/mtdconfig",
  },
  {
    .size    = 0x800000, /* 8MiB */
    .devpath = "/dev/mtdlog_maintenance",
  },
  {
    .size = 0, /* Calculate the rest of the size */
    .devpath = "/dev/mtdlog_trace",
  }
};

static const size_t g_mtd_partition_table_size =
    nitems(g_mtd_partition_table);
static struct mtd_dev_s
*sam_w25qxxxjv_alloc_mtdpart(struct mtd_dev_s *mtd,
                             struct mtd_geometry_s *geo,
                             struct mtd_partition_s
                             *part)
{
  ASSERT((part->offset + part->size) <= geo->neraseblocks * geo->erasesize);
  ASSERT((part->offset % geo->erasesize) == 0);
  ASSERT((part->size % geo->erasesize) == 0);

  size_t startblock = part->offset / geo->blocksize;
  size_t blocks = part->size / geo->blocksize;

  return mtd_partition(mtd, startblock, blocks);
}

int sam_w25qxxxjv_init(void)
{
  int ret = OK;

  struct qspi_dev_s *qspi = sam_qspi_initialize(QSPI_PORTNO);
  if (!qspi)
    {
      syslog(LOG_ERR, "ERROR: sam_qspi_initialize failed\n");
      return ERROR;
    }

  /* Use the QSPI device instance to initialize the W25QxxJV device.  */

  struct mtd_dev_s *mtd = w25qxxxjv_initialize(qspi, true);
  if (!mtd)
    {
      syslog(LOG_ERR, "ERROR: w25qxxxjv_initialize failed\n");
      return ERROR;
    }

  /* Get geometry of W25QxxxJV */

  struct mtd_geometry_s geometry;
  ret = MTD_IOCTL(mtd, MTDIOC_GEOMETRY, (unsigned long)&geometry);
  if (ret < 0)
    {
      syslog(LOG_ERR, "ERROR: w25qxxxjv ioctl failed: %s\n",
             strerror(errno));
      return ret;
    }

  size_t i;
  size_t offset = 0;
  for (i = 0; i < g_mtd_partition_table_size; i++)
    {
      struct mtd_partition_s *part = &g_mtd_partition_table[i];
      part->offset = offset;
      if (part->size == 0)
        {
          part->size = (geometry.neraseblocks * geometry.erasesize) - offset;
        }

      part->mtd = sam_w25qxxxjv_alloc_mtdpart(mtd, &geometry, part);
      if (!part->mtd)
        {
          syslog(LOG_ERR, "ERROR: sam_w25qxxxjv_alloc_mtdpart failed\n");
          return ERROR;
        }

      /* Use the FTL layer to wrap the MTD driver as a block driver */

      ret = ftl_initialize(W25QXXJV_MTD_MINOR + i, part->mtd);
      if (ret < 0)
        {
          syslog(LOG_ERR, "ERROR: Failed to initialize the FTL layer: %d\n",
                 ret);
          return ret;
        }
      offset += part->size;
    }
  return ret;
}
```

acassis, 2023-05-04:
@michallenc did you try more than a board with 1 GB flash? Maybe the flash could be damaged (not common, but it is possible). I never used it but I know it is possible to simulate a Flash on NuttX. Maybe it could help to spot some issue case the issue is not in the flash driver. Search for "Simulated SPI FLASH with SMARTFS"

ALTracer, 2023-05-05:
> Everything seems to be working fine on 512 MB size W25QxxJV flash, the problems only occur on 1 GB flash, therefore the bug is definitely in W25QxxJV driver.

Where did you get yourself a whopping 1 GiB (8Gbit) serial flash? The biggest I found is W25Q01JV, which is 1 Gbit (128MiB), and it actually contains two 512Mbit (64MiB) dies.

I think NuttX MTD w25q driver only supports single-die flash chips with up to 512 mbit density (64MiB). Dual-die chips have separate status registers etc.

Please post your exact flash part number and read its datasheet. Also see the Dual die stack operations appnote.

acassis, 2023-05-05:
Good point @ALTracer I think probably it is 1Gbit, it is not common to find 1GiB SPI NOR Flash

michallenc, 2023-05-06:
@ALTracer Yes, it is W25Q01JV with two 512Mbit dies. I initially thought the driver supports this one since W25Q01JV is recognized from flash ID and required defines (block page, size etc.). I am currently trying to get my hands on dual die stack application note as the document is not open but one has to send a request to technical support (dont get why...).

I will take a look into possible implementation of dual die flash once I get the document.

TimJTi, 2023-05-06:
Datasheet is readily available, though, and should be enough to work out what changes are needed?

michallenc, 2023-05-06:
@TimJTi If you mean datasheet [from this page](https://www.winbond.com/hq/product/code-storage-flash-memory/serial-nor-flash/?__locale=en&partNo=W25Q01JV) then there is nothing about dial die operation. Just mentions that _The two die stack operation detailed on Two Die Stack Operations Application Note_.

btashton, 2023-05-06:
From recollection of working with theses in the past (not NuttX) from the software side you just treat them as two different chips and use 
Software Die Select instruction to select which one is active.  If the driver is not using that instruction you will only ever use the first die.

TimJTi, 2023-05-07:
> From recollection of working with theses in the past (not NuttX) from the software side you just treat them as two different chips and use Software Die Select instruction to select which one is active. If the driver is not using that instruction you will only ever use the first die.

Ah, yes, I remember doing that with a similar device in the past, long ago. The die select instruction is listed in the datasheet but not documented that I could see, but I seem to recall it's exactly as you suggest.

acassis, 2023-05-11:
@michallenc did you get W25QxxJV working? I just tested W25Q512JVB here in a STM32F7 board and confirmed that it is working in QuadSPI mode. Unfortunately I need to change the CONFIG_W25QXXXJV_QSPI_FREQUENCY to 64MHz because there is a bug in this chip that doesn't accept 133MHz.

michallenc, 2023-05-14:
@acassis Partially. I got an application note from Winbond this week so I implemented dual die support according to this. I still need to do the tests tomorrow.

I was mostly stuck at not being capable to even get the first die working. This was caused by flash not enabling QuadSPI mode which should be fixed by this PR https://github.com/apache/nuttx/pull/9272. I am not sure why this problem did not occur on W25Q512JV flash (I also was able to make it work in QuadSPI mode), probably the mode is enabled there by default?

acassis, 2023-05-14:
@michallenc good finding!!!

Yes, probably it is enabled by default in W25Q512JV, really crash!

closed 2023-05-17 by xiaoxiang781216