apache/nuttx #9185: cp: open failed: 2

GooTal, 2023-05-04:
Hi, i built and run rv-virt:nsh64 and found that cp command is not working.

The command:
```
make distclean
./tools/configure.sh -l rv-virt:nsh64 && make -j12
qemu-system-riscv64 -semihosting -M virt,aclint=on -cpu rv64 -smp 1 -bios none -kernel nuttx -nographic -s -S
```

Built on the branch `releases/12.0`

Then i mounted hostfs.

I try to copy file from hostfs, but is seems not OK:
[image]

I try to copy the status file, also failed:
[image]

Is this a bug? Or could you provide me some advice in case i used in a wrong way? Thanks.

acassis, 2023-05-04:
@GooTal I think it is not a BUG, you are trying to copy to a read-only filesystem.
You need to mount a RAMDISK or TMPFS to get a read/write FS!

GooTal, 2023-05-05:
Oh thanks. i tried this and it worked.
```
mkrd 1024
mkfatfs /dev/ram0
mount -t vfat /dev/ram0 /tmp
mount -t hostfs -o fs=. /host
cp /host/AUTHORS /tmp
cat /tmp/AUTHORS
```

closed 2023-05-05 by GooTal