apache/nuttx #9246: Implementing IPV6_MULTICAST_HOPS and IPV6_UNICAST_HOPS.

g2gps, 2023-05-11:
I'm currently porting porting and existing application to Nuttx which expects to be able to set the IPv6 unicast and multicast hop limit to 255. From what I can see in `net/inet/ipv6_setsockopt.c`, these socket options are currently not implemented, with the default value being set to 64 in `nuttx/nuttx/include/nuttx/net/netconfig.h`:
```c
/* The IP TTL (time to live) of IP packets sent by the network stack.
 *
 * This should normally not be changed.
 */

#define IP_TTL_DEFAULT 64
``` 
and subsequently fixed as a parameter of a UDP connection in `nuttx/nuttx/net/udp/udp_conn.c`:
```c
#if defined(CONFIG_NET_IPv4) && defined(CONFIG_NET_IPv6)
      conn->domain  = domain;
#endif
      conn->lport   = 0;
      conn->ttl     = IP_TTL_DEFAULT;
#if CONFIG_NET_RECV_BUFSIZE > 0
      conn->rcvbufs = CONFIG_NET_RECV_BUFSIZE;
#endif
```

It is possible to implement the socket options, for UDP only, with the following:
```diff
@@ -35,6 +35,7 @@
 #include "mld/mld.h"
 #include "inet/inet.h"
 #include "socket/socket.h"
+#include "udp/udp.h"
 
 #if defined(CONFIG_NET_IPv6) && defined(CONFIG_NET_SOCKOPTS)
 
@@ -112,21 +113,57 @@ int ipv6_setsockopt(FAR struct socket *psock, int option,
         }
         break;
 
+#ifdef NET_UDP_HAVE_STACK
+      case IPV6_MULTICAST_HOPS:   /* Multicast hop limit */
+        {
+          FAR struct udp_conn_s *conn;
+          uint8_t ttl;
+          if (psock->s_type != SOCK_DGRAM ||
+              value == NULL || value_len == 0)
+            {
+              ret = -EINVAL;
+              break;
+            }
+
+          ttl = *(FAR uint8_t *)value;
+          conn = (FAR struct udp_conn_s *)psock->s_conn;
+          conn->ttl = ttl;
+          ret = OK;
+        }
+        break;
+#endif
       /* The following IPv6 socket options are defined, but not implemented */
 
-      case IPV6_MULTICAST_HOPS:   /* Multicast hop limit */
       case IPV6_MULTICAST_IF:     /* Interface to use for outgoing multicast
                                    * packets */
       case IPV6_MULTICAST_LOOP:   /* Multicast packets are delivered back to
                                    * the local application */
 #endif
-      case IPV6_UNICAST_HOPS:     /* Unicast hop limit */
       case IPV6_V6ONLY:           /* Restrict AF_INET6 socket to IPv6
                                    * communications only */
         nwarn("WARNING: Unimplemented IPv6 option: %d\n", option);
         ret = -ENOSYS;
         break;
 
+#ifdef NET_UDP_HAVE_STACK
+      case IPV6_UNICAST_HOPS:
+        {
+          FAR struct udp_conn_s *conn;
+          uint8_t ttl;
+          if (psock->s_type != SOCK_DGRAM ||
+              value == NULL || value_len == 0)
+            {
+              ret = -EINVAL;
+              break;
+            }
+
+          ttl = *(FAR uint8_t *)value;
+          conn = (FAR struct udp_conn_s *)psock->s_conn;
+          conn->ttl = ttl;
+          ret = OK;
+        }
+        break;
+#endif
       case IPV6_RECVPKTINFO:
       case IPV6_RECVHOPLIMIT:
         {
```

This works for our application, however, I do not think it's a great solutions as:
 - It's only suitable for UDP, and adds protocol dependent characteristics back into the more generic IVP6 socket options. I noticed this was just cleaned up in #9214.
 - The limit is applied to multicast and unicast. I'm not sure on the implications here.
 - If the ttl, or hop limit socket option was to be added for other transports, a case would need to be added for each underlying connection type.

I'm not overly familiar with networking stack typologies, so I'm not sure what the best approach is in this case. Should the ttl, or hop limit be an attribute of the socket, with the transport layer using its value when it constructs the IP header?

xiaoxiang781216, 2023-05-11:
@zhhyu7 could you take a look?

zhhyu7, 2023-05-12:
> @zhhyu7 could you take a look?

Ok, I'll look at the relevant appropriate implementation and add it.

xiaoxiang781216, 2023-05-13:
let's wait the upcoming patch to add more ttl setting.

closed 2023-05-13 by xiaoxiang781216