apache/nuttx #9268: ARM64: Lack of other options about interrupt controller

zhoukejun, 2023-05-14:
In arch/arm64/Kconfig,  the ARM_GIC_VERSION is 3 by default. But some other arm64 chip-set don't have the GIC, such as BCM2837B0 (raspberry pi 3b+). The BCM2837B0 has a local interrupt controller.
How about add more options about interrupt controller for ARM64 variant chip-sets.

xiaoxiang781216, 2023-05-14:
one approach could extend the range from -1 to 4 and select -1 to disable all gic related code.

zhoukejun, 2023-05-14:
> one approach could extend the range from -1 to 4 and select -1 to disable all gic related code
That is a good idea. The following patch works fine.

------------------------------
diff --git a/arch/arm64/Kconfig b/arch/arm64/Kconfig
index 7aeb721f47..bfdac6f1fd 100644
--- a/arch/arm64/Kconfig
+++ b/arch/arm64/Kconfig
@@ -214,7 +214,7 @@ config ARM_GIC_VERSION
        int "GIC version"
        default 2           if ARCH_CHIP_A64
        default 3
-       range 2 4
+       range -1 4
        ---help---
                Version of Generic Interrupt Controller (GIC) supported by the
                architecture

xiaoxiang781216, 2023-05-29:
The problem is fixed, let's close this issue.

closed 2023-05-29 by xiaoxiang781216