apache/nuttx #9270: Please add "sudo" to all Ubuntu Getting Started / Installing patterns

dok-net, 2023-05-14:
On Ubuntu, one doesn't log in as root, but uses "sudo". Only the very first "sudo apt install" instruction acknowledges this, the remaining setup instructions don't work copy&paste-style without adding "sudo" manually. Maybe remove the "$" prompt and implement a "copy" icon, too, like on so many other sites.

btashton, 2023-05-14:
This is a good suggestion.  We should drop the $ and make sure it is indicated as bash code.  

The sphinx-copybutton will provide the easy copy button as you suggest.

patacongo, 2023-05-14:
`sudo `is not supported by Cygwin or MSYS.  Add sudo to the commands will break those build.

So it probably should be define as SUDO in tools/Config.mk with SUDO evaluating to "" for those platforms.

btashton, 2023-05-14:
> `sudo` is not supported by Cygwin or MSYS.  Adding sudo to the commands would break those build.
> 
> So it probably should be qualified for those platforms.
> 
> But that apt is not supported on those platforms either.

This is done per os in the docs. See https://nuttx.apache.org/docs/latest/quickstart/install.html

closed 2023-05-20 by xiaoxiang781216