apache/nuttx #9273: trace not working on rv-virt:knsh64?

GooTal, 2023-05-15:
I did as [task trace user guide](https://nuttx.apache.org/docs/latest/guides/tasktraceuser.html) told. Enabled these configuration:
[image]

I also enabled these configuration:
```
CONFIG_TRACE=y
CONFIG_TRACE_ARCH=y
CONFIG_TRACE_AUDIO=y
CONFIG_TRACE_BOARDS=y
CONFIG_TRACE_CRYPTO=y
CONFIG_TRACE_DRIVERS=y
CONFIG_TRACE_FS=y
CONFIG_TRACE_GRAPHICS=y
CONFIG_TRACE_LIBS=y
CONFIG_TRACE_NET=y
CONFIG_TRACE_SCHED=y
CONFIG_TRACE_VIDEO=y
CONFIG_TRACE_WIRELESS=y
```

Then i built rv-virt:knsh64 from branch releases/12.1. But trace is still not found.
[image]

Thanks if you could help.

xiaoxiang781216, 2023-05-15:
you need turn on CONFIG_SYSTEM_TRACE to get the trace command.

GooTal, 2023-05-15:
Yes, i have turned it on. Exactly as [task trace user guide](https://nuttx.apache.org/docs/latest/guides/tasktraceuser.html) told.

[image]

But rv-virt:knsh64 is not working. 
[image]

BTW, rv-virt:nsh64 (flat build) works fine. Is there something wrong with kernel build?

xiaoxiang781216, 2023-05-15:
@GooTal are you looking the opportunity of internship?

xiaoxiang781216, 2023-05-15:
> BTW, rv-virt:nsh64 (flat build) works fine. Is there something wrong with kernel build?

do you mount hostfs? trace should be put into apps/bin/ folder, you may need luanch the program with the full path.

GooTal, 2023-05-15:
> @GooTal are you looking the opportunity of internship?

Yes i have been looking for the opportunity of intership. But the tutor does not allow us to have summer intership. :( 
Maybe i could get in touch with the mentor or HRs of companies by submitting my resume.

GooTal, 2023-05-15:
> > BTW, rv-virt:nsh64 (flat build) works fine. Is there something wrong with kernel build?
> 
> do you mount hostfs? trace should be put into apps/bin/ folder, you may need luanch the program with the full path.

Yes, i mounted hostfs. I could use the `cat /host/AUTHORS` cmd to read the authors.
And trace is successfully compiled and put into apps/bin/ folder.
[image]

[image]

And hello works fine. trace still not working.
[image]

GooTal, 2023-05-15:
Oh, it called task_spawn [here](https://github.com/apache/nuttx-apps/blob/0acd68b391244d8edf5c3bc78e2b0ce818868a5b/system/system/system.c#L139), which is not supported for kernel build.
[image]

So the kernel build requires CONFIG_SYSTEM_SYSTEM_SHPATH to enable posix_spawn [here](https://github.com/apache/nuttx-apps/blob/0acd68b391244d8edf5c3bc78e2b0ce818868a5b/system/system/system.c#LL135C8-L135C35). 

Additionally, CONFIG_SYSTEM_SYSTEM_SHPATH depends on `SYSTEM_NSH=m` [here](https://github.com/apache/nuttx-apps/blob/0acd68b391244d8edf5c3bc78e2b0ce818868a5b/system/system/Kconfig#L20) , which was `SYSTEM_NSH=y` previously.

So i changed SYSTEM_NSH=m and enabled CONFIG_SYSTEM_SYSTEM_SHPATH. Problem solved for rv-virt:knsh64.

GooTal, 2023-05-15:
Thanks a lot. I\`ve sent my wechat number to your gmail.

xiaoxiang781216, 2023-05-29:
The problem is resolved, let's close it now.

closed 2023-05-29 by xiaoxiang781216