apache/nuttx #9275: Requesting help to reproduce a memory corruption crash in task creation

slorquet, 2023-05-15:
Hello,

I found a crash when trying to read /proc/%d/cmdline on a kthread manually created with argv=NULL.

After some unresponded messages on the mailing list I got no feedback so here is a github issue.

The crash happens when trying to read /proc/%d/cmdline, but using gdb watchpoints, I have determined that the memory address that stores tcb->group->tg_info->argv[n] is written  by an later arm exception, which is most probably a context switch.

This leads me to believe that the stack adjustment does not work correctly in certain cases.

This is a very annoying bug with a memory corruption and I would like to fix it, but I need to reproduce it in an isolated case that developers can reproduce.

Who has nucleo h7 boards and is able to help me fix this?

Thank you for your help.

slorquet, 2023-05-15:
Can some one please help me with the CI? like it failed successfully but I have no idea why

https://github.com/apache/nuttx/pull/9277

slorquet, 2023-05-17:
No one cares, ok. worked around but still no root cause for this.

closed 2023-05-17 by slorquet

slorquet, 2023-05-17:
No worries, I worked around it.

I was waiting to integrate watchdog kicker support, so people can reproduce it, thats why there are not a lot of info as of now.

However, lots of tests showed that I could not reproduce it with a canned config on the nucleo board.

The best fix was to avoid the issue, which is not very satisfying, but seem to work.

If someone finds weirdness around corruption of kthread arguments, that might be it reappearing.

btashton, 2023-05-17:
I have the board your PR was for so if you do end up finding some kind of repro with it I am happy to dig it out of the box and try to help.

mu578, 2023-05-17:
@slorquet get some love and be patient, they are virtues. Do you know or being fully aware of that you are addressing to other human beings like you? not robots, think about it. Thank you.