apache/nuttx #9298: Signal must be masked when it is delivered to a signal handler

patacongo, 2023-05-16:
Signal must be masked when it is delivered to a signal handler per:

> https://pubs.opengroup.org/onlinepubs/007904875/functions/sigaction.html
>     When a signal is caught by a signal-catching function installed by sigaction(), a new signal mask is calculated and installed for the duration of the signal-catching function (or until a call to either [sigprocmask()](https://pubs.opengroup.org/onlinepubs/007904875/functions/sigprocmask.html) or [sigsuspend()](https://pubs.opengroup.org/onlinepubs/007904875/functions/sigsuspend.html) is made). This mask is formed by taking the union of the current signal mask and the value of the sa_mask for the signal being delivered [XSI] [Option Start] unless SA_NODEFER or SA_RESETHAND is set, [Option End] and then _including the signal being delivered_. If and when the user's signal handler returns normally, the original signal mask is restored.

Any action queued for that signal while the signal is masked should be deferred.  It should go into the group pending signal list and should not be processed until until the signal is unmasked (which should occur when the signal handler returns).

mu578, 2023-05-16:
I think the issue is a signal maybe delivered while its handler is running, so yes, it needs to be withheld or block. `Are nested signals should be banned? (absolutism)` I don't know might break legit things out there.

patacongo, 2023-05-16:
Close with #9299

patacongo, 2023-05-16:
> Should nested signals be banned? (absolutism)

I am not sure.  Certainly be default, handles for the same signal number are forced to be sequential (by the POSIX the requirement to mask the signal on entry into the signal handler).  But:

- You can always modify the sigprocmask within the signal handler, and
- That would not apply to other signals.

Interrupting one signal handler to process another implies some priority handling of signals.  Why would you interrupt one handler to run another versus letting the current handler run to completion first.  I think that the dispatch logic currently will not permit nested signal handlers (but I would need to verify that).

patacongo, 2023-05-16:
> Why would you interrupt one handler to run another versus letting the current handler run to completion first.

Signals do have a priority associated with them and should run in priority order.  So it might not be a bad thing to permit a higher priority signal to interrupt a lower priority signal.  Unfortunately signal prioritization is not currently supported:  #8899

patacongo, 2023-05-16:
> I think the issue is a signal maybe delivered while its handler is running (the nested case being the shortest path to expose it) , so yes, it needs to be withheld or block (to let that handler a chance to exit).

This PR fixes the problem because it eliminates _all_ nesting.  There was a window in which xcp.sigdeliver was NULL when it should not have been.  That permitted an unsupported nested signal action to occur because of the logic in  https://github.com/apache/nuttx/blob/master/arch/arm/src/armv7-m/arm_schedulesigaction.c#L88

In the current design, any nested signal will clobber xcp.sigdelver and crash the system.  This PR does not change that.

closed 2023-05-17 by xiaoxiang781216