apache/nuttx #9314: how to manage hardware encryption modules

xiaotailang, 2023-05-18:
Hello everyone, first of all, thank you to the community for your enthusiastic answers to my previous questions and for providing valuable advice. I have some hardware encryption modules, such as AES, RAS, etc., that I want to use in the NuttX system. However, I'm not sure how to manage these hardware encryption modules in the NuttX system. Through my research, I found a crypto directory under NuttX, which seems to be used for encryption-related purposes, but it only seems to implement various encryption algorithms in pure software. At the same time, I found that support for some hardware encryption modules, such as ESP32C3, was recently added to NuttX 12.1, but I couldn't find how to use these encryption modules at the application layer. I plan to manage each hardware encryption module through a character device. Is managing them through character devices feasible? Are there better ways to manage them in NuttX?

mu578, 2023-05-18:
Better interfacing in memory via ioctl, you really don't want using a character device for many obvious reasons. However, to my knowledge, I don't think there is any existing layer ; would be new to architect, new territory to explore.

xiaoxiang781216, 2023-05-18:
> Hello everyone, first of all, thank you to the community for your enthusiastic answers to my previous questions and for providing valuable advice. I have some hardware encryption modules, such as [...]

here has some demo: https://github.com/apache/nuttx-apps/tree/master/testing/crypto
NuttX verson come from openbsd: https://github.com/openbsd/src/tree/master/sys/crypto
So you can find many example from there.

> I plan to manage each hardware encryption module through a character device. Is managing them through character devices feasible? Are there better ways to manage them in NuttX?

I would suggest that you follow the crypto framework since many other components will base on the new crypto framework:

1. New IPSec will call the crypto framework directly
2. Bridge mbedtls to crypto framework for hardware acceleration

mu578, 2023-05-18:
You can redefine crypto primitives and engine within mbedtls, however, it would require to adopt something like PSA Crypto API/Protocol within NuttX. This is a very long road ahead to get there.

xiaotailang, 2023-05-19:
Thank you for your valuable advice, have a happy life!

mu578, 2023-05-19:
@xiaotailang 超可愛的，我喜歡 , 你也一樣法語怎麼說吗 , that is just to make you aware of and comprehend the true underlaying complexity of such task and request.

Meanwhile, for educational purpose using hardware primitives from kern to user space can be done using ioctl ; make your own personal `syscalls` ; would be a first good step to achieve. Start small think big.

xiaoxiang781216, 2023-05-29:
> You can redefine crypto primitives and engine within mbedtls, however, it would require to adopt something like PSA Crypto API/Protocol within NuttX (from bottom to top and top to bottom). This is a very long road ahead to get there. Not a student or a single man project.
> 
> Update: zephyr team seems to move into/toward that direction: [zephyrproject-rtos/zephyr#43712](https://github.com/zephyrproject-rtos/zephyr/issues/43712) https://github.com/zephyrproject-rtos/zephyr/projects/24

@mu578 @xiaotailang we will implement both alt and psa on top of /dev/crytpo

closed 2023-05-29 by xiaoxiang781216

xiaotailang, 2023-05-30:
@xiaoxiang781216 @mu578 Okay，Thank you again for your valuable advice. I plan to follow the suggestions you gave me  and wish you a happy life。

mu578, 2023-05-30:
@xiaoxiang781216 ok, @xiaotailang it will give you a hand on primitives and how to handle them ; this is not so easy to get ; then you come back on PSA-project opening, with that understanding.