apache/nuttx #9327: Tiva board won't boot when FPU is enabled

fdcavalcanti, 2023-05-19:
This board works fine with all default configurations. I'm using it regularly without issues.

However, by simply enabling FPU Support (ARCH_FPU) causes the board to not event turn the LEDs on.
I also tried enabling all kinds of debug statements, and nothing shows on serial port.
Compilation and flashing all complete without errors.

I would appreciate any indications on where I could look for issues. So far I tried tracking CONFIG_ARCH_FPU but no success.

Board: Tiva-C TM4C1294-XL

Thanks!

acassis, 2023-05-20:
Thank you for reporting this issue!

@gustavonihei and @qinwei2004 I think you both have much experience with FPU issues, any idea why it is happening on TIVA? 
The issue seems to happening during the registers copy (FPU Stacking).

@fdcavalcanti maybe you can open a question on TI forum asking why this issue is only happening on their chip, really strange.

fdcavalcanti, 2023-05-20:
Not sure, but I think that line on 'arm_saveusercontext.S' is after the hard fault happens. Using GDB, I can track all the way to 'tiva_configgpio' and as soon as it enters this function, it goes to a hard fault.
Really can't see how enabling the FPU is causing this (because the FPU configuration comes after configuring those GPIOs!)

ldube, 2023-05-29:
@fdcavalcanti I don't see this problem on what I think is the same board you have. 

I used gcc version 9.2.1. This is what I did.

0)  git log
commit 1a927a6cf3fdadc57e6fd8923087ae1e68482164 (H
1) make distclean; ./tools/configure.sh -l  tm4c1294-launchpad:nsh
2) make menuconfig and enable FPU

... and use grep to confirm tha FPU is enabled in config file.
grep FPU .config

# CONFIG_ARM_HAVE_DPFPU32 is not set
# CONFIG_ARM_FPU_ABI_SOFT is not set
CONFIG_ARCH_HAVE_FPU=y
# CONFIG_ARCH_HAVE_DPFPU is not set
CONFIG_ARCH_FPU=y

3)  make CROSSDEV=arm-none-eabi- && lm4flash nuttx.bin
4)
NuttShell (NSH)
nsh> uname -a
NuttX  0.0.0 1a927a6cf3-dirty May 29 2023 15:37:26 arm tm4c1294-launchpad
nsh> 

Please check again.

fdcavalcanti, 2023-05-29:
Hi @ldube I just checked again. Even got a different board (same model) to test it.
```
filipe@fdcavalcanti:~/nuttxspace/nuttx $ git checkout 1a927a6
Previous HEAD position was bcd776b2a7 Fix build issues for ESP32 SoftAP mode
HEAD is now at 1a927a6cf3 mm: add invalid pid dump when malloc failed
filipe@fdcavalcanti:~/nuttxspace/nuttx $ make distclean
NuttX has not been configured!
To configure the project:
  tools/configure.sh <config>
For a list of available configurations:
  tools/configure.sh -L
filipe@fdcavalcanti:~/nuttxspace/nuttx $ ./tools/configure.sh tm4c1294-launchpad:nsh
filipe@fdcavalcanti:~/nuttxspace/nuttx $ make -j
--- Enabled FPU
filipe@fdcavalcanti:~/nuttxspace/nuttx $ ~/lm4tools/lm4flash/lm4flash nuttx.bin
Found ICDI device with serial: 0F007816
ICDI version: 12245
filipe@fdcavalcanti:~/nuttxspace/nuttx $
```
Here's grep on .config:
```
filipe@fdcavalcanti:~/nuttxspace/nuttx $ grep FPU .config
# CONFIG_ARM_HAVE_DPFPU32 is not set
# CONFIG_ARM_FPU_ABI_SOFT is not set
CONFIG_ARCH_HAVE_FPU=y
# CONFIG_ARCH_HAVE_DPFPU is not set
CONFIG_ARCH_FPU=y
```

Still, same result on both boards: only power LED stays on, no serial output.

ldube, 2023-05-29:
@fdcavalcanti Thanks for testing. Please zip your nuttx + nuttx.bin and attach (drag and drop) to this issue. I want to flash it and see what happens.

fdcavalcanti, 2023-05-30:
@ldube attached the binary. Entire directory is too large to add here.
If you confirm issues on your board I'll upload to a personal drive and share.

Also: 
```
filipe@fdcavalcanti:~/nuttxspace/nuttx $ arm-none-eabi-gcc --version
arm-none-eabi-gcc (15:10.3-2021.07-4) 10.3.1 20210621 (release)
```
[issue_9327_nuttx_binary.zip](https://github.com/apache/nuttx/files/11594914/issue_9327_nuttx_binary.zip)

Really appreciate the assistance.

ldube, 2023-05-30:
Your image does not boot on my board. The four LEDs are off too. Does nuttx.bin contain your code changes?

fdcavalcanti, 2023-05-30:
There are no code changes. It is the procedure I showed above.

I'll try and downgrade ARM GCC to the same version you used.

fdcavalcanti, 2023-05-30:
Well then. Not good I guess.
For now I'll downgrade GCC, but not sure how to proceed from here.

@ldube thanks for the help!

ldube, 2023-05-30:
gcc 12.2.0 created a working image too.

ldube, 2023-05-30:
I will try to find the root cause next.

ldube, 2023-05-30:
Because gcc will use FPU registers, the FPU should be enabled early. 
In this case vpush is being used by gcc 10 before the FPU is enabled. 

```
GCC 9.2.1
00000834 <tiva_configgpio>:
     834:       e92d 4ff0       stmdb   sp!, {r4, r5, r6, r7, r8, r9, sl, fp, lr}
     838:       b089            sub     sp, #36 ; 0x24
     83a:       f000 0207       and.w   r2, r0, #7

GCC 10.3.1
0000086c <tiva_configgpio>:
     86c:       e92d 4ff0       stmdb   sp!, {r4, r5, r6, r7, r8, r9, sl, fp, lr}
     870:       ed2d 8b02       vpush   {d8}   <-------- requires enabled FPU
     874:       b085            sub     sp, #20
```

ldube, 2023-05-30:
Here is a patch that corrects the issue

```diff
diff --git a/arch/arm/src/tiva/common/lmxx_tm4c_start.c b/arch/arm/src/tiva/common/lmxx_tm4c_start.c
index 3b04c8aba3..ed7e6b8955 100644
--- a/arch/arm/src/tiva/common/lmxx_tm4c_start.c
+++ b/arch/arm/src/tiva/common/lmxx_tm4c_start.c
@@ -108,9 +108,9 @@ void __start(void)
 
   /* Configure the UART so that we can get debug output as soon as possible */
 
+  arm_fpuconfig();
   tiva_clock_configure();
   tiva_lowsetup();
-  arm_fpuconfig();
   showprogress('A');
 
   /* Clear .bss.  We'll do this inline (vs. calling memset) just to be
```

fdcavalcanti, 2023-05-30:
Works for me! Thanks a lot @ldube .
Will you be opening a PR or should I?

ldube, 2023-05-31:
@fdcavalcanti Please open a PR. Thanks.

closed 2023-05-31 by xiaoxiang781216