apache/nuttx #9349: "C++ Example using CMake" in latest documentation and main branch is broken

dok-net, 2023-05-20:
https://nuttx.apache.org/docs/latest/guides/cpp_cmake.html states:
```
Using the ‘build as a library’ procedure of NuttX, it is possible to build NuttX applications using C++ language and also the cmake build tool.
```

This is not correct, at least on Ubuntu "jammy" or "" (toolchain `Ubuntu (deb)` as per `Installing/Toolchain`).
Using pristine git checkouts (but "download stable release" fares no better), plus `make distclean`, then following the instructions to the word, there are multiple obvious issues.
The include files `cstdio` and `string` are not found.
The exported tar archive does not contain the libs board, gcc (IIRC), supc++.

At least the `sim` target should be supported for this out-of-tree CMake-based build procedure and documented in the step-by-step tutorial style.

With the later C++ releases, more exactly the standard library, I think they are a great showcase for the POSIX support in NUTTX.

acassis, 2023-05-20:
@dok-net thank you for reporting the issue. This Guide was written by @fraviofii ! I'll ask him to review it.

fdcavalcanti, 2023-06-25:
@dok-net did you manage to compile from a export? Just had the same issue as you.

dok-net, 2023-06-25:
@fdcavalcanti Has there been any activity on this issue? I keep checking regularly. No changes, no improvement.

fdcavalcanti, 2023-06-28:
Got it to work with a few modification for my board (TM4C):

- src/CMakeLists.txt:

1. - Commented the linking of supc++
2. - libgcc had to be linked with a harcoded path such as "/usr/lib/gcc/arm-none-eabi/10.3.1/thumb/v7e-m+fp/softfp/libgcc.a"
3. - libxx requires the user to activate the use of C++ on `make menuconfig`
4. - On "add_custom_command" I had to change the .elf directory that was missing `/src/.`

```
set(HEADER_FILES
    HelloWorld.h
)

set(SOURCE_FILES
    HelloWorld.cpp
)

link_directories(${EXE_NAME} ${NUTTX_PATH}/libs)

add_executable(${EXE_NAME} ${SOURCE_FILES} main.cpp ${HEADER_FILES})

add_custom_command(
        TARGET ${EXE_NAME}
        POST_BUILD
        COMMAND ${CMAKE_OBJCOPY} ARGS -S -O binary ${CMAKE_BINARY_DIR}/src/ ${EXE_NAME}.elf ${CMAKE_BINARY_DIR}/${EXE_NAME}.bin
)
message(CMAKE_BINARY_DIR=${CMAKE_BINARY_DIR})

target_link_libraries(${EXE_NAME} --start-group)

target_link_libraries(${EXE_NAME} sched)
target_link_libraries(${EXE_NAME} drivers)
target_link_libraries(${EXE_NAME} boards)
target_link_libraries(${EXE_NAME} c)
target_link_libraries(${EXE_NAME} mm)
target_link_libraries(${EXE_NAME} arch)
target_link_libraries(${EXE_NAME} xx)
target_link_libraries(${EXE_NAME} apps)
target_link_libraries(${EXE_NAME} fs)
target_link_libraries(${EXE_NAME} binfmt)
target_link_libraries(${EXE_NAME} board)
target_link_libraries(${EXE_NAME} net)
target_link_libraries(${EXE_NAME} "/usr/lib/gcc/arm-none-eabi/10.3.1/thumb/v7e-m+fp/softfp/libgcc.a")
# target_link_libraries(${EXE_NAME} supc++)

target_link_libraries(${EXE_NAME} --end-group)
```

- CMakeLists.txt:

1. Modified `include/libcxx` to `include/cxx`

I think that was it. I had some issues with linking to boards and net, but nto sure why it works now.
I also could not get hard-abi to work.

dok-net, 2023-08-05:
> I think that was it. I had some issues with linking to boards and net, but not sure why it works now.
I also could not get hard-abi to work.

"Some issues" :-) I have adapted your changes but due to this problem the issue is still unresolved. It doesn't spontaneously succeed by random chance here.

> libxx requires the user to activate the use of C++ on make menuconfig

Could you please be specific on this?

fdcavalcanti, 2023-08-13:
Hey @dok-net, sorry to keep you waiting.
By activate C++ on menuconfig I meant to go in "Library Routines -> Have C++ compiler" and check this entry.

Regarding the "linking boards and net", I noticed the `libboard.a` was missing, however `libboards.a` was present. I had to go on the NuttX dir, compile everything as usual and copy the now available `libboard.a` file to the exported app.

```
filipe@fdcavalcanti:~/nuttxspace/nuttx$ find . -name "*board*.a"
./boards/libboards.a
./staging/libboards.a
filipe@fdcavalcanti:~/nuttxspace/nuttx$ make -j
CPP:  /home/filipe/nuttxspace/nuttx/boards/arm/tiva/tm4c1294-launchpad/scripts/ld.script-> /home/filipe/nuttxspace/nutLD: nuttx
CP: nuttx.bin
filipe@fdcavalcanti:~/nuttxspace/nuttx$ find . -name "*board*.a"
./boards/arm/tiva/tm4c1294-launchpad/src/libboard.a
./boards/libboards.a
./staging/libboards.a

```

trns1997, 2023-08-24:
@fdcavalcanti i came across the exact same issue where `libboard.a` is missing. There seems to be a bug with `make export` when used directly after configuring your board. The way to by pass this bug (i am guessing it is a bug) for the moment is to to call `make` right before calling `make export`. The problem seems to have already been resolved in the past https://github.com/apache/nuttx/issues/7232 but i am guessing something changed since. Here is a link to my issue: #10388

acassis, 2023-08-24:
@trns1997 @dok-net @fdcavalcanti I think we need to include "make export" in the CI and run some basic SIM build test using C++ to confirm it is working.

trns1997, 2023-08-25:
@acassis if I understand correctly, you are suggesting we create a test cpp project that consumes the output (archive) generated by `make export` to test if the export was successful? Couldn't we simply check that the files generated from the export are all there? Just a suggestion not sure which one is a better approach.

fdcavalcanti, 2023-08-25:
I think that would be a good approach. I would like to also point out that this is behaviour for the basic C++ compiler. When using  LLVM there is another issue pointed in #5530 that I think is relevant to the discussion.

trns1997, 2023-08-28:
@fdcavalcanti check out the following PR #10398. It automatically generates a cmake toolchainfile that you can use to build cpp applications. Note this does not solve the `make export` problem. I will look into that when i have the time.

closed 2023-08-29 by xiaoxiang781216