apache/nuttx #9350: Is it there a runnable version of sqlite on NuttX ?

GooTal, 2023-05-20:
Hi, i notice that a lady or gentleman [ported sqlite to NuttX](https://nuttx.yahoogroups.narkive.com/kMS1Y0mL/porting-sqlite-to-undefined-reference-to-utimes-and-ftruncate) 5 years ago. But it seems some problems occurred.

I wonder if it\`s possbile to run sqlite on NuttX. Specifically, on rv-virt:knsh64 or rv-virt:nsh64 build of NuttX.

Is it possible to read/write files using semihosting(on qemu) both on kernel build and flat build?

acassis, 2023-05-20:
@GooTal yes, it was a FAE from Microchip that did the port many years ago. He shared the port with me and I tried to update it for a new version. It still needs some improvement to integrate on mainline. I'm attaching here the patches that I was testing on ESP32 in 2021:

[0001-apps-Add-SQLite-support-on-NuttX.zip](https://github.com/apache/nuttx/files/11522729/0001-apps-Add-SQLite-support-on-NuttX.zip)
[0001-esp32-devkitc-Add-board-config-to-sqlite.zip](https://github.com/apache/nuttx/files/11522731/0001-esp32-devkitc-Add-board-config-to-sqlite.zip)

xiaoxiang781216, 2024-05-18:
@GooTal Please see this patch https://github.com/apache/nuttx/pull/12305, let's close this pr.

closed 2024-05-18 by xiaoxiang781216