apache/nuttx #9362: WireGuard

pmaddams, 2023-05-23:
@anjiahao1
@xiaoxiang781216
@gregory-nutt
@zx2c4

Hi,

I would like to request that WireGuard be integrated into NuttX. WireGuard is a new and modern VPN protocol that offers a number of advantages over traditional VPN protocols, such as OpenVPN and IPsec. WireGuard is faster, more secure, and easier to configure than traditional VPN protocols. It is also more lightweight, making it a good choice for embedded systems like drones.

Here are some of the advantages of WireGuard:

* **Speed:** WireGuard is significantly faster than traditional VPN protocols. This is because WireGuard uses a new and more efficient cryptographic algorithm.
* **Security:** WireGuard is more secure than traditional VPN protocols. This is because WireGuard uses a number of security features that are not available in traditional VPN protocols, such as perfect forward secrecy.
* **Ease of configuration:** WireGuard is easier to configure than traditional VPN protocols. This is because WireGuard uses a simpler configuration syntax.
* **Lightweight:** WireGuard is more lightweight than traditional VPN protocols. This is because WireGuard uses a smaller code base.

These advantages make WireGuard a good choice for embedded systems like drones. Drones are often resource-constrained, so it is important to use a VPN protocol that is lightweight and efficient. WireGuard meets these requirements, making it a good choice for drones.

[The OpenBSD kernel already has an implementation of WireGuard](https://undeadly.org/cgi?action=article;sid=20200622052207), which could be ported to NuttX. This would provide NuttX users with a secure and reliable VPN option. Searching the NuttX code for references to WireGuard, I found that NuttX already includes crypto code from OpenBSD: https://github.com/apache/nuttx/blob/bc0fe0ea16c8124e56309bf5bb98c363940d4a97/include/crypto/chachapoly.h#L70

I believe that WireGuard integration would be a valuable addition to NuttX. It would provide users with a secure and reliable VPN option, and it would make NuttX more competitive with other embedded operating systems.

Thanks,
Pavan

xiaoxiang781216, 2023-05-23:
What's component WireGuard depends on doesn't support on NuttX yet? It's very nice to support vpn natively on NuttX.

pmaddams, 2023-05-23:
I agree. Thanks for your interest. Here is more information on the [WireGuard protocol](https://www.wireguard.com/protocol/).

xiaoxiang781216, 2023-05-23:
Look like WireGuard just need UDP and the crypto algorithm. The lack crypto algos are Curve25519, SipHash24 and HKDF, which could be supported quickly since crypto is a pure software without any hardware/software dependence. Do you have a plan to port or need help from the community?

pmaddams, 2023-05-23:
I would appreciate help from the community porting WireGuard to NuttX, because it would be of great benefit to everyone. I believe it is a strong security recommendation. I think it is also a good practice for NuttX developers to study OpenBSD to learn about secure development.