apache/nuttx #9371: Security fault in exception_common

zhutouwangzha, 2023-05-24:
I want to port TF-M and NUTTX to my board. There is an error. TF-M can pull up NUTTX image correctly, but when NUTTX executes to bx r14 in exception_common, a Security fault occurs. 
 I see through GDB that the current LR value is 0xfffffff9, why? 
 The calling relationship is as follows ：
__start->go_nx_start->nx_start->nx_bringup->nx_create_initthread->nxthread_create->task_activate->up_unblock_task->up_switchcontext->exception_common。
 SVC 0 in the up_switchcontext will call exception_common, and an error will occur when it returns 。

acassis, 2023-05-24:
@michi-jung worked with TrustedFirmware-M, maybe is could give some hints.

michi-jung, 2023-05-25:
What you are seeing in the LR is the EXC_RETURN value.  Both bits 0 (ES) and bit 6 (S) are set, which means you are trying to return from an exception in the Non-Secure world to the Secure world.  This is no allowed and causes the Security Fault.

You have to set the CONFIG_ARCH_TRUSTZONE_NONSECURE option compile NuttX to be run in the Non-Secure world.  This will define the correct EXC_RETURN then in [exc_return.h](https://github.com/apache/nuttx/blob/efa2a9516311f249500b7065e58ee9f3cc1d74e2/arch/arm/src/armv8-m/exc_return.h#L93)

zhutouwangzha, 2023-05-26:
> EXC_RETURN

 Thanks for the hint, I'm using NUTTX 8.2 and it doesn't seem to have the CONFIG_ARCH_TRUSTZONE_NONSECURE compile option, so I manually changed the value of EXC_RETURN in exc_return.h and it worked fine.

closed 2023-05-26 by zhutouwangzha

mu578, 2023-05-26:
@zhutouwangzha update, or patch diff, there is a natural hierarchy to understand master/slave. You should read arm documentation and conceptual to clarify.