apache/nuttx #9375: esp32c3: uart1 not working

XuNeo, 2023-05-24:
Based on configuration, 
`tools/configure.sh boards/risc-v/esp32c3/esp32c3-devkit/configs/usbconsole`,
enable following configs 
```
+CONFIG_ESP32C3_UART1=y
+CONFIG_ESP32C3_UART1_RXPIN=20
+CONFIG_ESP32C3_UART1_TXPIN=21
```

On device, It will stuck when writing data to ttyS0. Backtrace shows it's waiting in function uart_tcdrain.

nsh> ls /dev
/dev:
 console
 null
 ttyACM0
 ttyS0
 zero
nsh> echo 123 > /dev/ttyS0 ## stuck here

acassis, 2023-05-24:
@XuNeo if you use UART0 and send data do UART1, does it work? Maybe the issue is related to USB SERIAL/JTAG driver

XuNeo, 2023-05-24:
@acassis Hi Alan, 

Thanks for your quick responding. 

I tried it with defconfig of `esp32c3-devkit/configs/nsh`, and  added `+CONFIG_ESP32C3_UART1=y`, unfortunately issue still presents.

fdcavalcanti, 2025-07-23:
`esp32c3-devkit` is not supported anymore, we recommend using `esp32c3-generic`.
This issue is not happening on it.
Closing this issue due to inactivity.

closed 2025-07-23 by fdcavalcanti