apache/nuttx #9392: RP2040 Raspberry Pico W  SPI not working

paolovolpi, 2023-05-25:
I'm facing some issue with the Raspberry Pico W spi (both). Compiling for example the  raspberrypi-pico-w:st7735  project  using  raspberry SDK v1.4.0 and attaching an oscilloscope to the spi1 pins I can't  see any transaction executing the fb application. Same example but for the pico let me see the spi transaction. I already tested a bunch of pico-w without success.

acassis, 2023-05-25:
@paolovolpi can you please try with some old version? Maybe test some version just after st7735 support was added to confirm it was working.

paolovolpi, 2023-05-26:
Same test done with  commit 74f1bfbfd7d578ba74f1885548a8be0901df0c34 confirm the problem. Be careful is not an issue related to the specific application, I'm facing same problem with my personal projects, spi working fine on raspberry pico and fails on raspberry pico-w

paolovolpi, 2023-05-26:
After some more investigation I found that compiling nuttx with raspberry pico sdk 1.5.0 fix the issue BUT the wifi doesn't work with the newer sdk because the firmware blob files are changed between sdk 1.4 and 1.5.

acassis, 2023-05-26:
Hi @paolovolpi thank you for this investigation! We need to find some way to do hardware debug.
Some CI test on real hardware is definitely required!

Few days ago I tested ESP32-Devkit  with GT911 touchscreen and discovered that I2C is not sending all the data correctly. It sends the I2C Address, two bytes of GT911 internal register (ie. 0x8040) but doesn't send the value to that register. Same gt9xx.c driver tested on STM32F4Discovered board worked fine.

paolovolpi, 2023-07-14:
Hi after a lot of more investigation I was able to fix the issue using the  raspberry pico sdk 1.5.0. Still not working with the 1.4.0 but with the 1.5.0 the spi is working as expected (we had some hardware issue with our dev board). I had to replace the sdk 1.5.0 wifi driver (/lib/cyw43-driver) with the 1.4.2 one, because the file structures is changed and is not compatible with the nuttx sources.

acassis, 2024-02-22:
@paolovolpi it worked here! I clone 1.4.0 updated the firmware submodule lib/, then cp -a lib /tmp, clone 1.5.0 removed lib and cp -a /tmp/lib .

paolovolpi, 2024-02-23:
The problem with the spi was related to my custom pcb. The issue related to the SDK still remain. If you want to use sdk 1.5 you need to replace the sdk direcotry lib/cyw43-driver with the one from version 1.4

closed 2024-02-23 by paolovolpi