apache/nuttx #9420: Makes our CI more human readable

acassis, 2023-05-29:
Currently we spend a lot of time trying to figure out why our CI is breaking. Unfortunately it is not human readable and I decided to open this issue to discuss how we could improve it to simplify our life or the life of ours contributors.

I have already some list of issues and suggestion to improve it based on my own experience that I will list below:

--------------------------------------------------------------------------------------------------------------------------------

Issue 1: We need to scroll all the way in the CI building log to find the issue.

Suggested solution: Create a summary of probably causes of the issue at the bottom of the log, it will simplify identifying the issue.

--------------------------------------------------------------------------------------------------------------------------------

Issue 2: CI doesn't explain what to do when board config is outdated

Suggestion: This issue is also related to "Issue 1" (because we need to scroll to find the issue), but it is now clear for new users wait is the issue and what they need to do.
For instance: if some defconfig isn't updated, they could be report in RED color at bottom: You forgot to update the board configs after you modification in the Kconfigs, please run the refresh.sh script.

--------------------------------------------------------------------------------------------------------------------------------

TimJTi, 2023-05-29:
> ...You forgot to update the board configs after you modification in the Kconfigs, please run the refresh.sh script.

never even know about that script! Where is it documented? And (many of) the other tools too for that matter?

acassis, 2023-05-29:
@TimJTi is it not documented, but if you search here in the github PR history you will find other developers explaining how to use it. Unfortunately documentation is our Achilles tendon, this is why I insist to people include documentation when adding a new features. I think we need to do a task force to create more documentation and to document everything that is important to new users

TimJTi, 2023-05-29:
@acassis - OK, I do understand. Can you/others perhaps give a 1-line explanation of what all the tools do, I'll try them and work it out from that, and I will be happy to add some words to help others. Not for 2 weeks though as off on holiday to Greece tomorrow :)

mu578, 2023-05-30:
@acassis CI are dependent of makefile outputs; personally, I prefer give some love to makefiles first, i.e `standardizing` what they spit out, because, I do know that scripting over will be easier, e.g control the mess first.

btashton, 2023-05-30:
> @acassis CI are dependent of makefile outputs; personally, I prefer give some love to makefiles first, i.e `standardizing` what they spit out, because, I do know that scripting over will be easier, e.g control the mess first.
> 

There is a manifest file we generate for each target containing all the expected output files.

btashton, 2023-05-31:
Created this WIP PR to help make the tests results more accessible.  It is not complete yet but I think is in the right direction https://github.com/apache/nuttx/pull/9438

acassis, 2023-06-07:
Sorry I missed those new comments. I need to find configure github to send notication when someone cite my name. I disable github notification because I was receiving tons of messages.

Have a nice vacation @TimJTi 

Nice work @btashton maybe we could give some hints to user on those fails (like failed to mkimport, import, etc)

TimJTi, 2023-06-07:
@acassis 
> Have a nice vacation @TimJTi
> 
I did - thanks!

I have had a quick look and I see that all the tools are described in the readme.md in the /tools directory.

I also see that all of the make file options are covered in the "Building Nuttx" section of the readme.md in the top level of nuttx.

And I hadn't trawled through the readmes as they are too bit and not easy to work with IMHO.

So it seems that what is missing is these being used in web-based documentation with clickable links.

Would that be a fair interpretation or have I , again, missed something?

And, if that's right, what is the best way of making them appear in the online documentation in a nice way?