apache/nuttx #9428: Configuration error

pryan1k69, 2023-05-30:
### I want to add my board to Nuttx. 
Added as in the official instructions.
When I enter `tools/configure.sh -L` then my board is displayed in the list.
But when I try to configure `tools/configure.sh -l myboard:nsh` an error occurs:

```
  Copy files
  Select CONFIG_HOST_LINUX=y
  Refreshing...
CP: arch/dummy/Kconfig to /home/allwinner/Desktop/nuttx-master_1/arch/dummy/dummy_kconfig
CP: boards/dummy/Kconfig to /home/allwinner/Desktop/nuttx-master_1/boards/dummy/dummy_kconfig
LN: platform/board to /home/allwinner/Desktop/apps/platform/
LN: include/arch to arch//include
No directory at /home/allwinner/Desktop/nuttx-master_1/arch//include
make[1]: *** [tools/Unix.mk:272: include/arch] Error 1
make: *** [tools/Unix.mk:655: olddefconfig] Error 2
ERROR: failed to refresh
```

I can't figure out what the problem is. Can you tell me?

TimJTi, 2023-05-30:
A double forward slash is probably the problem but why it's there I don't know!

pryan1k69, 2023-05-30:
I have solved this issue. It was necessary to add the configuration setting to `defconfig` file:
```
CONFIG_ARCH="arm" # for example
CONFIG_ARCH_BOARD="board-name"
CONFIG_ARCH_BOARD_board-name=y
CONFIG_ARCH_CHIP="chip-name"
CONFIG_ARCH_CHIP_chip-name=y
```
 located in `configs/nsh` folder in `boards` folder.

patacongo, 2023-05-30:
> I have solved this issue. It was necessary to add the configuration setting to `defconfig` file:

Are you editing the defconfig files by hand?  That would be a bad idea.  You will make a lot of mistakes and this will cost you time to fix.  You should do:

    make menuconfig

pryan1k69, 2023-05-31:
Yes, it was a bad idea to modify `defconfig` file. When I enter 
`make menuconfig`
 I get a new error. But the problem is that if you do not change `defconfig` file, then I do not know how to configure without the error that occurred above

btashton, 2023-05-31:
One of the common issues I have seen that looks like this when adding a new board, it not making all the required changes.

in `boards/Kconfig` there are three changes you need to make

You need to create an entry here
```

choice
        prompt "Select target board"
        default ARCH_BOARD_CUSTOM
        ---help---
                Select the board hosting the architecture.  You must first select the
                exact MCU part number, then the boards supporting that part will
                be available for selection.  Use ARCH_BOARD_CUSTOM to create a new
                board configuration.

config ARCH_BOARD_AMBER
        bool "Amber Web Server"
        depends on ARCH_CHIP_ATMEGA128
        ---help---
                This is placeholder for the SoC Robotics Amber Web Server that is based
                on the Atmel AVR ATMega128 MCU.  There is not much there yet and what is
                there is untested due to tool-related issues.

config ARCH_BOARD_ARDUINO_MEGA2560
        bool "Arduino Mega 2560"
        depends on ARCH_CHIP_ATMEGA2560
        ---help---
                This option selects the Arduino Mega 2560 board featuring the Atmel
                Atmega2560 MCU running at 16 MHz.

config ARCH_BOARD_ARDUINO_DUE
        bool "Arduino Due"
        depends on ARCH_CHIP_ATSAM3X8E
        select ARCH_HAVE_LEDS
        ---help---
                This options selects the Arduino DUE board featuring the Atmel
                ATSAM3X8E MCU running at 84 MHz.
```
Add to the list here
```
config ARCH_BOARD
        string
        default "amber"                     if ARCH_BOARD_AMBER
        default "arduino-mega2560"          if ARCH_BOARD_ARDUINO_MEGA2560
        default "arduino-due"               if ARCH_BOARD_ARDUINO_DUE
        default "arduino-m0"                if ARCH_BOARD_ARDUINO_M0
```

and here 
```
comment "Board-Specific Options"

if ARCH_BOARD_PCDUINO_A10
source "boards/arm/a1x/pcduino-a10/Kconfig"
endif
if ARCH_BOARD_BEAGLEBONE_BLACK
source "boards/arm/am335x/beaglebone-black/Kconfig"
endif
if ARCH_BOARD_C5471EVM
source "boards/arm/c5471/c5471evm/Kconfig"
endif
if ARCH_BOARD_SPRESENSE
source "boards/arm/cxd56xx/spresense/Kconfig"
endif
if ARCH_BOARD_NTOSD_DM320
```

If you have made those changes you should be able to take an existing config, change the chip and see your board available.

pryan1k69, 2023-06-01:
> One of the common issues I have seen that looks like this when adding a new board, it not making all the required changes.
>
> in `boards/Kconfig` there are three changes you need to make
>
> You need to [...]

Yes, I did as you say. But the error remains exactly the same, although my board is displayed in the list of available

pryan1k69, 2023-06-01:
The error is still the same. That is, it still fails to execute the command: `tools/configure.sh -l myboard:nsh`
If I understand you correctly, then `make distclean` or `make olddefconfig` is possible only after configuration

pryan1k69, 2023-06-01:
Yes

pryan1k69, 2023-06-01:
I think there should be board_name, but for some reason he does not see it

acassis, 2023-06-02:
@btashton AFAIK when you just create a new board (adding it to those 3 positions at Kconfig) you yet still need to have a boardname/configs/nsh/defconfig adapted to this new board. I think it is a dilemma: we cannot edit the defconfig, but we need a defconfig to use with ./tools/configure.sh to this new board. @patacongo am I missing something on this process?

TimJTi, 2023-06-02:
Always easier to start from a board that is similar and use make menuconfig for changes, then make savedefconfig and copy to a new location.

That's what I did, adding my own boards Kconfig as I grew more familiar with it all.

patacongo, 2023-06-02:
> @btashton AFAIK when you just create a new board (adding it to those 3 positions at Kconfig) you yet still need to have a boardname/configs/nsh/defconfig adapted to this new board. I think it is a [...]

I am not aware of many specific problems with this.  The procedure outlined by @TimJTi generally works.  I am aware of a couple of things:

- You can't switch between a POSIX build and a Windows build or vice versa.  The path styles are too different.
- If the average user edits the defconfig file by hand, it will probably break.
- If you reconfigure the architecture, MCU, or board, then the symbolic links that are in place will be wrong.  You need to exit the configuration, `make clean_context`, then `make menuconfig`

TimJTi, 2023-06-02:
Make clean_context also needs documenting

closed 2023-06-20 by pryan1k69