apache/nuttx #9467: assert can cause -Werror=implicit-function-declaration when NDEBUG is enabled

PetervdPerk-NXP, 2023-06-02:
The NuttX assert implementation is as follows when NDEBUG is defined
https://github.com/apache/nuttx/blob/61a46ad39ef215213cb831d5b58834bf2a5a7ad0/include/assert.h#L103

Which can cause a `-Werror=implicit-function-declaration` when code assertion checks like this

```
#ifndef NDEBUG 
static inline bool isokay(bool okay)
{
 return okay
}

void foo() 
{
 assert(isokay(1))
}
#endif
```

When `NDEBUG` is defined the function `isokay` doesn't get declared but the assertion stills refers to it. Causing a `-Werror=implicit-function-declaration` error.
It seems that other implementations of assert() when NDEBUG is as follows
```
#  define assert(f) ((void)(1))
```
Avoiding the  `-Werror=implicit-function-declaration` error.

PetervdPerk-NXP, 2023-06-05:
Sure but everything targeting glibc would be potentially be incompatible rearing the usage of assert() see
```
# define assert(expr)		(__ASSERT_VOID_CAST (0))
```
https://github.com/bminor/glibc/blob/67f704ab69e2305c2b9043d3451df33edbf99b05/assert/assert.h#L50

xiaoxiang781216, 2023-06-05:
Ok, create you create a patch to revert the change of assert.

casaroli, 2026-11-08:
Fixed by #12303.

closed 2026-11-08 by casaroli