apache/nuttx #9501: How to run nuttx(kernel build) on real hardware instead of qemu semihosting?

GooTal, 2023-06-09:
Hi, i`ve already ported Nuttx(kernel build) to LoongArch referring to qemu-rv:virt.

I notice that qemu-rv:virt used semihosting to load nsh_main\`s code into memory from host file system.

But on real hardware, it\`s necessary to load nsh_main\`s code from disk. I\`m trying to figure out how to load the related code.

I\`ve successfully run nuttx(flat build) on real hardware. Here\`s how i run it: I put the elf file under the `/boot/` directory of Linux. And loaded the elf by grub or pmon, then boot the kernel.

I hope to put the nuttx(kernel build) and `apps/bin/sh` under the `/boot/` directory of Linux. And use grub to boot my kernel, too. Is this a feasible way? 

Could anyone provide some suggestions? Thanks a lot.

patacongo, 2023-06-09:
There are some general instructions for a few ARM platforms.  I used the SAMA5D4-EK in the past with the root file system on an SD card.  I put instructions in the board README.txt file:

Using SAMA5D with file system on SD Card
    https://github.com/apache/nuttx/blob/master/boards/arm/sama5/sama5d4-ek/README.txt#L4091

I also noticed these:

Using qemu-system-arm:
    https://github.com/apache/nuttx/blob/master/boards/arm/imx6/sabre-6quad/README.txt#L892
    https://github.com/apache/nuttx/blob/master/boards/arm/imx6/sabre-6quad/README.txt#L929

You could also check the  RISC-V board README files to see if instructioins were provided:

    boards/risc-v/litex/arty_a7/configs/knsh/defconfig:CONFIG_BUILD_KERNEL=y
    boards/risc-v/mpfs/icicle/configs/knsh/defconfig:CONFIG_BUILD_KERNEL=y
    boards/risc-v/qemu-rv/rv-virt/configs/knetnsh64/defconfig:CONFIG_BUILD_KERNEL=y

GooTal, 2023-06-12:
> 
> Yes, but you need mount /boot inside nuttx before nuttx can launch sh
> 
But the `/boot` is on my SSD hard disk and  i didn\`t find the code for SATA driver in nuttx. Is this still a feasible way? 

I can only find out nand flash support. But i\`m running on a Desktop CPU so it\`s quite troublesome to port SATA driver all over again.😀

> 
> If the debugger can connect to your device through JTAG, you can continue use semihosting.
>
Thanks, but the JTAG doesn\`t support it for now.

patacongo, 2023-06-12:
> But the `/boot` is on my SSD hard disk and i didn`t find the code for SATA driver in nuttx. Is this still a feasible way?

Probably not.  There is no SATA support in NuttX.  NAND would not work either because there is no supported NAND-compatible file system.

If you want to use a NuttX native file system, the easiest thing to do would be to use and internal ROMFS file system.  I referred to this above:

https://github.com/apache/nuttx/blob/master/boards/arm/sama5/sama5d4-ek/README.txt#L4127

This requires some special initialization logic:

https://github.com/apache/nuttx/blob/master/boards/arm/sama5/sama5d4-ek/src/sam_bringup.c#L233

GooTal, 2023-06-12:
> Probably not. There is no SATA support in NuttX. NAND would not work either because there is no supported NAND-compatible file system.
>
> If you want to use a NuttX native file system, the easiest [...]

Thank you so much. Let me read the related code.

lupyuen, 2023-07-16:
Hi: I have the same question while porting NuttX to [Star64 JH7110 SBC](https://github.com/lupyuen/nuttx-star64).

Fortunately the SBC runs on U-Boot Bootloader, so I'll try to use the [Initial RAM Disk (initrd)](https://docs.kernel.org/admin-guide/initrd.html) to load the Apps Filesystem with ROMFS.

Thanks for the suggestions! :-)