apache/nuttx #9504: how to use mempool to reslove memory fragement?

Jan9401, 2023-06-09:
Memory pooling is a solution to memory fragmentation issues, but I don't understand MEMPOOL_ CHUNK_ SIZE MEMPOOL_ EXPAND_ SIZE DICTIONARY_ EXPAND_SIZE  What do the three parameters  mean?

Meanwhile, When I set EXPAND_SIZE=64,MEMPOOL_ EXPAND_ SIZE=4096, THRESHOLD=128, system will crash dump immediately.

anjiahao1, 2023-06-09:
MM_HEAP_MEMPOOL_THRESHOLD is a limit, memory less than this value will be allocated from mempool

MM_HEAP_MEMPOOL_EXPAND_SIZE means that if there are no free blocks in the mempool, memory will be requested from the main heap to expand the mempool

anjiahao1, 2023-06-09:
MM_HEAP_MEMPOOL_THRESHOLD set 256,Keep other values as default

anjiahao1, 2023-06-09:
If there are a large number of small memory allocations in the application mempool can optimize memory fragmentation, if not, the effect will not be very obvious.

The reason is that mempool can allocate small memory together to reduce memory fragmentation

Jan9401, 2023-06-09:
> MM_HEAP_MEMPOOL_THRESHOLD is a limit, memory less than this value will be allocated from mempool
> 
> MM_HEAP_MEMPOOL_EXPAND_SIZE means that if there are no free blocks in the mempool, memory will be requested from the main heap to expand the mempool

OK, I understand, by the way, What do the MEMPOOL_ CHUNK_ SIZE parameters mean?

anjiahao1, 2023-06-09:
MEMPOOL_ CHUNK_ SIZE can be set to a super large value, such as 1MB or larger, so that mempool will apply for chunk memory from the heap at one time to manage expend mempool

Jan9401, 2023-06-09:
Thanks for your replay

closed 2023-06-10 by Jan9401