apache/nuttx #9507: ESP32-S3: Add BLE support

pkarashchenko, 2023-06-09:
I want to build a product based on ESP32-S3 and I need BLE running.
I've made some initial effort to enable it based on ESP32 and ESP32-C3 versions in https://github.com/apache/nuttx/pull/9506 and https://github.com/espressif/esp-wireless-drivers-3rdparty/pull/6
I still had to copy `libbtdm_app` from esp-idf to get binary linked. I have not tried to start it on the board still and expect it will not run.
I would appreciate if someone from Espressif can follow-up and finalize my effort.

pkarashchenko, 2023-06-09:
It also would be good if we can align the esp-idf version used for all ESP32 chip family.

donghengqaz, 2023-06-13:
@pkarashchenko If ESP32-S3 BLE is used in real product, we should integrate this [PR](https://github.com/apache/nuttx/pull/4886), because if BLE ISR is blocked by closing interrupt, it will cause an assert when re-enable interrupt and BLE ISR triggers.

elasticdotventures, 2024-10-28:
@tmedicci any update?

tmedicci, 2024-10-28:
> @tmedicci any update?

Hi @elasticdotventures , I didn't get what is missing in your side. ESP32-S3's BLE is implemented and @pkarashchenko even commented on how to use it with nimBLE [here](https://github.com/apache/nuttx/issues/12685#issuecomment-2409152143).

@pkarashchenko , can this issue be closed?

pkarashchenko, 2024-10-29:
@tmedicci I think we can close this.
Answering questions above: I never tried to use BLE while accessing flash programming, so there still may be issues there, but from BLE perspective I think ESP32-S3 is ready to use.