apache/nuttx #9532: Questions about netutils running rv-virt:netnsh64 on branch releases/12.1

GooTal, 2023-06-14:
Hi, i tested some commands about netutils and found something not working as expected. Are these bugs? Did i used these cmds with the right approach? Thanks a lot.

I run these tests on rv-virt:netnsh64, branch releases/12.1.

- tftp put cmd
The `get` cmd works fine. Files could be passed from my tftp server.
[image]
But the `put` cmd is not working well. I could see the `AUTHORS` file created on my tftp server, but nothing in there.
[image]

- nslookup
I try to use nslookup to query Internet domain name servers. As you could see, `baidu.com` is not working well.
[image]

- wget
I\`ve mounted hostfs. `wget` cmd is not working as expected.
[image]

GooTal, 2023-06-15:
Thanks @acassis . But it\`s not working.
[image]

But it did not create any file under `testauthors` with dirname specified.

Without the dirname, it could create the file but nothing in it.

acassis, 2023-06-15:
@GooTal error 110 means ETIMEDOUT
Maybe you will need to use wireshark to understart why the timeout is happening, maybe the tftp server is ignoring the PUT command.

GooTal, 2023-06-16:
Thanks, I\`ll look into it.

GooTal, 2023-06-24:
Thanks. I added `ping` for nuttx of loongarch referring to netnsh64 of riscv64 version.

I try to ping my host. But no interrupt about network arrived.(I checked qemu, no ineterrupt about network raised).

Here`s the description of the bug:
I noticed that `icmp_ping` called `sendto`
```
      nsent = sendto(sockfd, iobuffer, result.outsize, 0,
                     (FAR struct sockaddr *)&destaddr,
                     sizeof(struct sockaddr_in));
      if (nsent < 0)
        {
          icmp_callback(&result, ICMP_E_SENDTO, errno);
          goto done;
        }
      else if (nsent != result.outsize)
        {
          icmp_callback(&result, ICMP_E_SENDSMALL, nsent);
          goto done;
        }
```
`nsent` is the same value(0x40) as rv-virt:netnsh64. 
Then the `poll` is called. But return value always shows time out (1000ms). If i set -1 for time, poll never returns.
[image]

Wireshark is not working for me. Because i used user mode for qemu netdev not tap mode. Wireshark can only parse the log that qemu dumped. So i used the `-object filter-dump,id=o1,netdev=u1,file=./netdump `. But it seems nothing in netdump:
[image]

I want to make sure host has received ping packet. Does the `nsent` value mean my host received packets successfully?
If i can make sure host received packets, it would be easier to debug. Thanks

GooTal, 2023-06-27:
Oh, thanks. I could successfully ping my host now.